ovro-lwa/lwa-issues #307: Beam observation of B1919+21

iveydavis, 2023-03-15:
Minimum of 2 beams:

1.  Beam on B1919+21
2.  Beam on Cygnus A
3.  Optional 3rd beam on zenith

iveydavis, 2023-03-17:
The original goal of this observation was to look at a pulsar we can absolutely see because there was some issues pulling out the pulses for other pulsar observations. But Nikita and Jayce were able to resolve the problem and the pulses are popping out! The OVRO LWA power beams can indeed see pulsars.

[image]

[image]

closed 2023-03-17 by iveydavis

rh-codebase, 2023-03-17:
Is there an issue number describing what was wrong and how it was resolved? Is so, please cross-reference.

iveydavis, 2023-03-17:
I don't think an issue was ever opened for the difficulty with pulsar detection. It ended up being an issue on the data processing side-- the pulsar signal was being flagged by the Presto software because it was so bright

caseyjlaw, 2023-03-17:
@rh-codebase We are using github issues to track commissioning tasks. They are organized into the sprint topics as a "project". Sometimes the issue description is not a problem, but a feature, like this one.

rh-codebase, 2023-03-17:
That's fine but without the explanation as to how this was resolved it left open whether it was a sw bug or something else.