ovro-lwa/lwa-issues #316: Need new account on major

caseyjlaw, 2023-03-28:
Ivey Davis would like an account on major (131.215.200.144) for OVRO-LWA observing purposes.

They would like username "idavis" using bash shell with the public key:
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABgQCjmqqtfLaA+pg8FXdQlgex0obRDLH7z2mZF/1u2B4tC3C6snrhZfu5HuZjCHkFtCz4009qXFP9TgCkr1ogi6Zo7VFr+lGrMVhI1AMdfn5szzx+wxLNKAjTYeMS/Hbr3CLPL+5974LnBn3CvbocgeET1cbYJ7lEy+9gZBJ6S2tkOtVrtXRwd4txKWZb/e/Pqksk6iVfzN4CgKEowrIkY1YK8hwKawzy/PbKa12eu/uxLzzNg5T2LkgBBUgscmyctDkovPXwI9LlKWQCdivSvGCGY4KMzwIqlXJjqT5DzPvrmxXaroCvrc9n6cz3PZn50xt6BvFxm1Bs2b6jLJV6HpY9kPqSTmsvx6kZ+qoUJx1B8MsqUeaXl8xDt5CUTuvYXlvA6BF+5avKCPPOj4urZWfo2Rm4k7/lDGf21Q0yo+CmsXbjXrYJCHw9vNlseFy4UDZCckJDK7938XWC3LGzAvzFD7GMBhQ8/JB05CU2jEvPo05E6PglbEYoBQKMythiDuM= idavis@pop-os

rh-codebase, 2023-03-28:
I don't have any access to 'major' or any root access to any machine in the IP 131.215.x.x space.

caseyjlaw, 2023-03-28:
Ok, thanks. We'll check with Vikram.

closed 2023-03-28 by caseyjlaw