ovro-lwa/lwa-issues #317: MS reading tool

caseyjlaw, 2023-03-29:
Many users want to read MS data. We should develop a tool to make that easy.
Features:
- One-command read for all subbands
- Select correlations (e.g., only self)
- Speed could be improved by reading into RAM and using async file handling.

Need to check on existing solutions (sdmpy, ngcasa, xarray-based stuff).

caseyjlaw, 2023-03-29:
@marinanderson @ldaddario @nmahesh1412: Can you post a link to your code or paste it below as examples of usage?

ldaddario, 2023-03-30:
Some code for extracting self correlations is at
lwacalim*:/home/ldaddario/extractSelfCorr.py

USAGE:
import extractSelfCorr as sc
(data, time) = sc.allFreq(path,file,outfile)

This will read measurement sets for all subbands of one integration, extract all self-correlations and combine the subbands into array 'data' of shape (352,3072,2), and optionally (if argument 'outfile' is given) write the results to a Matlab file.  Output 'time' is the time number from the measurement sets; it should be the same for all of them (this is checked, and if not a warning is printed).
'path' should be the path to the directory containing the full observation.  It is assumed to have a subdirectory for each subband with names ['13MHz, '18MHz', ..., '82MHz'].  If any of those subdirectories is missing, the corresponding elements of 'data' are filled with zeros. 
'file' must be the filename of one .ms directory for the desired integration, /without/ the '_<subband>.ms' ending.  It is assumed that a file with name 'file*.ms' exists in each subband's subdirectory, and it is those measurement sets that are read.  If the file is missing for any subband, the corresponding part of 'data' is filled with zeros.
'outfile' (optional) is the name of a Matlab binary data file (.mat) into which to write 'data' and 'time'.

caseyjlaw, 2023-04-06:
I looked into pyuvdata and find it much more intuitive than casacore. Instructions for the version I installed are at https://pyuvdata.readthedocs.io/en/v2.1.2/tutorial.html.

I did a quick test in the `development` environment of calim00 on data in /mnt/24-hour/run. Here's some example syntax to retrieve the autos (all-pols):

```
> from pyuvdata import UVData
> uvd = UVData()
> fl = sorted(glob.glob('*MHz/20230309_162301*ms'))
> uvd.read(fl, axis='freq')  # this takes about 2s per MS file in the list
> uvd.data_array.shape
 (352, 1, 2688, 4)
> uvd.freq_array
array([[13398437.5    , 13422363.28125, 13446289.0625 , ...,
        86826660.15625, 86850585.9375 , 86874511.71875]])
> uvd.time_array
array([2460013.18266165, ...
```

There is a lot more it can do, so it would be worth going over our use cases for MS work.

ldaddario, 2023-04-06:
On 4/6/2023 06:29, Casey Law wrote:
> I did a quick test in the |development| environment of calim00 on data 
> in /mnt/24-hour/run. Here's some example syntax to retrieve the autos 
> (all-pols):
> 
> |> from pyuvdata import UVData > uvd = UVData() > fl = 
> sorted(glob.glob('*MHz/20230309_162301*ms')) 

> |> uvd.read(fl, axis='freq')
> # this takes about 2s per MS file in the list
> |> uvd.data_array.shape
> (352, 1, 2688, 4)
> |> uvd.freq_array
> array([[13398437.5 , 13422363.28125,
> 13446289.0625 , ..., [...]

Casey,

I don't see anything in your syntax that specifies selecting only the 
self-correlations.  Of course, it's really not just the 
self-correlations because you include the cross-correlations between 
polarizations.

2s per MS is very slow.  There are 9112*14 MSs in the 24h run, so it 
will take 3 full 24h days to read all of them.  My script using casacore 
took about <2 full days.

Does the uvd.read() method close all files of the MS after reading? 
Yuping pointed out to me that casacore does not, so the caller must 
explicitly close each open MS.

I notice that the uvd.time_array is in JD, not MJD nor MJD*86400, so 
apparently it does not just return what's in the MS.

--Larry

-- 
=======================================================
Larry R. D'Addario
Department of Astronomy
California Institute of Technology
Pasadena, California 91125

291 Cahill, mail code 249-17
email:  ***@***.***
phone:  +1/626/395-1431 (office)
         +1/626/510-6492 (home)
         +1/626/657-6357 (mobile)
=======================================================

caseyjlaw, 2023-04-06:
Larry, there is also syntax for selecting polarizations. I believe it would be just one more line like this:
`uvd.select(polarizations=[-5, -8])`

But it is indeed slower than real time. @rlbyrne, you may have the most experience with pyuvdata. Do you know of any tricks to improve read and/or selection speed?