ovro-lwa/lwa-issues #318: Beamformed reading tool

caseyjlaw, 2023-03-29:
Many people are playing with beamformed data. We should develop a single package/tool to avoid duplication and divergence.
Features:
- handle metadata
- averaging
- plotting

@iveydavis, @nkosogor: can you share any relevant code you have (link to github or just paste it).

iveydavis, 2023-03-29:
My code lives on lwacalim01:/opt/devel/idavis/BeamTools.py
Definitely needs some sprucing up, but it can average over specific time and frequency ranges, to a specified time and spectral resolution, and can plot the associated dynamic spectrum, light curve, or spectrum.
I go over how to use it a little bit here: https://caltech.sharepoint.com/sites/ovro/projects/SitePages/Beamforming-Tutorial.aspx#working-with-the-data

iveydavis, 2023-03-29:
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
"""
Created on Wed Mar  1 10:45:58 2023

@author: idavis
"""
from enum import Enum
import h5py
import matplotlib.pyplot as plt
import numpy as np
import datetime
from astropy.time import Time

def PickPol(file,stokes):
    acceptable_stokes = ['XX','YY','XYI','I','V','Q','U']
    if stokes not in acceptable_stokes:
        raise Exception('Stokes needs to be I, V, Q, U, XX, YY, or XYI')
    if stokes == 'XX':
        data = file['Observation1']['Tuning1']['XX'][:,:]
    if stokes == 'YY':
        data = file['Observation1']['Tuning1']['YY'][:,:]
    if stokes == 'XYI':
        data = file['Observation1']['Tuning1']['XYI'][:,:]
    if stokes == 'I':
        data = file['Observation1']['Tuning1']['XX'][:,:] + file['Observation1']['Tuning1']['YY'][:,:]
    if stokes == 'V':
        data = 2 * file['Observation1']['Tuning1']['XY_imag'][:,:]
    if stokes == 'Q':
        data = file['Observation1']['Tuning1']['XX'][:,:] - file['Observation1']['Tuning1']['YY'][:,:]
    if stokes == 'U':
        data = 2 * file['Observation1']['Tuning1']['XY_real'][:,:]
    return data

def GetData(fn,stokes,n_time_avg = 0,n_freq_avg = 0):
    file = h5py.File(fn,'r')
    freqs = file['Observation1']['Tuning1']['freq'][:]
    times = timestamp_to_mjd(file['Observation1']['time'])[0:-2]
    data = PickPol(file, stokes)[0:-2,:]
    
    if n_freq_avg != 0:
        data,freqs = AverageInFrequency(data, freqs, n_freq_avg)
    if n_time_avg != 0:
        data, times = AverageInTime(data, times, n_time_avg)
        
    return data, times, freqs

def timestamp_to_mjd(times):
    time_final = []
    for t in times:
        ts = datetime.datetime.fromtimestamp(t[0]).isoformat()
        ts += str(t[1])[1:]
        time_final.append(Time(ts,format = 'isot').mjd)
    return np.array(time_final)

def AverageInTime(data, times, n_avg):
    times_new = times[::n_avg][()]
    data_new = np.zeros((len(times_new),data.shape[1]))
    for i in range(len(times_new)):
        data_new[i,:] = np.mean(data[i*n_avg:(i+1)*n_avg,:],0)
    return data_new,times_new

def AverageInFrequency(data, freqs, n_avg):
    freqs_new = freqs[::n_avg][()]
    data_new = np.zeros((data.shape[0],len(freqs_new)))
    for i in range(len(freqs_new)):
        data_new[:,i] = np.mean(data[:,i*n_avg:(i+1)*n_avg],1)
    return data_new,freqs_new

def mjd_to_hms(times):
    ts = Time(times,format = 'mjd').ymdhms
    tf = []
    for t in ts:
        hour = t[3] + float(t[4])/60 + t[5]/3600
        tf.append(hour)
        if tf[-1] < tf[0]:
            tf[-1] = tf[-1] + 24
    return tf

def mjd_to_timesincestart(times):
    ts = Time(times,format = 'mjd')
    t0 = ts[0]
    tv = []
    for t in ts:
        dt = t - t0
        dt_hr = dt.to('hr').value
        tv.append(dt_hr)
    return tv

def PlotDynSpec(data,times,freqs,stokes='',obj ='',n_time_avg=0,n_freq_avg=0,doLog = True, doAbs = False,vmin= None,vmax = None,title = '',freq_range = [],time_range = []):

    if doAbs:
        data = np.abs(data)
    if doLog:
        data = np.log10(data)
    if freq_range != [] and len(freq_range) == 2:
        inds = np.where((freqs > freq_range[0]) & (freqs < freq_range[1]))[0]
        freqs = freqs[inds]
        data = data[:,inds]
    if time_range != [] and len(time_range) == 2:
        inds =  np.where((times > time_range[0]) & (times < time_range[1]))[0]
        times = times[inds]
        data = data[inds,:]
    t_hour = mjd_to_timesincestart(times)
    extents = [t_hour[0],t_hour[-1],freqs[0]/1e6,freqs[-1]/1e6]
    plt.figure()
    plt.imshow(data.transpose(),aspect = 'auto',origin = 'lower',vmin= vmin,vmax = vmax,extent=extents)
    plt.colorbar()
    plt.xlabel('Time Since Start [hour]')
    plt.ylabel('Frequency [MHZ]')
    title = obj + ' Stokes ' + stokes + ', '+Time(times[0],format ='mjd').isot
    plt.title(title)
    plt.tight_layout()
    plt.show()
    return

def PlotLightCurve(data,times,freqs,stokes='',obj ='',n_time_avg=0,n_freq_avg=0,doLog = True,freq_range = []):
    
    if freq_range == []:
        print('No frequency range specified; Averaging over all frequencies')
        lc = np.mean(data,axis = 1)
    if freq_range != [] and len(freq_range) != 2:
        raise Exception('Expecting two frequencies, [low,high]')
    if freq_range != [] and len(freq_range) == 2:
        inds = np.where((freqs > freq_range[0]) & (freqs < freq_range[1]))[0]
        lc = np.mean(data[:,inds[0]:inds[-1]],axis = 1)
    ts = mjd_to_timesincestart(times)
    if doLog:
        lc = np.log10(lc)
    plt.plot(ts,lc)
    plt.xlabel('Time Since Start [hour]')
    title = obj+ ' Stokes ' + stokes + ', '+Time(times[0],format ='mjd').isot

    plt.title(title)
    plt.show()
    return

def PlotSpectrum(data,times,freqs,stokes='',obj= '',n_time_avg=0,n_freq_avg=0,doLog = True,time_range = []):
    if time_range == []:
        print('No time range specified; Averaging over all times')
        spec = np.mean(data,axis = 0)
    if time_range != [] and len(time_range) != 2:
        raise Exception('Expecting two times, [low,high]')
    if time_range != [] and len(time_range) == 2:
        inds = np.where((times > time_range[0]) & (times < time_range[1]))[0]
        spec = np.mean(data[inds[0]:inds[-1],:],axis = 0)
    if doLog:
        spec = np.log10(spec)
    plt.plot(freqs/1e6,spec)
    plt.xlabel('Frequency [MHZ]')
    plt.title(obj+ ' Stokes '+stokes)
    plt.show()
    return

```

jaycedowell, 2023-03-29:
The HDF5 should also be compatible with the scripts at https://github.com/lwa-project/commissioning/tree/main/DRX/HDF5

caseyjlaw, 2023-05-03:
Bin's team has developed this tool: https://github.com/binchensun/suncasa/blob/master/suncasa/dspec/dspec.py.
Example syntax in ipython:
```
# in ipython
from suncasa.dspec import dspec
d = dspec.Dspec()
# file path of the 8-hr beamforming data on 2023 May 5
fn = ‘/data07/bin.chen/20230502/060066_153223940292804009f’
# read and plot Stokes I in a given time range and frequency range. Default to [] to take everything. timebin & freqbin are number of measurements for binning.
d.read(fn, obstype=‘lwa’, stokes=‘I’, timerange=[‘2023-05-02T18:00:00’, ‘2023-05-02T20:00:00’], freqrange=[25., 80.], timebin=10, freqbin=1)
d.plot(freq_unit=‘MHz’, vmin=10, vmax=2e3)
# read and plot Stokes V
d.read(fn, obstype=‘lwa’, stokes=‘V’, timerange=[‘2023-05-02T18:00:00’, ‘2023-05-02T20:00:00’], freqrange=[25., 80.], timebin=10, freqbin=1)
d.plot(freq_unit=‘MHz’, vmin=-5., vmax=5., norm=‘linear’, cmap=‘bwr’)
```