ovro-lwa/lwa-issues #322: OVRO-LWA Scheduler proposal

caseyjlaw, 2023-03-31:
OVRO-LWA has multiple observing modes that operate commensally. This motivates careful thought about scheduling and coordination of observations. Currently, observations are controlled in a single python session. That raises issues when tracking with beams on an update loop (#291). It also does not allow for automation, such as triggered observing (#238). Here, I propose a scheme for a new scheme for automatic scheduling and observing via a REST API.

The core of the new scheme is a single service that receives scheduling blocks (i.e., from an SDF) and keeps track of all submitted blocks. This service takes the form of a REST API with a route for each data recorder (e.g., `/dr1`, `/drvs`, ...). Each route can receive "PUT" and "GET" calls.

A observing schedule can be "PUT" to a route (minimal requirement is a beam number, start time, and duration). The service will check whether that block can be scheduled. If yes, it will be saved to some internal state (maybe a database). If not, a warning is returned to the caller. Generally, data recorder functions should be executable via PUT calls (e.g., cancel something in a queue), but that could be implemented later.

A "GET" on a route will return the scheduled observation for a given data recorder. The default is to return the current scheduled observation, but a parameter "mjd" can be used to return other times. This can be used by observers, but primarily will be used by x-engine control services.

X-engine control requires a new service per beam. These services will run "GET" calls to look for the current schedule of a beamformer. Based on the schedule, the service will run x-engine control commands (e.g., calibration or track a beam). The control commands are already available in `xengine_beamformer_control.py`

Comments welcome: @jaycedowell @jack-h @yupinghuang.

jaycedowell, 2023-03-31:
I have comments but it's too late on a Friday afternoon to post them.  I'll try again on Monday.

caseyjlaw, 2023-04-03:
Good point, Rick. I was just trying to capture the idea after Ivey and I were brainstorming a bit.

Some proposed OVRO-LWA Scheduler requirements:
1. Set up observations in advance -- We want to be able to submit an observation up to 24 hours in advance. 
2. Control commensal observations -- Different observations (data recorders, beam control) should be schedulable and controllable independently. This requires separate threads/processes for beam control. #291 
3. Control of all subsystems -- The scheduler should be able to submit commands to f-engine, x-engine, and data recorder subsystems (i.e., execute `mnc-python` functions).
4. Visibility -- Users should be able to see what has been scheduled up to 24 hours in advance. Any user can submit, but submission should check on whether requested observations collide or interfere with other observations.
5. Calibrated beams -- Beamforming observations should be schedulable such that they apply good calibration for the time of observation (note: this means making and applying new tables across day/night boundaries).
6. Automated scheduling -- It should be possible for an automated process (a script in a loop) to decide to submit an observation. An important application of this is ASAP beamformed observations of FRBs detected by DSA-110. #238

jaycedowell, 2023-04-03:
I get the idea behind this kind of PUT/GET framework but I think it adds a lot of complication to running/controlling an observation.  From the description it sounds like observers PUT in a schedule and it is up to the subsystems to poll via GET to figure out what to do and when to do it.  That pushes a lot of the mechanics to run an observation down to the subsystems and makes it dealing with interdependencies between subsystems tricky.  Plus, if you want to add something like a new beam tracking mode then you have multiple pieces of software that need to be updated and tested to make sure everyone is on the same page.

At least for our stations this is what the station's monitor & control software is for.  It receives SDFs from observers (PUTs), validates that they can be run without conflicts, and then converts those observations into what we call command scripts.  Those command scripts are then used by the station's MCS to send the relevant commands to the various subsystems at the correct times needed to run the observation.  At the end of the observation the station's MCS also creates a set of metadata, including the command script, that describe what happened when and where data were recorded.  This makes our approach more like the PUT and service parts of this proposal but without GET and where the service is more like a more flexible/robust version of `run_sdf.py`.

For the other point mentioned of sending commands to subsystems like the "cancel something in a queue" I don't really see the need for another PUT/GET layer on top of or in place of the etcd-based command-response framework.  If a user wants to send a specific command to a specific subsystem they can do that now.  If "cancel something in a queue" is a more high-level concept like canceling an observation that would be in the realm of the station's MCS since canceling can include many operations.

caseyjlaw, 2023-04-03:
Fair point. The design I suggested also shares some features with the key-value store, which could add confusion.
One reason we thought of a new scheme was that we were a bit unclear how how to adapt the SDF (files and script) to OVRO-LWA, esp f-eng and x-eng control.
@jaycedowell could you give us an intro on Wednesday's meeting and we can discuss the requirements listed above?

jaycedowell, 2023-04-03:
The F- and X-engine control are baked into station's MCS so that what is needed in terms of commands is derived from the observation parameters.  For things outside of standard conditions there are a limited number of overrides that can be set through the SDF.  These would be things like the analog filters/gain settings and beamformer weightings to some extent.

I can but I'm only available for the first hour.

rh-codebase, 2023-04-03:
What's the definition of 'station's MCS'?  Hearing what Jayce has may help refine the requirements but I'd still prefer to get sign-off on the requirements before in depth implementation discussions. 

Is requirement 6. above - External triggered observations? Is there a need to distinguish internal 'interrupt  triggers' vs 'external' ones?
req 2 and 3 still have implementation embedded.

Not sure if this issue is the best place to hack on requirements.  ?? Perhaps a new section in the existing Req. doc.?

jaycedowell, 2023-04-03:
I can talk more about this on Wednesday but I would say our "station MCS" has four parts:
 * There is a definition of how the subsystems are to communicate with each other via UDP packets, what the minimum set of required commands that a subsystem needs to support are, and what the minimum set of required monitoring points are.
 * "Task processor" which takes in SDFs and does a preliminary validation of the session (valid observing mode, tuning within the frequency range of the digital system, etc.) and checks for conflicts.
 * The "executive" receives valid sessions from the task processor and converts the observation into a series of subsystem commands and execution times which it sends out at the appropriate time.  The executive also does a regular polling of key monitoring points with the goal of getting them into the logs.
 * The "scheduler" which deals with the actual communication with the subsystems and keeping track of the command responses and maintaining the state of the monitoring points.  It receives commands from the executive as well as operators.

Here are a few memos that talk a little more about the station MCS and observing:
 * [MCS0024](https://leo.phys.unm.edu/~lwa/lwavt/doc/091120_MCS0024_CDR_Pres.pdf)
 * [MCS0030](https://leo.phys.unm.edu/~lwa/lwavt/doc/MCS0030v5_Observing.pdf)
 * [MCS0031](https://leo.phys.unm.edu/~lwa/lwavt/doc/MCS0031v5_StationMIB.pdf)

Sitting above the station MCS is the HAL system which implements the higher level parts of station control.  Things like how to attempt to resolve observing conflicts, how to handle triggered observation, and running basic station/environmental monitoring.  It's our attempt to largely replace observers with software that can react to things more quickly.

caseyjlaw, 2023-04-06:
First test of `run_sdf.py` looking good. See #326.

caseyjlaw, 2023-07-26:
SDF writer will be added to `mnc-python` by Ivey.
SDF parser and scheduler will be implemented in new repository, `lwa-observing`.

caseyjlaw, 2023-08-04:
Taking another look at this with Jayce's suggestion and considering etcd features.

We can create two services that run on cal-im:
- Processor -- takes SDF files, validates them, checks for scheduling conflicts, and (if valid) creates unique name for the session
- Executor -- takes validated SDFs, parses and adds them a command queue, and runs commands (translates sdf to `mnc-python` commands)

Etcd callback functions can connect them:
- Processor watches `/cmd/observing/processor`
  - expects an SDF filename
  - validates by trying to read file
  - compares requested execution times to schedule in `/mon/observing/schedule` (start/stop time for sessions; one per mode)
  - after parsing, it sets `/resp/observing/processor` with session name (or error message)
  - after successful parsing, it sets SDF filename in `/cmd/observing/executor`
  - cancelling an observation with command "cancel" and a session name
- Executor watches `/cmd/observing/executor`
  - expects a validated, schedulable SDF filename
  - reads it, adds commands to a queue (kept in memory or etcd?)
  - updates `/mon/observing/schedule` with new session
  - submits commands for execution at scheduled time

Features:
- Python client can hide etcd API behind functions (e.g., "validate", "submit", "get_schedule", etc.)
- Users can submit an SDF from anywhere and get a confirmation
- Can schedule a session for any mode at any time
- allows automated, low-latency observations
- Schedule is available to anyone (easy to make web page to display it)

Things that need more thought:
- calibration -- what scenarios might we expect
- parallelism model for executing commands (esp tracking beams)

caseyjlaw, 2023-08-07:
Executor protoype working in [#3](https://github.com/ovro-lwa/lwa-observing/pull/3).

Submit SDF from command line with `lwaschedule submit-sdf <full-path-to-sdf>`.
View schedule for each observing mode with `lwaschedule show-schedule`.

caseyjlaw, 2023-09-21:
Correction: `lwaobserve submit-sdf <sdf file>`

caseyjlaw, 2023-11-21:
New feature idea: database of scheduled observations, data products, and array state information.
The scheduler manages submission of sessions and their execution. However, it does not know whether data recorders produced any output or whether the array was able to produce useful data. A goal is to allow someone to see a data product (e.g., beamformed h5 file) and infer whether it is useful. We could also use this to expose info to a web app for the scheduler and/or data recorders.

Potential features:
- [ ] database 
- [ ] table with an entry per scheduled observation
- [ ] table with an entry per recorded product that references a row of scheduled observations
- [ ] fields in a table should include array state (e.g., day/night settings, good/bad antennas, calibration loaded)
- [ ] the table should be extensible to include pipeline products (e.g., images)
- [ ] web app that lists rows of scheduled and recorded data tables
- [ ] Python module in `lwa-observing` that provides API to adding observations/data to database

Issues to think about:
- [ ] What's the best database for this purpose? I'd use sqlite3 in Python, unless there are strong opinions.
- [ ] How to support visibility recorder start/stop as an "observation"?
- [ ] Submitting an SDF should produce an entry to the observation database
- [ ] Do we need data recorders to use this API or is it bad to couple these togther?

yupinghuang, 2023-11-22:
Incidentally, I installed MariaDB on calim10. It's basically mysql.

I mean...if you're not doing anything terribly fancy (joins, transactions), what you're describing with sqllite sounds like a text file buuuut I've never used sqllite and I am sure I am missing things.

You could probably have the data recorder send things to etcd and then have your consumer make the database if that's what you want to do.

Re settings -- I feel strongly that the info should live with the data. e.g. extra field in the hdf5. Also there's gotta be one of those 10000 fields in a measurement set that we can co-opt to indicate nighttime vs daytime https://casa.nrao.edu/Memos/229.html#SECTION000611000000000000000.

Is this enough for keeping track of observations? Part of this is to keep sensible directory structures for all data products so that we don't need a mapping of say, time, product, and frequency to files because...the files are already ordered that way.
```
(base) [yuping@lwacalim10 slow]$ pwd
/lustre/pipeline/slow
(base) [yuping@lwacalim10 slow]$ tree -L 2
.
├── 13MHz
│   ├── 2023-11-16
│   ├── 2023-11-17
│   ├── 2023-11-18
│   ├── 2023-11-19
│   ├── 2023-11-20
│   └── 2023-11-22
├── 18MHz
│   └── 2023-11-22
├── 23MHz
│   └── 2023-11-22
├── 27MHz
│   └── 2023-11-22
├── 32MHz
│   └── 2023-11-22
├── 36MHz
│   └── 2023-11-22
├── 41MHz
│   └── 2023-11-22
├── 46MHz
│   └── 2023-11-22
├── 50MHz
│   ├── 2023-11-17
│   ├── 2023-11-18
│   ├── 2023-11-19
│   ├── 2023-11-20
│   └── 2023-11-22
├── 55MHz
│   └── 2023-11-22
├── 59MHz
│   └── 2023-11-22
├── 64MHz
│   └── 2023-11-22
├── 69MHz
│   └── 2023-11-22
├── 73MHz
│   └── 2023-11-22
├── 78MHz
│   └── 2023-11-22
└── 82MHz
    └── 2023-11-22
```

```
(base) [yuping@lwacalim10 pipeline]$ tree -L 2 beam??
beam01
beam02
├── 060269_154500000000e472a83
├── 060269_16155999900038fbaf1
├── 060269_1647000000008d7e33c
├── 060269_171759999000e23e59d
├── 060269_17485999900036f6e06
├── 060269_1820000000008b7c637
├── 060269_185059999000e047fe4
├── 060269_19215999900034bd1a1
├── 060269_1953000000008982c17
├── 060269_202359999000de0437d
├── 060269_202659999000497eb65
├── 060269_20550000000032ae695
├── 060269_21260000000087712ae
├── 060269_215659999000dbee9f1
├── 060269_22280000000030c2914
├── 060269_225900000000854052e
├── 060269_232959999000da08faf
├── 060270_1545000000000ea03ed
├── 060270_161559999000636d7bb
├── 060270_164700000000b7ef123
├── 060270_1717599990000cc1f25
├── 060270_1748599990006137f11
└── 060270_182000000000b5fd433
beam03
├── 060269_193439830259fa2982a
└── 060270_054854498084c943878
beam04
beam05
└── 060270_043259999000331d986
beam06
└── 060270_04325999900033211f9
```

caseyjlaw, 2023-11-29:
Correct metadata stored in headers is definitely the right thing to do. @iveydavis can you open an issue to summarize what you need added/fixed in the beamformer headers? I think calibration and settings files are two important ones that likely are not there yet.
Today, I realized that my propose scheme above assumes that settings/calibration are tied to sessions, but that is not true. They can be changed independently of the scheduler and/or the data recorders. I think good metadata attached to data products is the right way to go.
I'll try to refocus this issue on providing visibility into the scheduler. So:
- [ ] table for scheduled sessions (and completed sessions)
- [ ] submitting sdf adds to the session table
- [ ] table for outputs of the data recorders 
- [ ] recorders adds to that table with time stamp that helps associate it to a session
- [ ] web app displays both tables

Optionally:
- [ ] table for settings file changes (replacing log file kept on disk)
- [ ] table for caltable changes
- [ ] table of data products could support imaging pipeline products
- [ ] web app displays these extra tables

iveydavis, 2023-11-30:
info on the metadata in the beam hdf5 files: https://github.com/ovro-lwa/lwa-issues/issues/497

caseyjlaw, 2023-12-04:
Some of these features are implemented in a service running at lwacalim10:8082.

caseyjlaw, 2024-02-02:
The basic function of the scheduler is working, so I'll close this.
The data products are not included in the `obsstate` database because I'm not sure we really need it there. A better focus would be to get observing state (settings, good antennas, etc) as metadata tied to data products.

closed 2024-02-02 by caseyjlaw