ovro-lwa/lwa-issues #323: Problems finding pipelines for beamforming

iveydavis, 2023-04-03:
Sherry, Marin, Brian, and I were trying to beamform this morning. Since SNAP01 had a digitizer replaced on Friday, we started by reprogramming it and then restarting all of the boards from the CR node:

```
from lwa_f import snap2_fengine 
CONFIG = '/home/ubuntu/proj/lwa-shell/caltech-lwa/control_sw/config/lwa_corr_config.yaml'
FPGFILE = '/home/ubuntu/proj/lwa-shell/caltech-lwa/snap2_f_200msps_64i_4096c/outputs/snap2_f_200msps_64i_4096c_2023-03-23_1851.fpg'

SNAPS = [*range(1,12)]  

ff = [] 

for f in ff: 

     ff += [snap2_fengine.Snap2Fengine('snap%.2d' % s)] 

f[0].program(FPGFILE)
for f in ff: 
     f.cold_start_from_config(CONFIG)
```

This went relatively smoothly. We then loaded in the daytime settings on calim00:
`python ~/proj/lwa-shell/mnc_python/scripts/loadFengineAndArxSettings.py  ~/proj/lwa-shell/mnc_python/data/20230309-settingsAll-day.mat`

In ipython and the datarecorder environment, we went through the typical steps of prepping the beamformers:
```
from mnc import control 
con = control.Controller('config/lwa_config_calim.yaml') 
con.stop_xengine() 
con.configure_xengine(recorders=['dr1'], full = True, calibratebeams = True) 
```
At this last step though, we ran into an issue where none of the relevant pipelines were found to calibrate the beam. I've attached the output of the attempt here as `calibration_output.txt`
[calibration_output.txt](https://github.com/ovro-lwa/lwa-issues/files/11142762/calibration_output.txt)

I checked on the datarecorder log file for beam 1 and found that it was reporting a missing data fraction of 28.125% (this fraction corresponds to 9 pipelines, but I'm not sure that's meaningful). It was also changing it's center frequency regularly. I've attached the relevant info from the log file here as `log_output.txt`
[log_output.txt](https://github.com/ovro-lwa/lwa-issues/files/11142772/log_output.txt)

We'd run into a similar problem of flipping frequencies when the config file had been changed, but I don't think that's the case here. Did I use the wrong config file for the SNAP2s?

yupinghuang, 2023-04-03:
I think what you're seeing is 1. bad tuning, and 2. no interleaving spws. It's a config issue.

The firmware appears correct. Note that `snap2_f_200msps_64i_4096c.fpg` under the `control_sw/config` directory is always symlinked to the latest firmware.

Also -- I'd really really recommend reprogramming & initializing the FPGAs from the calim nodes. The caltech-lwa repo on cr is really only meant for development. @marinanderson 

Also -- the config file should be the one in the `mnc_python` repo (i.e. the one that we've been using), instead of the one under caltech-lwa. https://github.com/realtimeradio/caltech-lwa/blob/master/control_sw/config/lwa_corr_config.yaml this looks wrong (see the channels for GPU nodes).

yupinghuang, 2023-04-03:
oh wait a caveat there is....`cold_start_from_config` takes a path to the config file on the `cr` node....so...............it's more chaos.

So I'd probably try this (and Jack recommends( from calim00 under `caltech-lwa/control-sw/scripts/`)

```
./lwa_observe_etcd.py -s -c ../../../mnc_python/config/lwa_config_calim_standard.yaml
```
which reprograms and reinitializes everything (if the fpga-etcd service can talk to the snap boards).

If that doesn't work (say Kathryn just loaded her custom fw on snap2), then you'd do a `program` call with the fpg file from `calim` and then call the above command and then
```
./lwa_observe_etcd.py -s -c ../../../mnc_python/config/lwa_config_calim.yaml --skipprog -i 2
```
to get this to work.

Too things I don't like about this:
1. This should be part of the mnc client...
2. it's kinda hard to figure out whether the etcd calls actually succeeded and where it went wrong without going into the `cr` node and looking at the log files.

But, if you're running into weird tuning issues. It's very likely the config file (i.e. reintiializing with another config file should fix it).

marinanderson, 2023-04-04:
Tried this out -- specifically the `./lwa_observe_etc.py` command above to reinitialize, but did not run the `program` call -- and 

`con.configure_xengine(recorders=['dr1'], full = True, calibratebeams = True) `

now works (with the exception of an `host lxdlwagpu07 (pipeline 0) failed to respond to etcd command!` error).

Before we close this issue out, can we update all of this information on reprogramming / reinitializing SNAPs to <strike>the</strike> a new(?) System Use Recipe document? I'm happy to do this and then have someone check my work. @Yuping, did you already start such a document?

yupinghuang, 2023-04-04:
Still very much a WIP but at least the snap stuff is there

https://caltech.sharepoint.com/sites/ovro/projects/SitePages/System-use-recipe-20220403.aspx#resetting-the-f-engine

I'll add an entry to the troubleshooting guide re failed to respond to ectd command https://github.com/ovro-lwa/lwa-issues/wiki/Troubleshooting-Guide-(what-if...)

yupinghuang, 2023-04-04:
actually I dunno what to do there -- do you have a stacktrace for the failed to respond to command error?

caseyjlaw, 2023-04-04:
I'll work on the `mnc-python` issues raised here. I have some provisional changes developed, but need to test more.

caseyjlaw, 2023-04-11:
I saw an odd pattern for slow MS files on lwacalim01. The recording was ongoing with two subbands that were widely spaced, as expected for interleaved subbands (13 and 50 MHz on lwacalim01). However, the time stamps in the files were about 8.5 hours off between the two subbands.

I restarted the recorder with the "standard" (linked to "interleaved") x-engine configuration. After that, it went back to the old setup, which had two low, adjacent subbands. The time stamps were similar, so that was fixed.

This is not directly related to the original post, but it may connect to config management. Does anyone know how this pattern of MS files could occur?

```
drwxr-xr-x. 15 pipeline pipeline 20 Apr 11 00:08 slow/20230411_080056_13MHz.ms
drwxr-xr-x. 15 pipeline pipeline 20 Apr 11 00:09 slow/20230411_163011_50MHz.ms
drwxr-xr-x. 15 pipeline pipeline 20 Apr 11 00:09 slow/20230411_080106_13MHz.ms
drwxr-xr-x. 15 pipeline pipeline 20 Apr 11 00:09 slow/20230411_163021_50MHz.ms
<restarted data recorder services; started data recording...>
drwxr-xr-x. 15 pipeline pipeline 20 Apr 11 00:20 slow/20230411_001932_20MHz.ms
drwxr-xr-x. 15 pipeline pipeline 20 Apr 11 00:20 slow/20230411_001932_18MHz.ms
drwxr-xr-x. 15 pipeline pipeline 20 Apr 11 00:20 slow/20230411_001942_20MHz.ms
drwxr-xr-x. 15 pipeline pipeline 20 Apr 11 00:20 slow/20230411_001942_18MHz.ms
drwxr-xr-x.  2 pipeline pipeline  6 Apr 11 00:20 slow/20230411_001952_20MHz.ms

```

yupinghuang, 2023-04-11:
It's almost certain that at least one of the SNAP boards was initialized with the non-interleaved config (the F engine decides which x engine to send what channels; I don't want to find out what happens if one snap board is sending the x engine different channels than the other snap boards @jack-h ). I think what the SNAP boards are programmed with is what decides what filenames you get.

I can see this on the other nodes too, where before restarting the data recorder the upper band's filename timestamps are a few hours in the future. My theory is that the x engine started a new sequence due to SNAP changes but the data recorder didn't catch that? Bit of a freak condition...

caseyjlaw, 2023-04-11:
One small change that may help is that the "standard" file is now the default. 
Using etcd for config is tracked in #223. I'll add draft up some requirements in that issue.

caseyjlaw, 2023-06-01:
I think the initial issue is resolved and other issues are being tracked. I'll close.

closed 2023-06-01 by caseyjlaw