ovro-lwa/lwa-issues #324: Migrating off the cosmic ray node onto lwacalim10

marinanderson, 2023-04-04:
We agreed in the LWA meeting today to migrate everything and everyone that isn't named Kathryn Plant off of the cosmic ray node.

All services, cron jobs, dashboards, etc. will now be run from **`lwacalim10`**.

To complete this transition, the following needs to be completed:

- [ ] what user should these services run as?
- [ ] create Jack an account on calim
- [ ] remove all ssh keys from cosmic ray node _except_ Kathryn's, Jack's, and Rick's (is this the right list of people?)
- [ ] verify cron jobs that were previously on `lxdlwagpu09` run as expected on lwacalim10
- [ ] what else am I missing?

marinanderson, 2023-04-04:
- [x] what user should these services run as?
**Answer:** `pipeline` user

jack-h, 2023-04-05:
This should be up and running, and I can talk to the new etcd services the same as before. Whatever rfi scripts run on gpu09 should keep working, so will be a good test to see if they do!

caseyjlaw, 2023-04-05:
@greghell Can you confirm that the RFI scripts still work? We are also recommending that they be moved to lwacalim10, but for now just want to know that the snap etcd change isn't breaking your code.

caseyjlaw, 2023-04-07:
So, I have the dashboard moved over to lwacalim10, too.
However, it seems that I cannot get it to run with systemd by setting `User=pipeline`. I suppose others are logging in as pipeline and setting it up, but not actually using that setting.

yupinghuang, 2023-04-11:
> `pipeline`

If it's under `/home/pipeline/.config/systemd` then it has to run as `pipeline` no matter what one specifies as `User` because it's not in the root user's systemd config and no one can run as other users other than root.

So, one has to be `pipeline`  to manage services running under the user `pipeline`. Long-term I think this is not a bad decision.

There's the other issue you reported where if one `sudo su` to `pipeline` then they run into `Failed to connect to bus: No such file or directory`. Adding

```
export XDG_RUNTIME_DIR=/run/user/`id -u`
```

to pipeline's `.bashrc`  solves it. ref https://askubuntu.com/questions/813588/systemctl-failed-to-connect-to-bus-docker-ubuntu16-04-container