ovro-lwa/lwa-issues #326: SDF writing script

iveydavis, 2023-04-05:
Script for writing beamformed SDF file given list of positions, start times, and durations.

iveydavis, 2023-04-05:
An example sdf file that I ran (semi-successfully) today:
```
PI_ID            81
PI_NAME          Ivey

PROJECT_ID       LD012
SESSION_ID       001

OBS_ID          1
OBS_START_MJD   60039
OBS_START_MPM   68400000
OBS_START       UTC 2023 04 05 19:00:00.000
OBS_DUR         10000
OBS_DUR+        00:00:10.000
OBS_MODE        TRK_RADEC
OBS_RA          0.000000000
OBS_DEC         +0.000000000
OBS_FREQ1       1161394218
OBS_FREQ1+      53.000000009 MHz
OBS_FREQ2       1599656187
OBS_FREQ2+      73.000000010 MHz
OBS_BW          7
OBS_BW+         19.600 MHz
OBS_DRX_BEAM    2

OBS_ID          2
OBS_START_MJD   60039
OBS_START_MPM   68430000
OBS_START       UTC 2023 04 05 19:00:30.000
OBS_DUR         10000
OBS_DUR+        00:00:10.000
OBS_MODE        TRK_RADEC
OBS_RA          0.000000000
OBS_DEC         +0.000000000
OBS_FREQ1       1161394218
OBS_FREQ1+      53.000000009 MHz
OBS_FREQ2       1599656187
OBS_FREQ2+      73.000000010 MHz
OBS_BW          7
OBS_BW+         19.600 MHz
OBS_DRX_BEAM    2
```

jaycedowell, 2023-04-05:
There are also the `sessionGUI.py` and `shiftSDF.py` scripts at https://github.com/lwa-project/session_schedules

iveydavis, 2023-06-05:
There was some discussion about whether the sdf should also be used for e.g. starting slow visibilities. Would it be reasonable to introduce something like a SESSION_MODE tag?

jaycedowell, 2023-06-06:
We had looked into something like that when we were updating our M&C software for LWA-SV.  Our approach was to expand allowed values of `OBS_MODE` by adding `COR` to control the correlator output.  We ended up not going this route because the correlator runs all the time and its output is imaged in realtime.

caseyjlaw, 2023-06-06:
Would you suggest implementing a feature in that field or that we develop a parallel system for digital control. It would be nice to do the former and benefit from all the experience.

iveydavis, 2023-07-17:
I've been spending more time on a script that _interprets_ SDFs than one that actually _writes_ SDFs.

I've been working on `sdf_parser.py` ( in `lwacalimXX:/opt/devel/idavis/scripts/` right now) that translates an SDF into a nested dictionary that contains 1. ) a dictionary with information on the session (_e.g._ the observing mode, which beam to use, the configuration file) and 2.) a nested dictionary that holds the information for each observation (each observation of a session is its own dictionary, with _e.g._ RA, DEC, object name, observation duration defined).

From this dictionary, session and observation objects are made. Given these objects, it determines the commands that should be run (and the order they should be done), as well as approximate times they should be run. A dataframe with this information is what is returned.

In my head, the parser should only be used for translating the SDFs into a list of commands; it shouldn't execute the commands. It should be the scheduler's job to consolidate all commands from all relevant SDFs and figure out the proper order to execute things (and do the execution).

As an example:

```
In [1]: import sdf_parser as sp

In [2]: fn = 'test.sdf'

In [3]: d = sp.sdf_to_dict(fn)

In [4]: d['SESSION']
Out[4]:
{'_ID': '81',
 'PI_NAME': 'Ivey',
 'PROJECT_ID': 'LD012',
 'SESSION_ID': '001',
 'SESSION_SPC': ['1024', '768'],
 'SESSION_MODE': 'POWER',
 'SESSION_DRX_BEAM': '1',
 'CONFIG_FILE': '/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_calim_config.yaml',
 'CAL_DIR': '/home/opt/devel/dgary/'}
...
In [7]: d['OBSERVATIONS']['OBSERVATION_1']
Out[7]:
{'OBS_ID': '1',
 'OBS_START_MJD': '60143',
 'OBS_START_MPM': '3600000',
 'OBS_START': ['UTC', '2023', '07', '18', '01:00:00.000'],
 'OBS_DUR': '10000',
 'OBS_INT_TIME': '56',
 'OBS_DUR+': '00:00:10.000',
 'OBS_MODE': 'TRK_RADEC',
 'OBS_RA': '0.000000000',
 'OBS_DEC': '+0.000000000',
 'OBS_FREQ1': '1161394218',
 'OBS_FREQ1+': ['53.000000009', 'MHz'],
 'OBS_FREQ2': '1599656187',
 'OBS_FREQ2+': ['73.000000010', 'MHz'],
 'OBS_BW': '7',
 'OBS_BW+': ['19.600', 'MHz']}
In [8]: df = sp.main('test.sdf')
/opt/devel/idavis/scripts/sdf_parser.py:155: UserWarning: Instructions to calibrate not specified. Assuming the beam should be calibrated
  warnings.warn('Instructions to calibrate not specified. Assuming the beam should be calibrated')

In [9]: from astropy.time import Time

In [10]: for i in range(len(df)):
    ...:     r = df.iloc[i]
    ...:     tv = r.name
    ...:     t = Time(tv, format = 'mjd').isot
    ...:     print(f"{t}: {r['command']}")
    ...:
2023-07-18T00:49:05.000: con = control.Controller('/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_calim_config.yaml')
2023-07-18T00:49:25.000: con.configure_xengine(['dr'+str(1)], calibrate_beams = True)
2023-07-18T00:59:55.000: con.control_bf(num = 1, ra = 0.0, dec = 0.0, track=True)
2023-07-28T01:00:05.000: con.start_dr(recorders=['dr'+str(1)], duration = 10000, time_avg=56,t0 = 60143.041666666664)
2023-07-18T01:00:25.000: con.control_bf(num = 1, ra = 0.0, dec = 0.0, track=True)
2023-07-28T01:00:35.000: con.start_dr(recorders=['dr'+str(1)], duration = 10000, time_avg=56,t0 = 60143.04201388889)
```

iveydavis, 2023-07-21:
A preliminary sdf writing script now lives at `/opt/devel/idavis/scripts/` as `sdf_writer.py`

To use, from the command line (in deployment or development environment) run:
`python sdf_writer.py out_name.sdf` where `out_name.sdf` is the name of the sdf file you want to write out to. Just follow the prompts that come up after running this, and you should be able to make an sdf easy peasy!

closed 2023-07-23 by iveydavis

caseyjlaw, 2023-07-24:
How shall we manage this software? I see two options:
- add it to `mnc-python` or
- create new repo related to observing and scheduling.

If we choose the latter, the repo could grow to include code for executing the observations.

rh-codebase, 2023-07-24:
It's Idiomatic python to have multiple packages in the same repo. Currently, mnc-python has 1 package named 'mnc'. Others could be added w/out ruffling any python feathers or breaking any rules that I'm aware of.  Kind of depends on how tightly related the new package is(will be) with the mnc package.