ovro-lwa/lwa-issues #329: Quota Script and VS-Code issue

Hallflower20, 2023-04-07:
The check_quota.py program that was in the shared bashrc seems to stop vscode from properly logging in at all. It causes this error:
```
[14:51:20.505] Running script with connection command: "C:\Windows\System32\OpenSSH\ssh.exe" -T -D 49686 lwacalim09 bash
[14:51:20.506] Terminal shell path: C:\Windows\System32\cmd.exe
[14:51:20.761] > �]0;C:\Windows\System32\cmd.exe�
[14:51:20.761] Got some output, clearing connection timeout
[14:51:20.783] > 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
[14:51:23.583] > Home directory quota: 72.8% out of 15.0 GB used.
[14:51:25.319] "install" terminal command done
[14:51:25.319] Install terminal quit with output: Home directory quota: 72.8% out of 15.0 GB used.
[14:51:25.319] Received install output: Home directory quota: 72.8% out of 15.0 GB used.
[14:51:25.319] Failed to parse remote port from server output
[14:51:25.320] Resolver error: Error: 
	at m.Create (c:\Users\20xha\.vscode\extensions\ms-vscode-remote.remote-ssh-0.98.0\out\extension.js:1:584388)
	at t.handleInstallOutput (c:\Users\20xha\.vscode\extensions\ms-vscode-remote.remote-ssh-0.98.0\out\extension.js:1:582948)
	at t.tryInstall (c:\Users\20xha\.vscode\extensions\ms-vscode-remote.remote-ssh-0.98.0\out\extension.js:1:681793)
	at async c:\Users\20xha\.vscode\extensions\ms-vscode-remote.remote-ssh-0.98.0\out\extension.js:1:644041
	at async t.withShowDetailsEvent (c:\Users\20xha\.vscode\extensions\ms-vscode-remote.remote-ssh-0.98.0\out\extension.js:1:647359)
	at async t.resolve (c:\Users\20xha\.vscode\extensions\ms-vscode-remote.remote-ssh-0.98.0\out\extension.js:1:645091)
	at async c:\Users\20xha\.vscode\extensions\ms-vscode-remote.remote-ssh-0.98.0\out\extension.js:1:720839
[14:51:25.323] ------
```

I've posted on vscode about the bug so maybe they can offer a fix but maybe there is a different way to remind users of their quota usage?

[VS-Code Github](https://github.com/microsoft/vscode-remote-release/issues/8249)

yupinghuang, 2023-04-11:
I moved the command from `.bashrc` to `.bash_profile` so that the it doesn't get executed when vscode does `ssh -T` (no ptty) and then run `bash` since running a subshell does not source `.bash_profile`. @Hallflower20 try this out? It worked for me (not that I tried reproducing your issue but my vscode worked, still). Mark this issue closed if it's working.

Hallflower20, 2023-04-11:
@yupinghuang I tried logging in again and now it just hangs until it eventually times out. It gets stuck on the storage thing. 
```
[13:11:18.824] Log Level: 2
[13:11:18.840] SSH Resolver called for "ssh-remote+lwacalim09", attempt 1
[13:11:18.841] "remote.SSH.useLocalServer": true
[13:11:18.843] "remote.SSH.path": undefined
[13:11:18.843] "remote.SSH.configFile": undefined
[13:11:18.844] "remote.SSH.useFlock": true
[13:11:18.845] "remote.SSH.lockfilesInTmp": false
[13:11:18.845] "remote.SSH.localServerDownload": auto
[13:11:18.846] "remote.SSH.remoteServerListenOnSocket": false
[13:11:18.846] "remote.SSH.showLoginTerminal": false
[13:11:18.847] "remote.SSH.defaultExtensions": []
[13:11:18.847] "remote.SSH.loglevel": 2
[13:11:18.847] "remote.SSH.enableDynamicForwarding": true
[13:11:18.848] "remote.SSH.enableRemoteCommand": false
[13:11:18.848] "remote.SSH.serverPickPortsFromRange": {}
[13:11:18.849] "remote.SSH.serverInstallPath": {}
[13:11:18.866] VS Code version: 1.77.1
[13:11:18.866] Remote-SSH version: remote-ssh@0.100.0
[13:11:18.866] linux x64
[13:11:18.872] SSH Resolver called for host: lwacalim09
[13:11:18.872] Setting up SSH remote "lwacalim09"
[13:11:18.882] Acquiring local install lock: /tmp/vscode-remote-ssh-fb1c0e14-install.lock
[13:11:18.886] Looking for existing server data file at /home/xhall/.config/Code/User/globalStorage/ms-vscode-remote.remote-ssh/vscode-ssh-host-fb1c0e14-b7886d7461186a5eac768481578c1d7ca80e2d21-0.100.0/data.json
[13:11:18.888] Using commit id "b7886d7461186a5eac768481578c1d7ca80e2d21" and quality "stable" for server
[13:11:18.893] Install and start server if needed
[13:11:18.902] PATH: /home/xhall/miniconda3/bin:/home/xhall/miniconda3/condabin:/home/xhall/.cargo/bin:/home/xhall/.julia/juliaup/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
[13:11:18.903] Checking ssh with "ssh -V"
[13:11:18.922] > OpenSSH_8.9p1 Ubuntu-3ubuntu0.1, OpenSSL 3.0.2 15 Mar 2022

[13:11:18.929] askpass server listening on /run/user/1000/vscode-ssh-askpass-9aaaba1110b9aaf8627ab170f39d0cf88f4e9f6a.sock
[13:11:18.929] Spawning local server with {"serverId":1,"ipcHandlePath":"/run/user/1000/vscode-ssh-askpass-36a25afe8822a5dc18ca30ad005b82ff89df1a84.sock","sshCommand":"ssh","sshArgs":["-v","-T","-D","43807","-o","ConnectTimeout=120","lwacalim09"],"serverDataFolderName":".vscode-server","dataFilePath":"/home/xhall/.config/Code/User/globalStorage/ms-vscode-remote.remote-ssh/vscode-ssh-host-fb1c0e14-b7886d7461186a5eac768481578c1d7ca80e2d21-0.100.0/data.json"}
[13:11:18.930] Local server env: {"SSH_AUTH_SOCK":"/run/user/1000/keyring/ssh","SHELL":"/bin/bash","DISPLAY":":0","ELECTRON_RUN_AS_NODE":"1","SSH_ASKPASS":"/home/xhall/.vscode/extensions/ms-vscode-remote.remote-ssh-0.100.0/out/local-server/askpass.sh","VSCODE_SSH_ASKPASS_NODE":"/usr/share/code/code","VSCODE_SSH_ASKPASS_EXTRA_ARGS":"--ms-enable-electron-run-as-node","VSCODE_SSH_ASKPASS_MAIN":"/home/xhall/.vscode/extensions/ms-vscode-remote.remote-ssh-0.100.0/out/askpass-main.js","VSCODE_SSH_ASKPASS_HANDLE":"/run/user/1000/vscode-ssh-askpass-9aaaba1110b9aaf8627ab170f39d0cf88f4e9f6a.sock"}
[13:11:18.939] Spawned 6214
[13:11:19.128] > local-server-1> Running ssh connection command: "-v -T -D 43807 -o ConnectTimeout=120 lwacalim09"
[13:11:19.136] > local-server-1> Spawned ssh, pid=6223
[13:11:19.149] stderr> OpenSSH_8.9p1 Ubuntu-3ubuntu0.1, OpenSSL 3.0.2 15 Mar 2022
[13:11:21.284] stderr> debug1: Server host key: ssh-ed25519 SHA256:jiax//wNBObEFp51QCWIihh7KXVoLMmLPVd/t6x+N04
[13:11:21.392] stderr> No Kerberos credentials available (default cache: FILE:/tmp/krb5cc_1000)
[13:11:21.392] stderr> 
[13:11:21.392] stderr> 
[13:11:21.398] stderr> No Kerberos credentials available (default cache: FILE:/tmp/krb5cc_1000)
[13:11:21.399] stderr> 
[13:11:21.399] stderr> 
[13:11:21.493] stderr> Authenticated to lwacalim09 (via proxy) using "publickey".
[13:11:21.747] > Activate the web console with: systemctl enable --now cockpit.socket
> 
> Register this system with Red Hat Insights: insights-client --register
> Create an account or view all your systems at https://red.ht/insights-dashboard
[13:11:22.424] > Home directory quota: 73.9% out of 15.0 GB used.
```

closed 2023-04-19 by yupinghuang