ovro-lwa/lwa-issues #341: Comparison of DW and NM beam models

marinanderson, 2023-04-11:


nmahesh1412, 2023-04-12:
I have placed the LWA antenna beam at /data05/nmahesh: 
LWA_x_10to100.ffe
LWA_y_10to100.ffe

It is simulated at every 1 MHz between 10 and 100 MHz and at every 1 deg theta and phi. These files dont have the soil in model but they have the exact dimensions of the antenna based on the blue prints in Sharepoint and the ground plane is 3m x 3m.

nmahesh1412, 2023-04-12:
The beam Jones matrix can be constructed as:
FIle1_Col7  File1_Col8
FIle2_Col7  File2_Col8

nmahesh1412, 2023-04-12:
A python def to read in that .ffe files into a numpy array
`def read_feko(file_name_prefix,freq_p,theta_p,phi_p):
    """
    Reads in a typical feko output file and creates a beam object.
    """
    beam_square = np.zeros((freq_p,theta_p,phi_p))
    etheta_square90 = np.zeros((freq_p,theta_p,phi_p),dtype = 'complex')
    ephi_square90 = np.zeros((freq_p,theta_p,phi_p),dtype = 'complex')
   
    f1 = open(file_name_prefix)
   

    z = theta_p*phi_p +10# ---> change this to no.of theta * no.of phi + No.of header lines
    c=0
    for line in f1:
        if c%z ==0:
            co=0
        if c % z >= 10: 
            x = list(map(float,line.split()))
            beam_square[int(c/z), co%181,int(co/181)] = 10**(x[8]/10)##*** beam_square [freq,theta,phi]***
            etheta_square90[int(c/z), co%181,int(co/181)] = x[2] + 1j*(x[3])
            ephi_square90[int(c/z), co%181,int(co/181)] = x[4] + 1j*x[5]
            co = co+1
        c = c+1

    return beam_square, etheta_square90, ephi_square90
`

jaycedowell, 2023-04-14:
I'd really like to see these simulations make it into LSL.  Are there any objections to me including these with LSL?  I'd also be happy with a PR that includes the files/reader.

nmahesh1412, 2023-04-14:
What is LSL?

jaycedowell, 2023-04-14:
The LWA Software Library - https://github.com/lwa-project/lsl

jaycedowell, 2023-04-26:
From the NM side here are what I think are the most relevant references on the beam models we use:
 * [Hicks et al. (2012)](https://arxiv.org/abs/1210.0506) - Overall antenna design and simulations of the 2D dipole response and impedance mismatch losses
 * [Ellingson (2011)](https://ieeexplore.ieee.org/document/5722983) - Simulations of mutual coupling for a LWA array
 * LWA Memos [#175](https://leo.phys.unm.edu/~lwa/memos/memo/lwa0175.pdf) and [#178](https://leo.phys.unm.edu/~lwa/memos/memo/lwa0178a.pdf) - Parameritization of our NEC4 models of the dipole response that are used in LSL
 * [Reeve (2014)](https://www.reeve.com/Documents/Long%20Wavelength%20Array/Reeve_LWA-Model.pdf) - Models the response over a variety of ground screens and soil conductivities
 * [Dowell et al. (2017)](https://arxiv.org/abs/1705.05819) - Derives an empirical correction as a function of elevation to our NEC4 models using observations of bright pulsars
 * [DiLullo et al. (2023)](https://arxiv.org/abs/2303.14904) - Makes measurements of the antenna impedance, dipole-dipole coupling, and mutual coupling at LWA1, LWA-SV, and LWA-NA.

A few notes on these:
 * The parameritized models and the empirical correction are what are used in LSL when modeling the 2D response of the dipole.  These are the basis for what the LSL script `plotAntenna.py` generates.
 * LSL also contains the impedance mismatch simulations of Hicks et al. (2012) in the `data` directory.
 * The KiCAD files for the test fixtures used by DiLullo et al. (2023) will be posted to that github repo. "soon".

nmahesh1412, 2023-06-06:
Comparing David Woody's beam files with recently generated FEKO and CST files :
[image]

rlbyrne, 2023-06-19:
Here's a function that converts the `.ffe` files into pyuvdata-readable `.beamfits` files:
```
import numpy as np
import pyuvdata

def parse_ffe_files():

    beam_files = [
        "/data05/nmahesh/LWA_x_10to100.ffe",
        "/data05/nmahesh/LWA_y_10to100.ffe",
    ]
    per_feed_beams = []

    for feed_ind, file in enumerate(beam_files):  # Loop over feeds

        with open(file, "r") as f:
            data = f.readlines()
        f.close()

        start_freq_chunk_lines = np.where(
            ["Configuration Name:" in line for line in data]
        )[0]

        for freq_chunk_ind in range(
            len(start_freq_chunk_lines)
        ):  # Loop over frequencies

            if freq_chunk_ind < len(start_freq_chunk_lines) - 1:
                chunk_lines = np.arange(
                    start_freq_chunk_lines[freq_chunk_ind],
                    start_freq_chunk_lines[freq_chunk_ind + 1],
                )
            else:
                chunk_lines = np.arange(
                    start_freq_chunk_lines[freq_chunk_ind], len(data)
                )

            freq_line = (
                [line_num for line_num in chunk_lines if "Frequency:" in data[line_num]]
            )[0]
            freq_hz = float((data[freq_line].split())[-1])

            # Parse header
            header_intro_line = (
                [
                    line_num
                    for line_num in chunk_lines
                    if "No. of Header Lines:" in data[line_num]
                ]
            )[0]
            header_len = int((data[header_intro_line].split())[-1])
            header_line = header_intro_line + header_len
            header = np.array(data[header_line].strip("#").split())
            theta_col = np.where(header == '"Theta"')[0][0]
            phi_col = np.where(header == '"Phi"')[0][0]
            etheta_real_col = np.where(header == '"Re(Etheta)"')[0][0]
            etheta_imag_col = np.where(header == '"Im(Etheta)"')[0][0]
            ephi_real_col = np.where(header == '"Re(Ephi)"')[0][0]
            ephi_imag_col = np.where(header == '"Im(Ephi)"')[0][0]

            data_chunk_split = np.array(
                [
                    data_line.split()
                    for data_line in data[header_line + 1 : np.max(chunk_lines)]
                    if len(data_line.split()) == len(header)
                ]
            ).astype(float)

            theta_array = data_chunk_split[:, theta_col]  # Zenith angle
            phi_array = data_chunk_split[:, phi_col]  # Azimuth
[... 98 lines cut ...]
        not np.min(per_feed_beams[0].freq_array == per_feed_beams[1].freq_array)
        or not np.min(per_feed_beams[0].axis1_array == per_feed_beams[1].axis1_array)
        or not np.min(per_feed_beams[0].axis2_array == per_feed_beams[1].axis2_array)
    ):
        print("ERROR: Mismatched axes. Cannot combine beam objects.")
        print("Saving beam to /data05/rbyrne/LWA_x_10to100.beamfits")
        per_feed_beams[0].write_beamfits(
            f"/data05/rbyrne/LWA_x_10to100.beamfits",
            clobber=True,
        )
        print("Saving beam to /data05/rbyrne/LWA_y_10to100.beamfits")
        per_feed_beams[1].write_beamfits(
            f"/data05/rbyrne/LWA_y_10to100.beamfits",
            clobber=True,
        )
    else:
        per_feed_beams[0].data_array[:, :, 1, :, :, :] = per_feed_beams[1].data_array[
            :, :, 1, :, :, :
        ]
        print("Saving beam to /data05/rbyrne/LWA_10to100.beamfits")
        per_feed_beams[0].write_beamfits(
            f"/data05/rbyrne/LWA_10to100.beamfits",
            clobber=True,
        )
```

yupinghuang, 2023-06-25:
@nmahesh1412 can you share the code that reads and normalizes the DW beam?

rlbyrne, 2023-06-25:
@yupinghuang Nivedita might have a different script but this is what I use to read the DW beam:
```
def read_beam_txt_file(path, header_line=6):

    with open(path, "r") as f:
        lines = f.readlines()
    f.close()

    header = lines[header_line]
    npoints = len(lines) - (header_line + 1)

    za_deg = np.zeros(npoints, dtype=float)
    az_deg = np.zeros(npoints, dtype=float)
    freq_mhz = np.zeros(npoints, dtype=float)
    jones_theta = np.zeros(npoints, dtype=complex)
    jones_phi = np.zeros(npoints, dtype=complex)

    for point in range(npoints):
        line_data = [float(value) for value in lines[point + 7].split()]
        za_deg[point] = line_data[0]
        az_deg[point] = line_data[1]
        freq_mhz[point] = line_data[2]
        jones_theta[point] = line_data[7] + 1j * line_data[8]
        jones_phi[point] = line_data[9] + 1j * line_data[10]

    za_axis = np.unique(za_deg)
    freq_axis = np.unique(freq_mhz)
    az_axis = np.unique(az_deg)
    # Fill in other quadrants
    az_axis = np.concatenate(
        (az_axis, az_axis + 90.0, az_axis + 180.0, az_axis + 270.0)
    )
    az_axis[np.where(az_axis < 0.0)] += 360.0
    az_axis[np.where(az_axis >= 360.0)] -= 360.0
    az_axis = np.unique(az_axis)

    jones = np.full(
        (2, 2, len(freq_axis), len(za_axis), len(az_axis)),
        np.nan + 1j * np.nan,
        dtype=complex,
    )

    # Flip the Jones matrix in particular quadrants
    multiply_factors = np.ones(
        (2, 2, 4), dtype=float
    )  # Shape (nfeeds, npols, nquadrants)
    multiply_factors[0, 0, 0:2] = -1.0
    multiply_factors[0, 1, 1:3] = -1.0
    multiply_factors[1, 0, 0] = -1
    multiply_factors[1, 0, 3] = -1
    multiply_factors[1, 1, 0:2] = -1

    for point in range(npoints):
        za_ind = np.where(za_axis == za_deg[point])
        freq_ind = np.where(freq_axis == freq_mhz[point])
        for pol in [0, 1]:
            use_az = az_deg[point]
            if pol == 0:
                use_az = 90.0 - use_az  # Q pol is rotated 90 degrees

            az_ind_quad_1 = np.where(az_axis == use_az)
            jones[0, pol, freq_ind, za_ind, az_ind_quad_1] = (
                multiply_factors[0, pol, 0] * jones_theta[point]
            )
            jones[1, pol, freq_ind, za_ind, az_ind_quad_1] = (
                multiply_factors[1, pol, 0] * jones_phi[point]
            )

            az_ind_quad_2 = np.where(az_axis == 180.0 - use_az)
            jones[0, pol, freq_ind, za_ind, az_ind_quad_2] = (
                multiply_factors[0, pol, 1] * jones_theta[point]
            )
            jones[1, pol, freq_ind, za_ind, az_ind_quad_2] = (
                multiply_factors[1, pol, 1] * jones_phi[point]
            )

            az_ind_quad_3 = np.where(az_axis == 180.0 + use_az)
            jones[0, pol, freq_ind, za_ind, az_ind_quad_3] = (
                multiply_factors[0, pol, 2] * jones_theta[point]
            )
            jones[1, pol, freq_ind, za_ind, az_ind_quad_3] = (
                multiply_factors[1, pol, 2] * jones_phi[point]
            )

            az_ind_quad_4 = np.where(az_axis == 360.0 - use_az)
            jones[0, pol, freq_ind, za_ind, az_ind_quad_4] = (
                multiply_factors[0, pol, 3] * jones_theta[point]
            )
            jones[1, pol, freq_ind, za_ind, az_ind_quad_4] = (
                multiply_factors[1, pol, 3] * jones_phi[point]
            )
    jones = np.flip(jones, axis=1)

    # Polarization mode at zenith is undefined
    # Insert values that will make the basis conversion and normalization work
    # This discards any imaginary component
    zenith_points = np.where(za_axis == 0)[0]
    jones[1, 0, :, zenith_points, :] = np.sqrt(
        jones[0, 0, :, zenith_points, :] ** 2.0
        + jones[1, 0, :, zenith_points, :] ** 2.0
    )
    jones[0, 1, :, zenith_points, :] = -np.sqrt(
        jones[0, 1, :, zenith_points, :] ** 2.0
        + jones[1, 1, :, zenith_points, :] ** 2.0
    )
    jones[0, 0, :, zenith_points, :] = 0.0
    jones[1, 1, :, zenith_points, :] = 0.0
```
To normalize I turn it into a pyuvdata beam object and use the `peak_normalize` function:
```
    beam_obj = pyuvdata.UVBeam()
    beam_obj.Naxes_vec = 2
    beam_obj.Nfreqs = len(freq_axis)
    beam_obj.Nspws = 1
    beam_obj.antenna_type = "simple"
    beam_obj.bandpass_array = np.full((1, len(freq_axis)), 1.0)
    beam_obj.beam_type = "efield"
    beam_obj.data_array = np.copy(jones[:, np.newaxis, :, :, :, :])
    beam_obj.data_normalization = "physical"
    beam_obj.feed_name = ""
    beam_obj.feed_version = ""
    beam_obj.freq_array = (
        np.copy(freq_axis[np.newaxis, :]) * 1e6
    )  # Convert from MHz to Hz
    beam_obj.history = ""
    beam_obj.model_name = ""
    beam_obj.model_version = ""
    beam_obj.pixel_coordinate_system = "az_za"
    beam_obj.spw_array = [0]
    beam_obj.telescope_name = "LWA"
    beam_obj.Naxes1 = len(az_axis)
    beam_obj.Naxes2 = len(za_axis)
    beam_obj.Ncomponents_vec = 2
    beam_obj.Nfeeds = 2
    beam_obj.Npols = 2
    beam_obj.axis1_array = np.radians(az_axis)
    beam_obj.axis2_array = np.radians(za_axis)
    beam_obj.basis_vector_array = np.repeat(
        (
            np.repeat(
                (np.identity(2, dtype=float))[:, :, np.newaxis], len(za_axis), axis=2
            )
        )[:, :, :, np.newaxis],
        len(az_axis),
        axis=3,
    )
    beam_obj.feed_array = ["E", "N"]
    beam_obj.x_orientation = "east"
    beam_obj.peak_normalize()
    beam_obj.check()
```
This is the beam file I'm reading, for reference:
[DW_beamquadranttable20151110.txt](https://github.com/ovro-lwa/lwa-issues/files/11861158/DW_beamquadranttable20151110.txt)

nmahesh1412, 2023-06-25:
Ah yes, I have a slightly different function for the DW beam:

DW_files = np.loadtxt('/data05/nmahesh/OVRO-LWA_Beam-models/DW_beamquadranttable20151110.txt',skiprows=7) 

theta_dw = np.arange(0,95,5)
phi_dw = np.arange(0,95,5)
freq_dw = np.arange(20,90,10)
DW_beam = np.zeros((7,len(theta_dw),len(phi_dw))) ## combines individual components to calculate the power beam
DW_ebeam = np.zeros((7,len(theta_dw),len(phi_dw),4)) ## save the individual e-field components 

c =0
for i in range(7):
    for p in range(len(phi_dw)):
        for t in range(len(theta_dw)):
            DW_beam[i,t,p] = np.sqrt(np.abs(DW_files[c,3]+1j*DW_files[c,4])**2+ np.abs(DW_files[c,5]+1j*DW_files[c,6])**2)
            DW_ebeam[i,t,p,:] = DW_files[c,-4:]
            c = c+1

I am working on just making a central script that has all of these functions and can easily be run to produce normalized beams and Jones matrix products.
Will include your functions too, Ruby. I can see soon, a few of us will use pyuvobject. So it will be nice to have a central place for it.