ovro-lwa/lwa-issues #344: Determine elevaiton limit for usable observations of the Sun

marinanderson, 2023-04-11:
- [ ] Correction for XX vs YY at low elevation in the solar imaging pipeline

dgary50, 2023-07-29:
The test of horizon to horizon solar beam tracking shown by the plot below indicates that earlier than 14:00 UT on 28 July 2023, or later than 02:00 UT on 29 July 2023, we do not get any signal from the Sun.  The elevation limit at these times is 11 degrees.  Realistically, a time limit of 14:15 UT to 01:45 UT is probably optimum, at which times the Sun's elevation was 14 degrees.

[image]
The plot shows beam intensity at three frequencies (on a very active day).  The track of minimum values for 51.7 and 75.6 MHz shows more-or-less the quiet Sun signal.  Those signals decrease at low elevations, possibly due to the primary beam and/or refraction of the beam off the Sun due to the ionosphere.

dgary50, 2023-12-14:
We have been observing the Sun daily for at least a month (Nov. 14-Dec. 14), using 10 degrees as the elevation limit.  At least during the winter months this is possibly too low due to ionospheric refraction causing problems at low elevation.  However, we may continue with this elevation limit since there is no clear downside.