ovro-lwa/lwa-issues #346: Compare slow vs. fast mode quality in various solar activity (esp. quiet Sun)

marinanderson, 2023-04-11:


dgary50, 2023-05-20:
### Summary of Observations on 2023-05-19

We successfully took stand-along fast-mode data 2023-05-19 17:36 - 17:38 UT with older set of 48 antennas.  Then we took simultaneous fast- and slow-mode data 2023-05-19 17:46 - 17:48 UT with older set of 48 antennas.  

After this success, we changed the antenna list to a better set of antennas (10 in the core, 38 outer) chosen randomly and then checked for online status.  The antenna list is LWA-015, LWA-018, LWA-038, LWA-067, LWA-085, LWA-096, LWA-131, LWA-151, LWA-180, LWA-191, LWA-260, LWA-263, LWA-264, LWA-272, LWA-274, LWA-275, LWA-277, LWA-278, LWA-281, LWA-284, LWA-285, LWA-285, LWA-286, LWA-287, LWA-289, LWA-290, LWA-295, LWA-301, LWA-304, LWA-305, LWA-307, LWA-313, LWA-314, LWA-320, LWA-321, LWA-326, LWA-327, LWA-328, LWA-329, LWA-330, LWA-334, LWA-335, LWA-336, LWA-337, LWA-338, LWA-339, LWA-340, LWA-347.

With this list we took simultaneous fast- slow- and beamformed data 2023-05-19 21:48 - 21:50 UT.  Initial checks show the data are good, and the beamformed spectrogram shows a type-III storm was in progress, shown below.

[image]

dgary50, 2023-08-30:
@binchensun @surajit4444mondal Can this issue be closed now?  Are there other tests that are needed first?

surajit4444mondal, 2023-09-01:
I did a quick check and it seems that the quiet sun images using fast visibilities has a higher noise than just a mere scaled up version of the slow visibilities. My feeling is that the extra noise is due to RFI, which gets suppressed some how with having more antennas. I would prefer to keep this issue open, and maybe take some data during night time with both slow and fast vis to check the noise characteristics. I believe that the RFI situation will be better at night.