ovro-lwa/lwa-issues #347: Compare solar data with/without ARX solar attenuation

marinanderson, 2023-04-11:


binchensun, 2023-04-19:
Is the "ARX solar attenuation" equivalent to the "daytime ARX setting"? It appears that we have two attenuators per polarization per antenna. Are the nighttime and daytime ARX settings have fixed attenuator levels for all pols/antennas?

dgary50, 2023-04-19:
My understanding is that the daytime settings are the leveling attenuations and are distinct from the solar attenuator.  This second attenuator is to be used when activity is high and data would otherwise be compromised.  But use of this attenuator does mean non-solar observations are compromised, so we haven't worked out how a decision will be made when to use them.

dgary50, 2023-04-20:
@marinanderson Is there a software hook, and documentation, about how to switch the solar attenuator in and out?

dgary50, 2023-04-20:
> Yes, these are essentially distinct things. However, the "solar attenuator" is not a distinct bit of hardware. It would be a new way to set the ARX attenuation. It would be a third set, distinct from "daytime" and "nighttime".
> Details:
> Larry has worked to optimize the ARX and digital gain with a goal of having noise be dominated by the sky brightness. The sky is assumed to be dominated by Galactic synchrotron, rather than any individual source. The filtering and gain are optimized to reduce the impact of RFI, which in power terms, is the biggest foreground. This process used data recorded in typical observing conditions at day and night.
> I can't find Larry's ARX memo at the moment, but the control system is documented at https://github.com/ovro-lwa/ARXtesting/blob/main/arxCommandDictionary.txt. There are two filters that can be set (high and low sides of the band).

Thanks Casey. If the attenuation is set for sky brightness to dominate, I would propose to just insert a fixed additional attenuation in each ARX channel (5 dB, or 10 dB). More later…DaleSent from my iPhone