ovro-lwa/lwa-issues #348: Measure delay and gain differences of solar attenuation setting

marinanderson, 2023-04-11:


binchensun, 2023-07-10:
### Setup
On 2023 July 7, I did the following practice:

- At 10:53 UT, ran ./scripts/restoreSettings.sh data/20230608-settingsAll-day.mat, which took ~1 min to complete.
- At 10:59 UT, ran ./scripts/restoreSettings.sh data/20230608-settingsAll-night.mat, which took ~1 min to complete.

According to @ldaddario's message on July 6, compared to the night-time setting, the day-time setting has 15 dB of more analytical attenuation, but an additional 10 dB of digital gain to compensate for the former, resulting in an expected net attenuation of 5 dB.

### Data used for analysis:

- [dataset (a)] Data with **daytime** attenuator settings: /data07/bin.chen/20230707_calgain/cal_day/20230707_105839_*.ms
- [dataset (b)] Data with **nighttime** attenuator settings: /data07/bin.chen/20230707_calgain/cal_night/20230707_110139_*.ms
- [dataset (c)] Data with **nighttime** attenuator settings ~5 min later (used as the "control"): /data07/bin.chen/20230707_calgain/cal_night/20230707_110640_*.ms

### Analysis procedure:
1. Derive bandpass solutions for the reference night-time setting data (dataset b)
- Put CasA and CygA models to the model column of the visibilities
- Solve for bandpass as usual
2. Derive differential bandpass solution for day-time setting data (dataset a)
- Put CasA and CygA models to the model column of the visibilities
- Solve for bandpass while applying the bandpass table from step 1 on the fly
3. Similar to step 2, but derive differential band-averaged complex gain solutions for day-time setting data (dataset a)
4. Similar to steps 2 and 3, but derive differential bandpass and band-averaged complex gain for the 5-min later night-time setting data (dataset c) for comparison.

### Results
#### Differential (band-averaged) complex gains from day-time setting dataset (a)

[image]

#### Differential (band-averaged) complex gains from the control night-time setting dataset (c)
[image]

### Summary and discussion:
- Gain solutions are quite stable (judging on the differential amps/phases) except for few antennas that have very regular spacings - “edge” of the correlator boards?
- Overall on-sky daytime gain (analytical + digital) is lower than the night-time gain by 2.5 dB (~2x higher than the expected 5 dB). Why? But they are quite consistent across all antennas except for these “edge” antennas, which is good.
- The majority of the antennas have very small differential phases (<+-2 deg), which means the image quality would be acceptable if one transfers the solution between the two settings. Self-calibration can likely do a good job in correcting for the rest.
- However, the few "edge" antennas and antenna indices 0~63 (first two correlator boards, perhaps?) have quite large differential amplitudes/phases, which may be the main cause of the reduction of the image dynamic range. What is the origin of those larger differential gains? Possible means to improve these?

binchensun, 2023-07-10:
Yes, these are antenna _indices_ going from 0 to 351, which I believe is the same as the "correlator number." Each antenna index maps to an antenna name "LWA###," which usually has a different number. CASA's convention calls the former "antenna indices" and the latter "antenna names."

ldaddario, 2023-07-10:
On 7/10/2023 07:54, Casey Law wrote:
> To help interpret, can you clarify what the ant index refers to? Is it 
> an index based on the shape of data when read by casa or casacore for 
> plotting? If so, that would be the 'correlator number', not antenna number.

Casey,

It was decided a few years ago that we will not use "antenna numbers" in 
the OVRO LWA, but rather antenna names.  An antenna name is a character 
string starting with 'LWA-'.  This is because, although the antenna name 
contains a numerical part, that number tells us very little.  The 
numbers are not in a describable sequence, and the largest is bigger 
than the number of physical antennas in the array because some are not used.

The antenna names are needed only when it's necessary to account for 
some physical property that varies among antennas.  (The only such 
property that we use now is the antenna's geographical location.  In the 
future, we may have individual beam patterns for all 704 antennas.) 
Most of the time, it's sufficient to use only the signal number within 
the processing system, ignoring the antenna names.  Whereas connectivity 
is regular from the digitizer inputs onward, digital signal numbers can 
be known by calculation; no table lookup is needed.  What you call the 
'correlator number' is assigned in the order of the SNAP2 boards and 
within each board in the order of the F engine's enumeration (0:63). 
This is almost the same as the digital signal number, which is
   dsig = (snapNo-1)*64 + fsigNo
There is no reason for this ever to change.  For a given dsig, the 
corresponding data will always be in the same place everywhere downstream.

At the moment, the two co-located antennas of opposite polarization are 
kept together in the signal processing.  Since 'correlator number' 
enumerates only antenna pairs, not signals, it is currently given by
   corrNo = int(dsig/2).
However, this could change in the future; there is no reason that both 
polarizations need to be kept together, and we might find reasons to 
separate them.  For this reason it's better to use signal numbers 
(0:703) rather than correlator numbers (0:351).

Since the correspondence between signal or correlator numbers and 
antenna names is irregular, a table lookup is needed when 
antenna-specific properties need to be taken into account.  I have a 
Python package that does this.  On a calim node:
$ cd /home/ldaddario
$ python
.>> import sigtab as s
Then:
.>> s.d2name(dsig)    # returns antenna name
.>> s.name2d(string)  # returns digital signal number (or None if string 
is not an antenna name that is used in the array)
Also, if it's necessary to know where in the hardware a signal was 
processed:
.>> s.d2feng(dsig)    # returns (snapNo,fsigNo)
.>> s.d2a(dsig)       # returns analog signal number asig
.>> s.a2arx(asig)     # returns (arxAddress,arxChannel)
.>> s.a2d(asig)       # returns digital signal number dsig

For the last three, the analog signal number (0:719) is in the order of 
ARX board addresses and channels.  The connectivity between the analog 
and digitial processing is irregular, so this also needs a table lookup. 
  While in principle this could change in the future, by policy it 
should remain fixed.  Changes in connectivity are intended to occur only 
upstream of the ARX boards (typically at the large bulkhead at the rear 
of the shielded section of the shelter).  There are 720 (not 704) values 
of asig because some ARX channels are not used.

Cheers,
Larry

-- 
=======================================================
Larry R. D'Addario
Department of Astronomy
California Institute of Technology
Pasadena, California 91125

291 Cahill, mail code 249-17
email:  ***@***.***
phone:  +1/626/395-1431 (office)
         +1/626/510-6492 (home)
         +1/626/657-6357 (mobile)
=======================================================

binchensun, 2023-07-10:
Each panel actually has both X and Y solutions plotted (red & blue symbols). So if I understand Larry's response correctly, the "odd" antennas seem to correspond to the "ends" of the (F engine of the) SNAP2 boards, each having 2 x 32 = 64 digital signal numbers?

caseyjlaw, 2023-07-11:
Hi Larry,
  Could I interest you in integrating your code with the software repositories used by the team? In particular, we have naming lookup in the lwa-antpos repo.
  That code helps users map inputs and see changes. It may be missing features that you've developed. I'd appreciate having a pull request if it helps the team.

    Casey