ovro-lwa/lwa-issues #350: `snap11` 40GbE not linking

jack-h, 2023-04-14:
Each SNAP board has two 40GbE connections. On `snap11` one of these should go to port 11 of the switch. However, the switch reports a transceiver installed, but no link detected.

Is it possible that either this cable is faulty, or the SNAP-side end isn't seated correctly?
I expect the link light on the switch side is currently not lit, and if the cable is at fault then trying a different cable (or swapping from a different SNAP) would cause the switch link lights to come on.

Any chance someone (@rh-codebase , @Princemongo ?) could check this out at a convenient moment, please?

rh-codebase, 2023-04-14:
Also note that there has been at least 2 SFP module failures in the past few months.

Princemongo, 2023-04-14:
There are 4 ports on the SNAP2s: 2 mounted close to the board and 2 
above those. Only the ports closest to the board are used in the current 
configuration. As the boards are mounted vertically in the chassis, one 
port is above the other, so I refer to them as "upper" and "lower". The 
upper port on snap11 goes to port #11 on the Cisco switch. The lower 
port goes to port #12 on the Cisco switch. There are no LEDs lit on any 
of the SNAP2s which would indicate the presence of an SFP module. There 
are however LEDs on the Cisco switch which indicate the presence and 
status of an SFP module:
1) no module installed - amber
2) module installed and connection is made - green
3) module installed but no connection - not lit

When I went to the shelter, port 11 was not lit, but port #12 was green.
I noticed that the SFP module going into the upper SNAP2 port was 
inserted upside down. I removed the module and inserted it in the 
correct orientation. The LED turned amber and then went out.
I tried swapping the SFP modules at the SNAP2, then both LEDs went out!
I tried using the other pair of unused ports on the SNAP2, but the LEDs 
remained out.
I put the SFP modules in the correct spots on the SNAP2s and tried 
another pair of ports on the Cisco switch, but the LEDs remained out.
I put everything back to the original state, with the SFPs going into 
the SNAP2 in the correct orientation. The LEDs remained out.
I didn't know what else to do at this point, so I cycled power on the 
SNAP2. The LEDs remained out.

All I can think of at this point is to swap in the spare SNAP2.

- Mark

jack-h, 2023-04-14:
Hi Mark,

It's possible that the link doesn't recover from unplugging and replacing
without a software intervention. In all probability the incorrectly
oriented transceiver was the issue and hopefully at the next software
initialization everything will fall into place. I'm not by a computer right
now but will see if I can bring this up tomorrow.
Thanks for checking it!

Princemongo, 2023-04-14:
The LEDs on the Cisco switch do require "software intervention". Kathryn re-initialized the SNAP2 and now both ports have green LEDs!

closed 2023-04-14 by Princemongo