ovro-lwa/lwa-issues #351: Fast baseline selection loading leads to no fast correlator output.

jack-h, 2023-04-14:
cc @caseyjlaw 

See https://github.com/ovro-lwa/mnc_python/pull/20

When loading the default solar-team-approved visibility set, the fast output pipeline seems to crash and nothing is output.

dgary50, 2023-04-20:
@jack-h Is there any resolution to this issue yet?  We have some tasks involving this mode and we'd like to spend some time on those next month during a visit to Caltech (May 18-19).

jack-h, 2023-04-20:
Looking at this now! Will report back soon

jack-h, 2023-04-20:
This should now be fixed!

This results in fast vis output from gpu01:

```
# from mnc_python import control
In [48]: con = control.Controller(xhosts=['lxdlwagpu01'], npipeline=1)
2023-04-20 16:29:09,672 - mnc.control - INFO - Loaded configuration for 1 x-engine host(s) running 1 pipeline(s) each
2023-04-20 16:29:09,672 - mnc.control - INFO - Supported recorder modes are: slow,fast,beam1,beam2,beam3,beam4
2023-04-20 16:29:09,673 - mnc.control - INFO - etcd server being used is: etcdv3service

In [49]: con.configure_xengine(recorders=['drvs', 'drvf'])
2023-04-20 16:29:22,687 - mnc.control - INFO - Stopping/starting pipelines
2023-04-20 16:29:40,741 - mnc.control.Lwa352CorrelatorControl - INFO - Pipelines all appear to be ready after 8.0 seconds
2023-04-20 16:29:40,742 - mnc.control - INFO - pipelines up? True
2023-04-20 16:29:40,742 - mnc.control - INFO - Configuring x-engine for slow visibilities
2023-04-20 16:29:40,743 - mnc.control.Lwa352CorrelatorControl - INFO - Setting max data output rate per pipeline to 20000.0 Mbit/s
2023-04-20 16:29:40,743 - mnc.control.Lwa352CorrelatorControl - INFO - Setting pipeline lxdlwagpu01:0 data destination to 10.41.0.76:10001
2023-04-20 16:29:40,809 - mnc.control.Lwa352CorrelatorControl - INFO - Arming correlator core
2023-04-20 16:29:41,817 - mnc.control.Lwa352CorrelatorControl - INFO - All pipelines in 'waiting' state as expected
2023-04-20 16:29:41,817 - mnc.control.Lwa352CorrelatorControl - INFO - Waiting 10 seconds for trigger
2023-04-20 16:29:51,829 - mnc.control.Lwa352CorrelatorControl - INFO - All pipelines in 'running' state as expected
2023-04-20 16:29:51,829 - mnc.control.Lwa352CorrelatorControl - INFO - Arming correlator accumulator core
2023-04-20 16:29:52,840 - mnc.control.Lwa352CorrelatorControl - INFO - All pipelines in 'waiting' state as expected
2023-04-20 16:29:52,840 - mnc.control.Lwa352CorrelatorControl - INFO - Waiting 10 seconds for trigger
2023-04-20 16:30:02,852 - mnc.control.Lwa352CorrelatorControl - INFO - All pipelines in 'running' state as expected
2023-04-20 16:30:02,852 - mnc.control - INFO - Configuring x-engine for fast visibilities
```

closed 2023-04-20 by caseyjlaw

jack-h, 2023-04-20:
Be aware, the way the control code is currently set up, I think you must
observe with slow to correctly configure the correlator even if you only
care about the fast output

caseyjlaw, 2023-05-01:
I've made a change to allow configuring x-eng and data recorders for 'drvf' without also including 'drvs'. This has been deployed in the `development` and `deployment` environments. It successfully writes data in either drvs/drvf modes.

However, I'm still seeing signs of falling behind in the data recorders, e.g.:
`Max. processing time at >125% of gulp (0.826 s = 823.3% )`

I also see this if I set the antenna selection to only four antennas. @jack-h is this a new issue or the same one? Our software deployment isn't very clear at the moment, so it is possible that I'm running some old code.

jack-h, 2023-05-02:
You will always get 48 antennas -- the `set_baseline_select` method asserts that the baseline selection array contains (2*48)^2 (or rather `48*49/2 * 4 [polarization]`) entries, so something upstream of this is padding your 4-antenna list. I.e., it's possible that the recorder doesn't keep up, but data rate doesn't depend on the number of antennas you are selecting.

caseyjlaw, 2023-05-02:
Good point. I'll open a new issue for the falling behind issue.

dgary50, 2023-05-02:
Casey and All, several of us will be visiting Caltech on May 18-19 and
would like to take fast visibility data during the visit in order to try
imaging in that mode.  So getting the mode working in the next couple of
weeks would be very timely.
--Dale

dgary50, 2023-05-06:
FYI, I was looking at one of the fast data files and here are the timestamps of the scans in the ms and the overall duration of a 1-s file:
Overall duration:
(EndTime - BeginTime)*86400 = 1.0031013283878565

Individual timestamps per scan:
2023-05-01 23:44:55.806
2023-05-01 23:44:55.907
2023-05-01 23:44:56.007
2023-05-01 23:44:56.107
2023-05-01 23:44:56.208
2023-05-01 23:44:56.308
2023-05-01 23:44:56.408
2023-05-01 23:44:56.509
2023-05-01 23:44:56.609
2023-05-01 23:44:56.709

So clearly the timestamps are losing time (I assume these should be exactly 100 ms apart).  Is these timestamps a function of the data recorder, or the correlator?

caseyjlaw, 2023-05-06:
I think it need not be exactly 100ms, but more likely an integer multiple of a fundamental sampling time.
That said, there may be indications of falling behind in the contents of the files. Could you try loading them and counting the zeros.

jack-h, 2023-05-09:
The integration time here is 2400 spectra, with each spectra consisting of
channels of width 196 MHz / 8192.
I.e. the integration time is `8192 / 196e6 * 2400 = 0.100310... seconds`.
So, I think what you are seeing is correct