ovro-lwa/lwa-issues #352: Two data recorders not working

rlbyrne, 2023-04-17:
For slow visibility capture over the last few days, two data recorders haven't been working, resulting in two missing subbands of data. I am following the System Use Recipe and doing data capture as follows:
```
from mnc import control
con = control.Controller()
con.stop_xengine()
con.configure_xengine(recorders=['drvs'])
con.start_dr(recorders=['drvs'])
```
Running `con.status_dr()` returns this:
```
['0 recording; 14 idle; 2 timed out',
 'WARNING: drvs not fully operational: 2 sub-bands have not updated in 120 s: drvs8301,drvs8302',
 '0 recording; 14 idle; 2 timed out',
 'WARNING: drvf not fully operational: 2 sub-bands have not updated in 120 s: drvf8301,drvf8302',
 'idle',
 'idle',
 'idle',
 'idle']
```

jaycedowell, 2023-04-18:
drvs8301 and drvs8302 are on lwacalim08.  The logs show that both those services were shut down on 4/12 and never restarted.

rlbyrne, 2023-04-18:
@jaycedowell Thank you! How did you reboot them?

jaycedowell, 2023-04-18:
I logged into lwacalim08 as pipeline and ran `~/ovro_data_recorder/services/start_dr_services.sh`.

rh-codebase, 2023-04-18:
what calim machines  are the DR being run on?

rlbyrne, 2023-04-18:
@rh-codebase calim00

rh-codebase, 2023-04-18:
did the shutdown of DR on calim08 correspond to a reboot? If so, then they did not come back up on boot.
(base) [ubuntu@lwacalim08 ~]$ uptime
 00:44:03 up 4 days,  9:16,  1 user,  load average: 0.17, 0.67, 0.89

rh-codebase, 2023-04-18:
@rlbyrne Jayce just said on calim08. Are you saying only on 08 and 00?

rlbyrne, 2023-04-18:
@rh-codebase I was controlling the data recorder from calim00, but the issue was on calim08

rh-codebase, 2023-04-18:
my question is which calim nodes run the start_dr_services.sh script?

closed 2023-04-26 by caseyjlaw