ovro-lwa/lwa-issues #353: restoreSettings.sh errors

kathrynplant, 2023-04-18:
I ran ./scripts/restoreSettings.sh data/20230309-settingsAll-day.mat

I got the following errors, although the script continued running. What do the errors mean? The script kept running and later gave an output of settings it was applying. How can I confirm whether the script finished correctly?

Received 2 of 1 expected responses
Command status from FID 2 returned: 'error'
Received 2 of 1 expected responses
Command status from FID 4 returned: 'error'
Received 2 of 1 expected responses
Received 2 of 1 expected responses
Received 2 of 1 expected responses

yupinghuang, 2023-04-19:
Yeah errors are not good.

The script will not stop for errors typically.

I added stuff in the system use recipe for checking etcd command logs

[https://caltech.sharepoint.com/sites/ovro/projects/SitePages/System-use-recipe-20220403.aspx#resetting-the-f-engine](https://caltech.sharepoint.com/sites/ovro/projects/SitePages/System-use-recipe-20220403.aspx#resetting-the-f-engine)

So in the logs if you see something like `responded to command x: Ok? True" at the right time for the right command then you're likely good.

Received 2 of 1...is probably fine. I'd just run that script again tbh and it might just go away. Otherwise I'd check the snap2 etcd service log files.

kathrynplant, 2023-04-19:
Thanks!  To add some more details in case it's useful, here is another error message I got when I ran restoreSettings.sh again late last night:

Watch callback failed
Traceback (most recent call last):
  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/site-packages/etcd3/watch.py", line 223, in _safe_callback
    callback(response_or_err)
  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/site-packages/lwa_f/snap2_feng_etcd_client.py", line 242, in response_callback
    for event in watchresponse.events:
AttributeError: 'RevisionCompactedError' object has no attribute 'events'
Received 0 of 1 expected responses

jack-h, 2023-04-19:
I also ran `for snap in snap0{1..9} snap{10..11}; do sudo systemctl disable lwa-feng-etcd@$snap.service; done` on `lwa-cr` which I hope will stop it ever coming back at boot. I think this is probably what happened, during the CR NIC-swap saga

jack-h, 2023-04-19:
That `RevisionConpactedError` I think @yupinghuang had a theory about -- I think it is exacerbated by the ~1s polling of various things that was turned on for the 24 hour data collection run and should probably be turned off because it just makes everything perform worse.

caseyjlaw, 2023-05-26:
I think there are multiple issues mixed here, but I ran the settings script without any issue today and have heard no other complains. I'll close this for now.

closed 2023-05-26 by caseyjlaw