ovro-lwa/lwa-issues #354: lwa-feng-etcd services died

yupinghuang, 2023-04-19:
logs to come unless @jack-h gets to them first.

```
(base) [pipeline@lwacalim10 logs]$ for i in {01..11};do
> systemctl status --user lwa-feng-etcd@snap${i};
> done
● lwa-feng-etcd@snap01.service - LWA SNAP etcd service for host snap01
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap02.service - LWA SNAP etcd service for host snap02
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap03.service - LWA SNAP etcd service for host snap03
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap04.service - LWA SNAP etcd service for host snap04
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap05.service - LWA SNAP etcd service for host snap05
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap06.service - LWA SNAP etcd service for host snap06
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap07.service - LWA SNAP etcd service for host snap07
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap08.service - LWA SNAP etcd service for host snap08
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap09.service - LWA SNAP etcd service for host snap09
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap10.service - LWA SNAP etcd service for host snap10
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
● lwa-feng-etcd@snap11.service - LWA SNAP etcd service for host snap11
   Loaded: loaded (/home/pipeline/.config/systemd/user/lwa-feng-etcd@.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
```

yupinghuang, 2023-04-19:
oh I restarted them btw

jack-h, 2023-04-19:
You have any idea what went on here? Did someone stop these deliberately to prevent interference?

jack-h, 2023-04-19:
Well, I am confused about why this stopped and why it didn't restart.

jack-h, 2023-05-25:
Is this still happening @yupinghuang ?

caseyjlaw, 2023-05-25:
The fact that the dashboard stopped at the same time suggests it is not an f-eng service issue. That might be reason enough to close this issue.
I suspect a user did this (it is during working hours in California).

closed 2023-05-25 by caseyjlaw