ovro-lwa/lwa-issues #357: where did dashboard logs go?

yupinghuang, 2023-04-26:
```
(base) [pipeline@lwacalim10 ~]$ journalctl -u dashboard
-- Logs begin at Tue 2023-04-25 17:27:25 UTC, end at Wed 2023-04-26 14:52:41 UTC. --
-- No entries --
```

caseyjlaw, 2023-04-26:
I had logs going to a file instead of syslog. I've updated the systemd config file.
I can't run `daemon-reload` without a password. Can someone (privately) remind me how I set that password?

rh-codebase, 2023-04-26:
shouldn't need a pwd if the cmd is run like so:

sudo systemctl .....

caseyjlaw, 2023-05-01:
Ok, I think I've set this up as is done for the data recorders. However, I'm still not seeing logs with `journalctl`. Does anyone have a good reference file I can compare to, so I can figure out what's wrong? I'd love to see more logging going into elastic.

yupinghuang, 2023-05-02:
syslog goes to `/var/log/messages` and not the journal it seems like
```
(base) [yuping@lwacalim10 ~]$ sudo grep dashboard /var/log/messages | tail
May  2 15:36:49 lwacalim10 dashboard[3338013]: 2023-05-02 15:36:48,121 {"mjd": 60066.6505569568, "proj": "dsa", "subsystem": "System", "app": "_", "version": "_", "module": "dsaStore", "function": "c-tor", "level": "info", "time": "2023-05-02T15:36:48.121052Z", "msg": "DsaStore created"}
May  2 15:37:36 lwacalim10 dashboard[3338013]: 2023-05-02 15:37:36,472 404 GET /favicon.ico (127.0.0.1) 0.89ms
May  2 15:37:36 lwacalim10 dashboard[3338013]: 2023-05-02 15:37:36,478 WebSocket connection opened
May  2 15:37:36 lwacalim10 dashboard[3338013]: 2023-05-02 15:37:36,489 ServerConnection created
May  2 15:44:10 lwacalim10 dashboard[3338013]: 2023-05-02 15:44:08,382 {"mjd": 60066.65565257589, "proj": "dsa", "subsystem": "System", "app": "_", "version": "_", "module": "dsaStore", "function": "_check_host", "level": "info", "time": "2023-05-02T15:44:08.382547Z", "msg": "TODO: implement"}
May  2 15:44:12 lwacalim10 dashboard[3338013]: 2023-05-02 15:44:10,527 {"mjd": 60066.6556773883, "proj": "dsa", "subsystem": "System", "app": "_", "version": "_", "module": "dsaStore", "function": "_check_port", "level": "info", "time": "2023-05-02T15:44:10.526328Z", "msg": "TODO: implement"}
May  2 15:44:15 lwacalim10 dashboard[3338013]: 2023-05-02 15:44:12,854 {"mjd": 60066.65570432706, "proj": "dsa", "subsystem": "System", "app": "_", "version": "_", "module": "dsaStore", "function": "c-tor", "level": "info", "time": "2023-05-02T15:44:12.853842Z", "msg": "DsaStore created"}
May  2 15:46:55 lwacalim10 dashboard[3338013]: 2023-05-02 15:46:54,473 {"mjd": 60066.65757491492, "proj": "dsa", "subsystem": "System", "app": "_", "version": "_", "module": "dsaStore", "function": "_check_host", "level": "info", "time": "2023-05-02T15:46:54.472629Z", "msg": "TODO: implement"}
May  2 15:46:57 lwacalim10 dashboard[3338013]: 2023-05-02 15:46:55,781 {"mjd": 60066.65759005354, "proj": "dsa", "subsystem": "System", "app": "_", "version": "_", "module": "dsaStore", "function": "_check_port", "level": "info", "time": "2023-05-02T15:46:55.780615Z", "msg": "TODO: implement"}
May  2 15:46:58 lwacalim10 dashboard[3338013]: 2023-05-02 15:46:57,219 {"mjd": 60066.65760670525, "proj": "dsa", "subsystem": "System", "app": "_", "version": "_", "module": "dsaStore", "function": "c-tor", "level": "info", "time": "2023-05-02T15:46:57.219323Z", "msg": "DsaStore created"}
```

it'd be easier to keep service logs in the journal. I'll play with it later today.

yupinghuang, 2023-05-11:
Ok I removed the syslog stuff and it turns out the command should be

```
journalctl --user-unit dashboard
```
for user units...

yupinghuang, 2023-05-11:
[image]

Still need to working on mapping...but the dashboard log is now also on kibana

closed 2023-05-11 by yupinghuang