ovro-lwa/lwa-issues #358: storage vdev failure.

rh-codebase, 2023-05-01:
Sometime after generating about 120TB( out of 175TB)  of fake data on a 12-wide z1 the vdev is in a degraded state with 1 drive failed and another degraded. Needs fixing.

kembyct, 2023-05-01:
The new 12-wide z1 vdevs have been created all storage nodes. There is not any indication of failure so far.

rh-codebase, 2023-05-01:
I think this comment belongs under #359.

rh-codebase, 2023-05-01:
I now see the vdev totally suspended. The degraded drive(sdj) is unavailable and the faulted drive(sdk) is still in place. Was the wrong drive pulled or has sdj totally failed?

            sdj     UNAVAIL    253     3     0
            sdk     FAULTED    255     3     0  too many errors

kembyct, 2023-05-02:
They are both signal red when I went to LWA shelter. I’m not sure what’s wrong, but I double check it tomorrow.

rh-codebase, 2023-05-02:
OK. So were any drives pulled from the chassis? Note, this is z1 so only the faulted drive can safely be removed.

kembyct, 2023-05-02:
Yes, I was doing it to check the warranty on the disk when you told me to do so, and then I put them right back up since the spare disks were not around.

rh-codebase, 2023-05-02:
What information on the disk is needed to check warranty? 

I now see the vdev is doing a resilver. Can you explain steps taken to get there?

The following info is good to have as it will likely impact the storage in both reading and writing:  

' [root@lwastor01 ubuntu]# zpool status -v
  pool: lwastor01
 state: ONLINE
status: One or more devices is currently being resilvered.  The pool will
	continue to function, possibly in a degraded state.
action: Wait for the resilver to complete.
  scan: resilver in progress since Tue May  2 08:22:37 2023
	23.9T scanned at 16.9G/s, 597G issued at 421M/s, 121T total
	49.9G resilvered, 0.48% done, 3 days 11:10:21 to go '

kembyct, 2023-05-02:
sudo zpool clear lwastor01 sdk
sudo zpool clear lwastor01 sdj 
To clear the mistakes.
I also went to LWA shelter this morning and find out that some disks were not properly inserted, that's why I could only list 11 disks on some udevs Yesterday. I'm fixing those udevs with the right required disks number

kembyct, 2023-05-02:
It has not been replaced. There is not any spare disk available. We're supposed to get them by tomorrow.