ovro-lwa/lwa-issues #363: fast visibility data recorders falling behind

caseyjlaw, 2023-05-03:
Digital system and python control code can now run `drvf` observations. A recent test was able to start the observation and record some data. However, the recorder status messages (reported by mcs client) reported that the recorders fell behind after 30 seconds or so.
No checks of the data were done, but many files were recorded on May 1.

jaycedowell, 2023-05-03:
From the logs it looks like it is the`WriterOp` that is failing to keep up.  Things seem fine until the recording starts.  The recording starts and packets start dropping and everything recovers after the recording ends.

Somewhat related I noticed that the logs on lwacalim01 showed the ~60 load that was reported at https://github.com/lwa-project/ovro_data_recorder/issues/31.

yupinghuang, 2023-05-03:
Is it just calim01 that's having issues?

One thing to try is to write to /fast (the NVMe).

jaycedowell, 2023-05-03:
It seems to be other nodes as well.  The logs reminded me of the high load problem in 01.

Writing to /fast would be a thing to try.  That should only require a service change and restart of the DRs.

jaycedowell, 2023-05-08:
I've updated the dr-vfast-5/6 services on lwacalim03 to write to /fast for testing.

caseyjlaw, 2023-05-11:
How did writing to /fast go? Can I help with testing for this?

jaycedowell, 2023-05-11:
I haven't had a chance to test /fast.

caseyjlaw, 2023-05-17:
It would be nice to have this working for the Thu/Fri visit by the solar team.
Can I run a test on this today? I could probably infer what you want to do from the changes you made to dr-vfast-5/6 services.

jaycedowell, 2023-05-17:
Sure, you can give it a shot.  I'm not sure how well it would roll out to other nodes since there appear to be other users of /fast.

caseyjlaw, 2023-05-26:
This seems to be working now.

closed 2023-05-26 by caseyjlaw