ovro-lwa/lwa-issues #367: new account request

caseyjlaw, 2023-05-10:
We have a new student, Ethan Alderete, starting on OVRO-LWA data analysis. We'll need an account set up on ssh.ovro.caltech.edu and LWA systems.
Username 'ea' is preferred, but 'ealderete' is also fine.

Public key:
```
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIJezf5QojlO9/RErnxjWp3PgiE0i5fkpRxt90feF1kDU git_key
```

kembyct, 2023-05-10:
The student has been given access to the servers.

caseyjlaw, 2023-05-11:
Thanks for the quick work, Kemby.
Ethan is still having trouble getting onto the ssh gateway. Here is a log:

```
> ssh -v -YA ea@ssh.ovro.caltech.edu
OpenSSH_9.0p1, LibreSSL 3.3.6
debug1: Reading configuration data /Users/ethanalderete/.ssh/config
debug1: /Users/ethanalderete/.ssh/config line 1: Applying options for [ssh.ovro.caltech.edu](http://ssh.ovro.caltech.edu/)
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 21: include /etc/ssh/ssh_config.d/* matched no files
debug1: /etc/ssh/ssh_config line 53: Applying options for *
debug1: Authenticator provider $SSH_SK_PROVIDER did not resolve; disabling
debug1: Connecting to [ssh.ovro.caltech.edu](http://ssh.ovro.caltech.edu/) port 22.
debug1: Connection established.
debug1: identity file /Users/ethanalderete/.ssh/id_rsa type -1
debug1: identity file /Users/ethanalderete/.ssh/id_rsa-cert type -1
debug1: identity file /Users/ethanalderete/.ssh/id_ecdsa type -1
debug1: identity file /Users/ethanalderete/.ssh/id_ecdsa-cert type -1
debug1: identity file /Users/ethanalderete/.ssh/id_ecdsa_sk type -1
debug1: identity file /Users/ethanalderete/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /Users/ethanalderete/.ssh/id_ed25519 type 3
debug1: identity file /Users/ethanalderete/.ssh/id_ed25519-cert type -1
debug1: identity file /Users/ethanalderete/.ssh/id_ed25519_sk type -1
debug1: identity file /Users/ethanalderete/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /Users/ethanalderete/.ssh/id_xmss type -1
debug1: identity file /Users/ethanalderete/.ssh/id_xmss-cert type -1
debug1: identity file /Users/ethanalderete/.ssh/id_dsa type -1
debug1: identity file /Users/ethanalderete/.ssh/id_dsa-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_9.0
kex_exchange_identification: read: Connection reset by peer
Connection reset by 192.100.16.11 port 22
```

Can you check the key or test the setup somehow?

kembyct, 2023-05-11:
Can he restart the sshd  service and try to connect again. Also, he should never try to connect as root because his account will be locked in /etc/host.deny file. He can also try to connect to @dana.ovro.caltech.edu, which is an alias of the same machine.

caseyjlaw, 2023-05-11:
Thanks for the ideas. There may be a problem with his key pair. Can you set this one up on dana, lwamaas, and calim?
```
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAICt7DcPsYDxkbY0l0zXSamqOVCMbhAcKFFOq2ItwFBmv ea@caltech.edu
```

kembyct, 2023-05-11:
The key pair has been added to the servers.

caseyjlaw, 2023-05-15:
Thank you. Ethan has access now.
Can I request another account? Xander Hall needs access to OVRO-LWA and the ssh gateway. Username "xhall". Public key:
```
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABgQCjvVp4poLnoiNp4of+KmViUdWPKsoUf8xzce9v15bMMGXV4fix74YcYAk+unCbitudQozIts756ODKpizR3a9bUqzqwVLQs3kNn+Q94uoGeZKwvRKADX63m2s/F7RAQs7YGFvX+q46y2Qff5voPN8zrrGSDind7C3a5y0Nq3XjB5irqF8rIzZsb0A22scLR+VOCOdWAR2Uqr/Vyy8kqJg1DavbvXhx1l+Umi0+8DfD2Lp51N+ZnSyxCOm0Jemh5nh20NJetjJCDvftFVS2fmq+G/2iaISY4LXFn9wa2oJmXvkKa/wfv9L4b7tNq5c8tqowh0rw5ZvKQpFzPHPxiFMQNGAjtffC+r1AOXi3BL4dChvYxjvpU3DgLykJzEaHqvN2kfxMB1Ua36cFb3ZMFqiM0hzRg7mi/v+SaOHy4sRsw+4LPsZynEgrANKZ6/8aKwVH6KRE9GsIsST3mBw51QnNEosuXcXRrQU8+kfXf9O9yJ1ZAQmaZ1ocCi8mSv7wuTc= 20xha@XandersDesktop
```

kembyct, 2023-05-15:
Great!
Xander has been given access to the servers.

closed 2023-05-15 by caseyjlaw

kembyct, 2023-05-15:
You are welcom.