ovro-lwa/lwa-issues #369: Can't access system health dashboard

akvydula, 2023-05-11:
I can open Grafana, WebUI but not system health dashboard. Casey suggested adding `LocalForward 5601 lwacalim10:5601` after this was moved, but I still get `channel 15: open failed: connect failed: Connection refused`

caseyjlaw, 2023-05-11:
I have noticed that the ssh forwarding for the health dashboard only works if you ssh directly to lwacalim10. You may need a new entry in your ssh config to make that easy.
For reference, here's the New User Tutorial at https://caltech.sharepoint.com/sites/ovro/projects/LWA%20Documents/Tutorials/New%20User%20Setup.docx?d=w9057af7e561946598ff41b6b80191bb9&csf=1&web=1&e=hwTYBM.

yupinghuang, 2023-05-11:
@caseyjlaw having the panel app listening on 0.0.0.0 instead of 127.0.0.1 and configuring the firewall to take traffic on that port should make it work.

akvydula, 2023-05-11:
@caseyjlaw I have the ssh config file set up for lwacalim10 with the local forwarding set up to 5601

caseyjlaw, 2023-05-11:
@yupinghuang I see you made a change to the IP in the config file. What is required to open that port in the firewall? Other apps/ports seem to pass through without changes to the firewall.

yupinghuang, 2023-05-12:
Actually -- it was set up at 0.0.0.0.

However -- 5601 is where kibana lives, not the dashboard.

Here's what I see on calim10
```
(base) [yuping@lwacalim10 ~]$ sudo firewall-cmd --zone=public --list-ports
5601/tcp 9200/tcp 24224/tcp 10001/udp 10002/udp 11001/udp 11002/udp 20001/udp 20002/udp 24224/udp
```
curiously 5601 is in there....But I added 5006 anyway
```
(base) [yuping@lwacalim10 ~]$ sudo firewall-cmd --zone=public --permanent --add-port 5006/tcp
success
(base) [yuping@lwacalim10 ~]$ sudo firewall-cmd --reload
success
```

And now doing the following works
```
ssh -NTL 5006:lwacalim10:5006 lwamaas
```

dgary50, 2023-05-12:
I am having the same problem, but I don't know how to interpret your solution.  Here is some output from my session, where I first tried to ssh to calim10 and then open the dashboard, which then gave the "connection refused" messages.  I then logged out of calim10 and ran the above command (ssh -NTL 5006:lwacalim10:5006 lwamaas) and got "address already in use."  Please advise...

dgary@R914F1P5:~$ ssh -XYA calim10
Warning: No xauth data; using fake authentication data for X11 forwarding.
Activate the web console with: systemctl enable --now cockpit.socket

Register this system with Red Hat Insights: insights-client --register
Create an account or view all your systems at https://red.ht/insights-dashboard
Last login: Fri May 12 18:08:27 2023 from 10.40.0.4
(base) [dgary@lwacalim10 ~]$ channel 13: open failed: connect failed: Connection refused
channel 13: open failed: connect failed: Connection refused
channel 13: open failed: connect failed: Connection refused

(base) [dgary@lwacalim10 ~]$
(base) [dgary@lwacalim10 ~]$ exit
logout
Connection to lwacalim10.mcs.pvt closed.
dgary@R914F1P5:~$ ssh -NTL 5006:lwacalim10:5006 lwamaas
bind [127.0.0.1]:5006: Address already in use
channel_setup_fwd_listener_tcpip: cannot listen to port: 5006

akvydula, 2023-05-12:
Marin helped edit my config file to include `LocalForward 5006 localhost:5006`  in lwacalim10. With this, when I ssh into lwacalim10 and open the system health dashboard it works fine.

caseyjlaw, 2023-05-12:
@dgary50 Do you use an ssh config file? The New User guide has some recommendations for that. Akshatha's change was to add a few lines to configure the ssh to lwacalim10.

caseyjlaw, 2023-05-13:
@dgary50 One more thing. I tested what you did and confirmed that message in the terminal. Try this instead: `ssh -NTL 5006:lwacalim10.mcs.pvt:5006 lwamaas`

dgary50, 2023-05-14:
@caseyjlaw I tried your version of the ssh command and although I still got a message "Address already in use" I **was** able to connect successfully to the system health dashboard.  Thanks.

closed 2023-05-26 by caseyjlaw