ovro-lwa/lwa-issues #370: Reverse proxy for all the web services

yupinghuang, 2023-05-11:
https://doc.traefik.io/traefik/getting-started/quick-start/

caseyjlaw, 2023-05-11:
I very much want to make the web services easier to use.
Our current plan was for Kemby to see if a VNC client can connect to the private network space. If so, then we'd ask people to use that (AnyConnect). 
Rick also put some effort into proxying, but ran into issues with websocket services. 

@kembyct have you had any luck using VNC with our switches?

@rh-codebase do you think traefik would simplify access while keeping the security you want?

kembyct, 2023-05-11:
The configuration is straight forward but doing it directly on the firewall in the production environment is kind of risky. The setup could make our entire system more vulnerable to bad guys if anything goes wrong.

yupinghuang, 2023-05-12:
> I've had good success elsewhere with setting up a basic Apache gateway
> with a login that uses OAuth. That then allows http traffic to other web
> services via reverse proxy. I don't know much about security, but we've run
> that for years without any problems.
> Could we set up some kind of web-based authentication on the first
> firewall and also allow proxying behind the firewall?

To be fair I was thinking about running a reverse proxy on a single node at
a single endpoint behind the firewall. The user still has to tunnel to that
node but can access the rest of the web services via, say
lwamaas:1234/grafana and stuff like that without having to do more hops and
forwarding.

kembyct, 2023-05-12:
We could implement Twingate service that does not require firewall configuration and port forwarding. It is a secure proxy and has a lot of security features to restrict the traffic to a specific service on a server.

rh-codebase, 2023-05-12:
@yuping that's already setup. I fired up nginx on lwamaas.lwa.pvt a while back to reverse proxy grafana and the webui for those at OVRO. A suggestion would be to\
 create an issue and populate with all the internal services for Kemby to finish up the nginx config. At that point, the VPN can be directed there if that's Kemb\
y's preferred solution. I thought that issue was created but maybe not.

kembyct, 2023-05-12:
The VPN setup on the firewall is just not advised after deeply researching it. I think the nginx should be enough to direct traffic to specific nodes.

caseyjlaw, 2023-05-12:
@kembyct Can you update the VPN ticket with some details? My impression was that it was the most secure way to do this, but I have not done research.
I'll open a new ticket about services that need nginx proxying.

caseyjlaw, 2023-05-12:
Oh wait, this issue describe what we need perfectly!
Kemby, for a list of services that need proxying, please see the table at the top of https://ovro-lwa.github.io/.

kembyct, 2023-05-12:
I think the VPN setup would need to be tested and need some inputs of maybe a security team before deploying on the system. 

I'll check the nginx setup to complete it.

kembyct, 2023-05-15:
@caseyjlaw I've checked the table, and since the proxy works on application level, I think each protocol needs to have some sort of host or domain identifier as part of its messages in order for the setup to work. We can't just proxy a multiple port on a server. I think @yupinghuang suggestion makes more sense.
@rh-codebase

kembyct, 2023-05-15:
Each port needs a separate host or domain. We can't proxy more than one service on a single host(IP). The table needs to be modified.

caseyjlaw, 2023-05-31:
@kembyct We can modify the location of services, if needed.
For now, could I ask that you start by proxying the first three on the list (webUI, grafana, dashboard)? They are on different servers and are the most used.

kembyct, 2023-05-31:
Hello Casey
I wonder if I can get the name of those servers.
Thanks.

caseyjlaw, 2023-05-31:
The three are:
- Web UI (webserveruiservice:9090)
- Grafana (grafana:3000)
- System Health Dashboard (lwacalim10:5006)