ovro-lwa/lwa-issues #371: mcs.py: AttributeError: 'NoneType' object has no attribute 'cancelled'

yupinghuang, 2023-05-12:
```
May 12 17:03:39 lwacalim10 bash[2488813]: Exception ignored in: <bound method Client.__del__ of <mnc.mcs.Client object at 0x7f4fdc7e2630>>
May 12 17:03:39 lwacalim10 bash[2488813]: Traceback (most recent call last):
May 12 17:03:39 lwacalim10 bash[2488813]:   File "/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/mnc/mcs.py", line 422, in __del__
May 12 17:03:39 lwacalim10 bash[2488813]:   File "/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/etcd3/client.py", line 174, in close
May 12 17:03:39 lwacalim10 bash[2488813]:   File "/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/grpc/_channel.py", line 1568, in close
May 12 17:03:39 lwacalim10 bash[2488813]:   File "/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/grpc/_channel.py", line 1550, in _close
May 12 17:03:39 lwacalim10 bash[2488813]: AttributeError: 'NoneType' object has no attribute 'cancelled'
```
from dashboard log when it got sigterm

caseyjlaw, 2023-06-29:
I'm seeing this often in logging when the dashboard fails to load.
The deletion of the mcs `Client` class runs `__del__`, which closes the etcd client `client.close()`. I'm deleting the `Client` in the notebook, which triggers this sometimes (not always?).
We made this change to deal with leaking threads (#430). Are we not implementing this correctly? Or do we need to install a new version of etcd3?

yupinghuang, 2023-06-29:
You can safely catch it if the cleanup function and ignore it.

mcs.py keeps track of its own watchers and the __del__ function sometimes tries to clean up wathces that the GC takes care of. An existence check in the code should work. I can work on this next week if no one gets to it first.

caseyjlaw, 2023-06-30:
On the `cjl/dev` branch, I just moved the client.close() call into the try/except. I'll run this for now and see if the dashboard stays up.

caseyjlaw, 2023-08-03:
This is working.

closed 2023-08-03 by caseyjlaw