ovro-lwa/lwa-issues #379: 24 Hour Run Incorrect Date

Hallflower20, 2023-05-16:
While running the mmode pipeline on the 69MHz dataset I get the following warning. I think it is mistaking MJD for JD but this might be  a problem with the data recorder or with the pipeline, but I'm unsure why casa would get it wrong.

```2023-05-15 04:27:28     INFO    MeasIERS::fillMeas(MeasIERS::Files, Double)     Requested JD 60012.5 is outside the range of the IERS (Earth axis data) table.```

caseyjlaw, 2023-05-16:
Thanks for the report, Xander. I scanned the ovro_data_recorder code, but could not find reference to how time is defined. @jaycedowell do you know what might cause this error?

jaycedowell, 2023-05-17:
Nope.

yupinghuang, 2023-05-17:
nvm casacore is just bad at giving correct messages. It really meant MJD.

https://github.com/casacore/casacore/blob/68e43f489abd3d4f1e2fe54a42695396703aa81a/measures/Measures/MeasIERS.cc#L103

Speaking of which, the final bulletin B earth orientation IERS data for MJD 60012 only came out last week.

```
browsetable('/opt/measures/geodetic/IERSeop2000')
```

from CASA gives the following for the last rows. 60018 just made it in.
Hopefully things are now working?

[image]

Hallflower20, 2023-05-17:
Alright, so it is just that the IERS tables aren't up to date yet, so nothing to worry about?

Hallflower20, 2023-05-20:
I am still seeing the same error.

yupinghuang, 2023-05-24:
@Hallflower20 what's the path to the julia wrapper to casacore you're using on the disk?

Hallflower20, 2023-05-24:
This seems to be where the MeasTable.cc is located for the casacore.
```/opt/src/casacore-3.5.0/measures/Measures/MeasTable.cc```