ovro-lwa/lwa-issues #380: new account request

caseyjlaw, 2023-05-16:
Andrea Isella (username "ai") needs an account on cal-im. He already has a key on lwamaas and the ssh gateway.
Public key:
```
ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIO2+bq5MYiopw2b0NpSZyxVL6c+vnB9NS6J2IlhLKAPR ai14@platax.rice.edu
```

kembyct, 2023-05-16:
Done!
Issella can actually connect to calim nodes.

closed 2023-05-16 by caseyjlaw

kembyct, 2023-05-16:
You are welcome.

isella, 2023-05-16:
I still cannot login using my user. ssh -v returns the following

(base) [ai14@platax ~]$ ssh -v calim0
OpenSSH_7.4p1, OpenSSL 1.0.2k-fips  26 Jan 2017
debug1: Reading configuration data /home/ai14/.ssh/config
debug1: /home/ai14/.ssh/config line 19: Ignored unknown option "usekeychain"
debug1: /home/ai14/.ssh/config line 51: Applying options for calim0
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 6: Applying options for *
debug1: Setting implicit ProxyCommand from ProxyJump: ssh -v -W %h:%p lwamaas
debug1: Executing proxy command: exec ssh -v -W lwacalim00.mcs.pvt:22 lwamaas
debug1: permanently_drop_suid: 56212
debug1: identity file /home/ai14/.ssh/id_rsa type 1
debug1: key_load_public: No such file or directory
debug1: identity file /home/ai14/.ssh/id_rsa-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_7.4
OpenSSH_7.4p1, OpenSSL 1.0.2k-fips  26 Jan 2017
debug1: Reading configuration data /home/ai14/.ssh/config
debug1: /home/ai14/.ssh/config line 19: Ignored unknown option "usekeychain"
debug1: /home/ai14/.ssh/config line 29: Applying options for lwamaas
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 6: Applying options for *
debug1: Setting implicit ProxyCommand from ProxyJump: ssh -v -W %h:%p ovro
debug1: Executing proxy command: exec ssh -v -W lwamaas.lwa.pvt:22 ovro
debug1: permanently_drop_suid: 56212
debug1: identity file /home/ai14/.ssh/id_rsa type 1
debug1: key_load_public: No such file or directory
debug1: identity file /home/ai14/.ssh/id_rsa-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_7.4
OpenSSH_7.4p1, OpenSSL 1.0.2k-fips  26 Jan 2017
debug1: Reading configuration data /home/ai14/.ssh/config
debug1: /home/ai14/.ssh/config line 19: Ignored unknown option "usekeychain"
debug1: /home/ai14/.ssh/config line 23: Applying options for ovro
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 6: Applying options for *
debug1: Connecting to ssh.ovro.caltech.edu [192.100.16.11] port 22.
debug1: Connection established.
debug1: identity file /home/ai14/.ssh/id_rsa type 1
debug1: key_load_public: No such file or directory
debug1: identity file /home/ai14/.ssh/id_rsa-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_7.4
debug1: Remote protocol version 2.0, remote software version OpenSSH_7.4
debug1: match: OpenSSH_7.4 pat OpenSSH* compat 0x04000000
debug1: Authenticating to ssh.ovro.caltech.edu:22 as 'ai'
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ecdsa-sha2-nistp256
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: curve25519-sha256 need=64 dh_need=64
debug1: kex: curve25519-sha256 need=64 dh_need=64
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ecdsa-sha2-nistp256 SHA256:iHMF0KwkIz1ZbIGeXnn8c7AUHFN9uzuo9I+zsBgsCw0
debug1: Host 'ssh.ovro.caltech.edu' is known and matches the ECDSA host key.
debug1: Found key in /home/ai14/.ssh/known_hosts:11
debug1: rekey after 134217728 blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey after 134217728 blocks
debug1: SSH2_MSG_EXT_INFO received
debug1: kex_input_ext_info: server-sig-algs=<rsa-sha2-256,rsa-sha2-512>
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey,gssapi-keyex,gssapi-with-mic,password
debug1: Next authentication method: gssapi-keyex
debug1: No valid Key exchange context
[... 81 lines cut ...]
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey,gssapi-keyex,gssapi-with-mic,password
debug1: Next authentication method: gssapi-keyex
debug1: No valid Key exchange context
debug1: Next authentication method: gssapi-with-mic
debug1: Unspecified GSS failure.  Minor code may provide more information
No Kerberos credentials available (default cache: KEYRING:persistent:56212)

debug1: Unspecified GSS failure.  Minor code may provide more information
No Kerberos credentials available (default cache: KEYRING:persistent:56212)

debug1: Next authentication method: publickey
debug1: Offering RSA public key: /home/ai14/.ssh/id_rsa
debug1: Authentications that can continue: publickey,gssapi-keyex,gssapi-with-mic,password
debug1: Offering ED25519 public key: ai14@platax.rice.edu
debug1: Authentications that can continue: publickey,gssapi-keyex,gssapi-with-mic,password
debug1: Next authentication method: password
ai@lwacalim00.mcs.pvt's password: debug1: channel 0: FORCE input drain
debug1: stdio forwarding: done
debug1: channel 0: free: direct-tcpip: listening port 0 for lwamaas.lwa.pvt port 22, connect from 127.0.0.1 port 65535 to UNKNOWN port 65536, nchannels 1
debug1: fd 0 clearing O_NONBLOCK
debug1: fd 1 clearing O_NONBLOCK
Killed by signal 1.

Authentication failed.

isella, 2023-05-16:
this is my ssh/config file:

IgnoreUnknown UseKeychain

ForwardAgent yes 

AddKeysToAgent yes 

ForwardX11   yes 

ForwardX11Trusted yes 

ForwardX11Timeout 1d 

ServerAliveCountMax 10 

ServerAliveInterval 30 

TCPKeepAlive no 

UseKeyChain yes 

IdentityFile ~/.ssh/id_rsa 

Host ovro 

    Hostname ssh.ovro.caltech.edu 

    User ai

Host lwamaas 

    user ubuntu 

    hostname lwamaas.lwa.pvt 

    ProxyJump ovro 

    LocalForward 3000 grafana.lwa.pvt:3000 

    LocalForward 5006 lxdlwacr:5006 

    LocalForward 5007 lxdlwacr:5007 

    LocalForward 5240 localhost:5240 

    LocalForward 9090 webserveruiservice:9090 

    LocalForward 5601 lwacalim10:5601 

 

Host calim0  

    user ai

    hostname lwacalim00.mcs.pvt 

    ProxyJump lwamaas 

    LocalForward 3000 grafana.lwa.pvt:3000 

    LocalForward 5006 lxdlwacr:5006 

    LocalForward 5007 lxdlwacr:5007 

    LocalForward 8889 localhost:8889 

    LocalForward 9090 webserveruiservice:9090 

    LocalForward 5601 lwacalim10:5601

isella, 2023-05-16:
ssh works ok now :-)

kembyct, 2023-05-16:
great!