ovro-lwa/lwa-issues #382: Single-antenna beamforming

iveydavis, 2023-05-16:
Editing cal files so that we can "beamform" with a single antenna. 

The general idea for how I've gone about it is:

1.  Make (or use previously-made) cal files that come from the `/opt/devel/pipeline/distributed-pipeline/orca/extra/ tempcal.py` script
2.  Run the `/opt/devel/idavis/scripts/flag_antennas.py` script to flag all antennas except the one I want a "beam" from
3.  Either move these new cal files to the `caltables/latest/` directory, or make a new config file that references the new caltable directory that gets called in `con = control.Controller(config_file = 'path/to/new/config_file')`

nmahesh1412, 2023-05-18:
Plot thickens... 
Carried out the following tests:
1.) Created new caltables with all the antennas (none flagged/zeroed). And collected data with two beams pointed at zenith. 
[image]
I expected to get the same data. 

2.) Created two sets of caltables : 1.) with all antennas flagged except one, 2.) with all antennas zeroed except one. And collected data with caltable 1 with Beam01 & beam02. Collected data with caltable 2 settings with Beam03 & Beam04
[image]

nmahesh1412, 2023-05-18:
Main take away - even if beam01 & beam02 are calibrated with the same caltables and pointed in the same direction, it records different data values into its XX_data. 
Beam01 & Beam03 returned the recorded the same data even though one was calibrated by flagging all the antennas and the other by zeroing the gains of all the antennas except one. So It was a good test that flagging and zeroing is the doing the same thing.

caseyjlaw, 2023-05-18:
Can you explain now the beam 1 and 2 were set up, including the commands? That's a very simple test that should produce identical results, so if we can't verify that, then all other tests are hard to interpret.

akvydula, 2023-05-18:
Is the difference in the order of noise levels (variance) in the data?

dannyjacobs, 2023-05-18:
What are the units of these plots?  Why are the differences log colorscale for B1-B2 but linear in others?  Judging by the scale ranges, B3-B4 could be similar to B1-B2, but its hard to tell since the latter is linear.

nmahesh1412, 2023-05-18:
@caseyjlaw I used the following commands for Test 1
`from mnc import control
con = control.Controller('/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim_drone_20230517.yaml') ##all_antennas
con.stop_xengine()
con.configure_xengine(recorders=['dr1','dr2','dr3','dr4'],full=True,calibratebeams=True)
con.control_bf(num=1, targetname='zenith', track=False)
con.control_bf(num=2, targetname='zenith', track=False)
con.start_dr(recorders=['dr1','dr2'], duration = 3*60*1e3, time_avg=128)`

iveydavis, 2023-05-18:
Wait, these commands suggest you configured the xengine for all 4 beams with the same configuration file

nmahesh1412, 2023-05-18:
Yes, for the test1 which was a simple control test before all the fancy flagging. Then for Test2: 
```
con1 = control.Controller('/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim_drone_flagged_20230517.yaml') ##flagged_antennas
con2 = control.Controller('/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim_drone_zeroed_20230517.yaml') ##zeroed
con1.configure_xengine(recorders=['dr1','dr2'],full=False,calibratebeams=True) ## configuring for flagged
con2.configure_xengine(recorders=['dr3','dr4'],full=False,calibratebeams=True) ## configuring for zeroed
con1.control_bf(num=1, targetname='zenith', track=False)
con1.control_bf(num=2, targetname='zenith', track=False)
con2.control_bf(num=3, targetname='zenith', track=False)
con2.control_bf(num=4, targetname='zenith', track=False)
con1.start_dr(recorders=['dr1','dr2'], duration = 3*60*1e3, time_avg=128)
con2.start_dr(recorders=['dr3','dr4'], duration = 3*60*1e3, time_avg=128) 

```

nmahesh1412, 2023-05-18:
@dannyjacobs Good questions! Sorry I should have been more clear
1.) So the y-axis is not any calibrated units. I am just plotting data from the XX beam (so just absolute units?)
2.) Sorry about the scales too. Everything in general is in log-scale except when the differences are exactly zero like beam1- beam3 and beam2-beam4

nmahesh1412, 2023-05-18:
@dannyjacobs Also here are some 2D plots to help us look at how much the variation is: 
[image]

nmahesh1412, 2023-05-18:
[image]

akvydula, 2023-05-18:
Why is the absolute difference negative values? (is that what you mean by abs?)

iveydavis, 2023-05-18:
> Why is the absolute difference negative values? (is that what you mean by abs?)

If you're referring to my plots-- they're all log scale (e.g. -3 refers to 10^-3)

nmahesh1412, 2023-05-18:
One more question that this test was able to answer/look into: 
The total amplitude of the data when using all the antennas Vs just one antenna
@gregghallinan what you mentioned on yesterday's call did make sense that we will expect one antenna to see more of the sky and hence get more amplitude in the data. But what I find from the recent test is that the amplitude of using all the antennas is roughly 100 times larger than just a single antenna. In line with what Larry expected us to find.
[image]

nmahesh1412, 2023-05-18:
> Why is the absolute difference negative values? (is that what you mean by abs?)

If you mean my plots. I did not take the absolute of the difference.

nmahesh1412, 2023-05-19:
It doesnt make sense to me that beam01 & beam03 are exactly the same and
beam02&beam04 are exactly the same. Here are the time stamps of each beam data collected. So beam01 & beam02 are different within the integration time (64ms). Whereas Beam03&beam04 were started solid 10 mins later
timestamps for each beam 
 Beam01-time: 2023-05-17T21:09:50.187185526 
 Beam02-time: 2023-05-17T21:09:50.179160833 
 Beam03-time: 2023-05-17T21:19:49.600841045 
 Beam04-time: 2023-05-17T21:19:49.697139025

nmahesh1412, 2023-05-19:
After looking at the timestamps and certainly expect things to be difference for beam3&4, I checked my code and found a dumb bug. Now the difference plots are here: 
[image]

All the beams differ by the same level. So, this answers the question for the drone team (@akvydula @dannyjacobs ) - that you can get a single antenna beamformed data with either flagged.py or zeroed.py. But it is recommed to flag rather than zero the cal tables because there is a normalization step that takes 1/cal and we dont recommend the divide by zero :D

Also, I apologise for the bug and the confusion

akvydula, 2023-05-19:
Thanks @nmahesh1412! This is good, and makes perfect sense. As for the recommendation, my understanding is that flag_antenna.py flags AND zeros out the gains, where as zero_antenna.py just zeros out the gain but does not alter the flags. So either ways, there is a divide by zero case (unless my understanding of this script is wrong).

akvydula, 2023-05-22:
> This script should become obsolete-- flagging antennas should be done through mnc code, not through an external script directly modifying caltables.

Do you mean this as a suggestion for future use or does this already exist? I am still using the external script to modify the cal tables.  @iveydavis

caseyjlaw, 2023-10-12:
To do:
- mnc-python beamformer control should select a set of antennas via a weights function that is normalized.
- beamformer control should be able to parse gain tables to get flagged antennas
- remove gain normalization

Goal is to remove scaling of beam power with antenna selection or gain table flag list.

caseyjlaw, 2023-10-19:
Details on vis-based (from an MS) beamforming in #475

caseyjlaw, 2023-10-19:
This has been implemented in "bfnorm" branch and deployed to `development` environment. Ready for testing.

caseyjlaw, 2023-10-20:
Ran two tests and Dale visualized output.
Test 1: flagging more antennas while tracking sun
- start data recorder for 10 minutes
- fiddled around a minute or two
- started tracking sun for 2 minutes with three antennas flagged (001, 002, 003)
- (maybe a small gap here)
- started tracking sun for 2 minutes with 100 antennas flagged (001-100)
- (maybe a small gap here)
- started tracking sun for 2 minutes with 200 antennas flagged (001-200)
- (maybe a small gap here)
- started tracking sun for 2 minutes with only one antenna (001)

[image]

Test 2: using caltable to ignore bad and outrigger antennas. track sun and let it drift out. So:
- start data recorder for 10 minutes
- started tracking sun for 2 minutes with nominal good core antennas
- stopped tracking for a few minutes (4?)
- started tracking for 2 minutes with nominal good core antennas. This last one may have slipped past the recording window.

[image]

caseyjlaw, 2023-10-20:
The first test shows that ignoring more antennas produces less power in beam. Seems to be in proportion to the number of antennas, but not clear whether that is due to changing beam shape or the scaling of beamformer output.

The second test shows the expected beam power drop, given the width of a core-only beam. Quality seems good and amplitude is ok, despite the lack of a self-normalization in x-engine. Looks ok.

surajit4444mondal, 2024-01-25:
Dale, did a test where he did a drift scan on the Sun with one beam and another beam was pointed on the Sun. The drift scan data was used to estimate the background and then the residual was scaled by a frequency dependent correction factor.  Next the corrected beam flux was compared with the flux obtained from the realtime solar imaging pipeline. The result is attached in the figure. The orange and blue lines show the flux obtained from the image and beam. It should be noted that the beam flux was obtained by doing XX+YY . We find that there is a factor of 2 discrepancy between the two fluxes. Several different things can contribute to this issue: a)primary beam correction was done for beam flux, but done on the image 2)Is there any difference because of definition of I  in the beam flux ? Understanding this discrepancy is work in progress and will be reported later. The correction factor is given by the Nant^2, where Nant is the  number of unflagged antenna in each channel. For now, we have used a polarisation independent correction factor, although, it can be easily changed to be polarisation dependent as well. The reasoning for this correction factor is also provided in the attached note. [beamformer_normalisation.pdf](https://github.com/ovro-lwa/lwa-issues/files/14056669/beamformer_normalisation.pdf) 
[image]