ovro-lwa/lwa-issues #403: Solar-specific configuration file

dgary50, 2023-05-18:
1. Copy latest configuration file to /opt/devel/dgary.
2. Edit cal_directory entry to point to /home/pipeline/caltables/solar_cal.
3. Put any calibration table updates into that /home/pipeline/caltables/solar_cal folder.
4. When starting mnc connection, specify configuration file as /opt/devel/dgary/lwa_config_calim.yaml

In principle, we can make other changes to the configuration for solar-only also.

To check that there are no changes to other parts of the configuration, do a diff between the solar configuration file and the latest file:
   diff /opt/devel/dgary/lwa_config_calim.yaml /home/pipeline/caltables/lwa_config_calim.yaml

iveydavis, 2023-05-18:
re: the caltables that you references, here's an outline of making them:

Making Calibration Files for the Solar Beam:

step 1: 
	Make sure day time settings are loaded (i.e. load them in yourself)
	
step 2:
	Collect a few minutes of slow visibilities
	
step 3:
	run `pdsh -w lwacalim[01-08] 'conda activate py38_orca && cd /opt/devel/pipeline/distributed-pipeline/ && pipenv run python orca/extra/tempcal.py YYYMMDD_HHMMSS ANT_IND'`
	
step 4:
	check estimated delay scatters (order 100ns or less)
	check # flagged channels (~ a few for good bands, hundreds for bad bands (e.g. below 30 MHz))
	
step 5:
	Check calibration outputs (e.g. delay tables K0) on one of the recording nodes (recommend lwacalim06) in the directory `/fast/pipeline/tempcal/YYYMMDD_HHMMSS/YYY-MM-DDTHH:MM:SS`
	
step 6:
	If the solutions are bad, repeat step 3 and 4 either with a different timestamp or a different refant. If the solutions are good, move the files from `/home/pipeline/caltables/staging/` to `/home/pipeline/caltables/solar_cal/`

caseyjlaw, 2023-12-14:
This procedure is working well. I think it is also documented via the tutorials or orca docs. If some details here are not included there, please update the docs.
I'll close for now.

closed 2023-12-14 by caseyjlaw