ovro-lwa/lwa-issues #404: Appropriate deconvolution options for solar imaging

surajit4444mondal, 2023-05-18:
WSCLEAN parameters

caseyjlaw, 2023-12-14:
@surajit4444mondal Can you document these now and close this issue?

surajit4444mondal, 2023-12-15:
@binchensun Can you please add the comparisons you did with the different weighting schemes here? I guess we can then close this issue, right?

binchensun, 2023-12-15:
[I just copied the report from the Slack solar channel here.]

Here is a brief summary of my experiment of fine-channel imaging spectroscopy for the Sun, which might be applicable to some other cases (e.g., strong sources in a relatively small field of view). I used CASA tclean and wsclean to image a single 10s snapshot slow vis data (20230502_174255) at all subbands (well, actually from ~29 MHz due to strong RFIs at lower frequencies). All images were performed on self-calibrated, non-solar-source-subtracted, and phase-shifted measurement sets, and have 1024 x 1024 pixels with 1 arcmin pixel size.

- For CASA tclean, only uniform weighting was used (as other weight schemes had errors). I looped over all channels with a 16-channel binning. Default specmode (mfs) and deconvolver (hogbom) were used.

- For wsclean, the “-channels-out” option is used for spectral imaging under the following schemes: 1) Uniform weighting; 2) Briggs = -1. 3) Briggs = -0.5; 4) Briggs = 0. Also tried is the -multiscale option with -multiscale-scale-bias of 0.5 and 0.8. I also tried another run with -no-negative added. All the latter three runs have Briggs = -1.

The attached pdf shows 38 selected images from 29 MHz to 85 MHz. Each panel corresponds to a test run with the corresponding method labeled in the title. Beam parameters as well as peak, rms, and dynamic range values are also computed and shown (in Jy/beam). Surprisingly the results of CASA tclean’s uniform weighting are nothing like those from wsclean’s uniform weighting, but are more closer to wsclean’s Briggs -0.5 or 0 results. I personally like wsclean Briggs -0.5 the best, which is a good comprise between image quality and resolution. Of course, a more negative Briggs value could be useful if resolution is the emphasis, and a more positive Briggs to emphasis sensitivity/dynamic range. The time it took to complete the 168 images (on a single calim7 thread) is as follows. They actually do not differ drastically.

- tclean uniform: 1085.6 s
- wsclean uniform: 820.7 s
- wsclean imaging briggs-1: 882.3 s
- wsclean imaging briggs-0.5: 982.5 s
- wsclean imaging briggs-0: 1034.9 s
- wsclean imaging briggs-1 multiscale bias 0.5: 1222.0 s
- wsclean imaging briggs-1 multiscale bias 0.8: 961.7 s
- wsclean imaging briggs-1 no-
[20230502_174255_solar_spec_img_compare.pdf](https://github.com/ovro-lwa/lwa-issues/files/13688028/20230502_174255_solar_spec_img_compare.pdf)
negative: 921.9 s

binchensun, 2023-12-15:
For the solar imaging pipeline, I am settled with wsclean with briggs = -0.5. Note this is based on imaging at a single time during the early phase of the commissioning. A similar practice may need to be done for the full array.

surajit4444mondal, 2023-12-15:
We also noted that the weighting scheme implemented in WSClean is slightly different from the one implemented in CASA and this might be one reason why the CASA image with uniform weighting is so different from that obtained using WSClean. The details of WSClean and CASA weighting schemes can be found in https://wsclean.readthedocs.io/en/latest/image_weighting.html#relation-to-visibility-weights and https://casadocs.readthedocs.io/en/stable/notebooks/synthesis_imaging.html#Imaging-Algorithms

caseyjlaw, 2023-12-15:
Thanks for the update. I'll close this issue, but you can add more later, if needed.

closed 2023-12-15 by caseyjlaw