ovro-lwa/lwa-issues #406: Timeout error connecting to calim10

keelder, 2023-05-19:
I am trying to login to calim10 but am running into this error: 

`connect 10.99.94.143 port 6000: Connection timed out`

Even though I am logged in to pipeline, my connection is slow and I get this error when I try to do anything more complicated than moving into new directories.

yupinghuang, 2023-05-24:
Can you give me a time when you tried logging in (or just try a gain).

Also, what did you have in your ssh config for the calim nodes? What command did you use for SSH'ing?

keelder, 2023-05-24:
I have tried to log in multiple times and am trying again right now. 

Here is my ssh config: 
```
Host lwamaas
    user ubuntu
    hostname lwamaas.lwa.pvt
    ProxyJump ovro
    LocalForward 3000 grafana.lwa.pvt:3000
    LocalForward 5240 localhost:5240
    LocalForward 9090 webserveruiservice:9090
    LocalForward 8081 lwacalimhead:8081

Host calimh
    hostname lwacalimhead.mcs.pvt
    user kelder
    ProxyJump lwamaas

Host calim0
    user kelder
    hostname lwacalim00.mcs.pvt
    ProxyJump lwamaas

Host calim1
    user pipeline
    hostname lwacalim01.mcs.pvt
    ProxyJump lwamaas
    LocalForward 5006 lwacalim10.mcs.pvt:5006
    LocalForward 5601 lwacalim10.mcs.pvt:5601

Host calim10
    user pipeline
    hostname lwacalim10.mcs.pvt
    ProxyJump lwamaas
    LocalForward 5006 lwacalim10.mcs.pvt:5006
```

The command is `ssh -YA calim10`

yupinghuang, 2023-05-24:
Gonna be real annoying and ask for your SSH config for `ovro`

keelder, 2023-05-24:
```
Host ovro
    Hostname ssh.ovro.caltech.edu
    User kelder
```

yupinghuang, 2023-05-24:
Um...so port 6000 is the X11 forwarding port. So first thing I'd try is doing ssh without the `-Y`.

Also, any idea what `10.99.94.143` is? It doesn't seem to be an OVRO IP address.

My best guess is that you're SSH'ing from a node that you SSH'd to (and `10.99.94.143` is somewhere along there) and somehow the X11 forwarding didn't get through?

keelder, 2023-05-24:
Trying to get on with just `ssh -A calim10` or even `ssh calim10` gives the same error. 

No idea what IP that is, it is not for my system. 

I am using the command directly from my computer's terminal, not SSH'ing from another node directly, so is it something wrong with the proxy jump? Although I can access calim00 without issue and it has the same config settings.

yupinghuang, 2023-05-24:
hmmm can you slack msg me your entire ssh config file?

caseyjlaw, 2023-05-26:
Did this get resolved? If not, feel free to reopen.

closed 2023-05-26 by caseyjlaw