ovro-lwa/lwa-issues #408: Inconsistent antenna names between slow and fast visibilities

binchensun, 2023-05-21:
We have been looking at the fast mode visibility data taken on May 19, 2023. An example dataset at /fast/pipeline/fast/20230519_214959_78MHz.ms. It seems to have a credible amplitude vs. uvdist distribution
[image]. 

We have transferred the antenna-based bandpass tables derived from the slow visibility data to the fast visibility dataset. However, we had trouble producing an image of the fast visibility. Then we discovered that the antenna-id-to-name mapping is inconsistent between the slow and fast visibilities:

We have provided the following list of antenna names for fast visibilities in the lwa_config_calim.yaml file:
LWA-015, LWA-018, LWA-038, LWA-067, LWA-085, LWA-096, LWA-131, LWA-151, LWA-180, LWA-191, LWA-260, LWA-263, LWA-264, LWA-272, LWA-274, LWA-275, LWA-277, LWA-278, LWA-281, LWA-284, LWA-285, LWA-285, LWA-286, LWA-287, LWA-289, LWA-290, LWA-295, LWA-301, LWA-304, LWA-305, LWA-307, LWA-313, LWA-314, LWA-320, LWA-321, LWA-326, LWA-327, LWA-328, LWA-329, LWA-330, LWA-334, LWA-335, LWA-336, LWA-337, LWA-338, LWA-339, LWA-340, LWA-347

However, the list of antennas in the fast visibility measurement set shows as (one can get using listobs() or msmd.antennanames()):
LWA259, LWA268, LWA267, LWA271, LWA269, LWA276, LWA273, LWA278, LWA277, LWA282, LWA281, LWA307, LWA285, LWA309, LWA308, LWA311, LWA310, LWA313, LWA312, LWA321, LWA314, LWA330, LWA327, LWA338, LWA332, LWA340, LWA339, LWA352, LWA341, LWA362, LWA353, LWA257, LWA255, LWA260, LWA258, LWA265, LWA263, LWA272, LWA270, LWA283, LWA280, LWA288, LWA284, LWA292, LWA291, LWA296, LWA295, LWA301

They obviously do not match each other (I have removed the hyphens when comparing, of course). We tried to transfer the slow visibility bandpass solutions for either list of the antenna names to the fast visibility but had no luck. Apparently, this is because the antenna-name mapping is inconsistent between the slow and fast visibilities. We will use the X-Y-Z coordinates themselves to determine the exact antenna and see how it goes. But it would make our life much easier if we have a consistent antenna–name mapping scheme between the slow and fast visibilities. 

@marinanderson @jaycedowell

jack-h, 2023-05-22:
The antenna names in the fast vis output files map as:
```
LWA259 = correlator input 1
LWA-268 = 2
LWA-267 = 3
... [I didn't check the rest]
```

Looking at the packet transmission code, the custom stand numbers are never written into packets -- the map currently needs to be known by the downstream receiver.

Questions (I think for @jaycedowell ):
1. Is it even possible to use bifrost's send function to step through a list of arbitrary stand numbers without one call per packet? https://github.com/realtimeradio/caltech-bifrost-dsp/blob/63b8cec8d39beb0c6d87c5864c3a74719be39047/pipeline/lwa352_pipeline/blocks/corr_output_part_block.py#LL392C57-L392C58
2. Will the data recorder tolerate arbitrary vales in the `stand_i`, `stand_j` fields of the output packets?
3. The send function doesn't quite look right anyway (surely `first_baseline` should increment with each iteration)?

Number (3) I'm looking at now. For 1 & 2 it would be useful to get input from others.

jack-h, 2023-05-22:
Turns out I just can't read -- `first_baseline` increments as it should, and in wireshark the packets look legitimate (though use headers from stand=1 to stand=48)

binchensun, 2023-05-22:
> The antenna names in the fast vis output files map as:
> 
> ```
> LWA259 = correlator input 1
> LWA-268 = 2
> LWA-267 = 3
> ... [I didn't check the rest]
> ```
> 
The antennas in the fast visibility map to antenna IDs 1-48 in the slow visibility (which seems to be the same as the correlator input).

jaycedowell, 2023-05-23:
This is a known issue with the fast visibility recorder that comes from it not having access to the antenna selection info.  I have a branch called [fastvis_ordering](https://github.com/lwa-project/ovro_data_recorder/tree/fastvis_ordering) that should address this but it hasn't been tested.  The fast recorder in this branch polls the X-engines for the antenna selection at the beginning of a recording session.

For @jack-h's questions:
 * (1) - That kind of custom indexing isn't currently supported.
 * (2) - Maybe-ish?  You could only send a subset of baselines but tell Bifrost to expect all N*(N+1)/2 and let it zero fill what doesn't come in.  You could then go through that and pull out what has actually been collected.  That seems kind of crazy.

jack-h, 2023-05-23:
Thanks @jaycedowell 
A while back when we were having etcd troubles I tried to reduce etcd load and one of the casualties was baseline selection readback. I'll put this back in (hopefully via a mechanism which doesn't mean the order map gets transmitted every second from every xeng pipeline,  which was the situation before)

jaycedowell, 2023-05-24:
Thanks @jack-h  I'll see if I can find some time next week to test this.  There is also the possible problem of #363 on the recorder side that needs to be looked at as well.

jaycedowell, 2023-05-30:
The mapping should be fixed with https://github.com/lwa-project/ovro_data_recorder/pull/34.  I've also restarted the fast vis. recorder services so these changes are active.

dgary50, 2023-07-03:
This problem was traced to having incorrect antennas selected relative to the database metadata (and wrong delays, etc.). Jayce made a fix to the antenna selection and now the antenna selection and metadata agree. Based on Bin Chen's investigation of data taken on 2023 Jun 22, fast and slow visibilties now agree, so this issue is closed.

closed 2023-08-03 by caseyjlaw