ovro-lwa/lwa-issues #409: Fast visibilities have different phase vs. frequency behavior from slow visibilities

binchensun, 2023-05-21:
This is related to issue #408, but perhaps has a different origin. The fast visibility data we are investigating is the same one in #408: /fast/pipeline/fast/20230519_214959_78MHz.ms.

First, we derived the bandpass solutions from the slow visibility data derived from /data07/slow/20230516_063559_78MHz.ms. 
We verified that the solutions work well in correcting the slow visibility data at the closest time to the fast visibility. (/data07/slow/20230519_214953_78MHz.ms). It produces a reasonable solar image below.
[image]

Then, we transfer the bandpass solutions from the slow visibility dataset to the fast visibility by only selecting the antennas in the fast visibility dataset. The assumption is that fast visibility has identical antenna-based gains to its slow visibility counterpart. However, we failed to make an image this way. We investigated it further, but it seems that the uncalibrated complex visibilities are different enough from each other in the first place. This is unexpected because the slow and fast visibilities see the same source in the sky (dominated by the bright type III burst) and should have the same antenna gains. Here is one example of comparison at two baselines
[image]
[image]

First, the phase vs. frequency variations are very different. Second, the fast visibility phases seem to have large wraps across the 4.6 MHz bandwidth. 
@surajit4444mondal did an experiment by placing the solar burst derived from the slow visibility image into the model column of the fast visibility data, and then solving for the delays across the 78 MHz band. The values are quite large, between 1000-3000 ns. See a plot below.
[image]

Therefore, we suspect that the cable delays were not applied properly. I went ahead and plotted the cable delays from /home/pipeline/proj/lwa-shell/mnc_python/data/cable_delays.csv for the fast visibility antennas, shown below.
[image]

They do not match exactly, but the magnitude of these delays are both at the x1000 ns level. Our conclusion is that something is wrong when applying the delays to the fast visibility data.

jack-h, 2023-05-22:
I think this might be a bug in packet header generation, mentioned in #408 , which I'm attempting to investigate/address now. In any case, it would likely be best to verify a recording with the system in test mode.

jack-h, 2023-05-22:
After looking a bit, I think the correlator output is "fine" in the sense that it is not buggy but also seemingly isn't quite compatible with how the data recorder treats antennas labels (see #408)

I think the first problem here is that the fast datasets contain the antennas which were selected, but they aren't labelled as such. So applying slow-vis calibrations based on these labels isn't going to work. Possibly it would work if you manually change the fast vis meta data to match the antennas you wanted, and their positions.

Plotting the bandpasses of the fast data, alongside the slow data bandpasses for both the desired and labelled baselines might be informative

binchensun, 2023-05-22:
Hi Jack, I have checked the uv points in the fast visibility data, which perfectly match the expectations from the antenna XYZ locations. See some plots below: 

This figure is the antenna configuration produced by CASA's plotants (left) and u vs. v with plotms (right). The latter comes directly from the actual visibility data.

[image]

The following plot is the antenna configuration using the known XYZ locations of the antennas list in the fast visibility (left) and my calculated uv points with the array phased toward the zenith at the time of the observation (right; the visibility is not used). You can that see they match exactly.

[image]

My conclusion from this practice is that the antenna labels should be fine because they match their XYZ locations and the expected uv(w) points in the recorded visibilities (unless the uvw records are wrong themselves). It is just the antenna selection did not transfer into the fast visibilities.

As shown in #408, the list of antenna names in the fast visibility data corresponds to correlator IDs 1-48 (not 0-47!). Is it possible that the cable delays were applied incorrectly due to the matching error of the antenna IDs? For example, LWA-273 corresponds to antenna ID #6 in the fast visibility, while it corresponds to antenna/correlator ID #7 in the slow visibility. As such, it may get the cable delay value from correlator ID #6 (LWA-276) instead?

jack-h, 2023-05-22:
> Hi Jack, I have checked the uv points in the fast visibility data, which
> perfectly match the expectations from the antenna XYZ locations. See some
> plots below:
>
> This figure is the antenna [...]

Unless a test interrogates the _data_ in the files, I don't think anything
can really be learned about what is or isn't being selected from the meta
data. Antenna labels, UVWs, and XYZs are all derived from the antenna
indexes sent from the correlator, which are always 1-48. Unless the data
recorders know the selected "fast" antennas, they can't possibly convert
these indices into the correct ones.

As shown in #408 <https://github.com/ovro-lwa/lwa-issues/issues/408>, the
> list of antenna names in the fast visibility data corresponds to correlator
> IDs 1-48 (not 0-47!).
This is because the LWA packet format 1-indexes the antennas, I think

Is it possible that the cable delays were applied incorrectly due to the
> matching error of the antenna IDs? For example, LWA-273 corresponds to
> antenna ID #6 <https://github.com/ovro-lwa/lwa-issues/issues/6> in the
> fast visibility, while it corresponds to [...]
Anything's possible, but since issue #408 identified that the meta data
sent to the data recorder is wrong (antennas are labelled 1-48 regardless
of fast selection), I think that issue alone makes all the observed results
consistent.

More generally though - if there are questions about data labelling, it is
always easier to identify them with test patterns rather than sky signal.

binchensun, 2023-05-22:
Thank you for your response, Jack. It makes sense to me. Let us do another test once the data labeling is fixed.

dgary50, 2023-05-22:
@jack-h You suggested a couple of times to run the system in "test mode."  We are happy to do so, but we would need someone to give us the details about how to put it into test mode and how to interpret any output.  Is this just a matter of setting a switch in the mnc control software somehow?

jack-h, 2023-05-22:
We should maybe discuss at tomorrow's meeting(?) - I'm not sure quite what
m&c exposes, but this is very easy to do from the lower-level code, and I
can send some examples tomorrow

jack-h, 2023-05-23:
I think this is what you might want to do via the low level interface. The interface deals in SNAP hostnames and per-SNAP zero-indexed signal numbers, so you'd have to use the configuration spreadsheet (of MNC library) to convert LWA antenna numbers into appropriate SNAP/ADC-ID pairs.

```
import numpy as np
from lwa_f import snap2_fengine

f = snap2_fengine.Snap2FengineEtcd('snap01')
# Turn on post FFT test vectors
f.eqtvg.tvg_enable()
# Write a test vector where input 5 of this board has the value 1 for all 8-bit samples in a spectra (this will be interpretted as 0 + 1j by the correlator
test_vec = np.zeros(f.eqtvg.n_chans, dtype='u1') # 8-bit uint
# initialize all vectors to zero
for i in range(64):
    f.eqtvg.write_stream_tvg(i, test_vec)
# Set input 5
test_vec[:] = 1
f.eqtvg.write_stream_tvg(5, test_vec)
```

There are some shortcut methods to set test vectors to a ramp varying with frequency (`eqtvg.write_freq_ramp()`) or a ramp constant in frequency but varying over the 64 inputs to a SNAP (`eqtvg.write_const_per_stream()`). See section 4.2.13 of the F-engine manual -- https://github.com/realtimeradio/caltech-lwa/blob/master/docs/build/latex/LWA352SNAP2F-Engine.pdf

dgary50, 2023-06-27:
This problem was traced to having incorrect antennas selected relative to the database metadata (and wrong delays, etc.).  Jayce made a fix to the antenna selection and now the antenna selection and metadata agree.  Based on Bin Chen's investigation of data taken on 2023 Jun 22, fast and slow visibilties now agree, so this issue is closed.

closed 2023-08-31 by jack-h