ovro-lwa/lwa-issues #410: t-engine services not starting up

marinanderson, 2023-05-23:
I got the following error message from the t-engine when trying to observe the HAARP COSMOS 1589 bounce this morning. Multiple restarts of the t-engine service had no effect.

> (casa) ubuntu@lwateng:~$ systemctl --user status dr-tengine
● dr-tengine.service - OVRO-LWA Voltage Beam T-Engine/Recorder
     Loaded: loaded (/home/ubuntu/.config/systemd/user/dr-tengine.service; disabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-05-23 11:50:34 UTC; 2min 55s ago
    Process: 2184895 ExecStartPre=/bin/bash -c [ -d /data/log ] || mkdir -p /data/log (code=exited, status=0/SUCCESS)
   Main PID: 2184896 (python3)
     CGroup: /user.slice/user-1001.slice/user@1001.service/dr-tengine.service
             └─2184896 /home/ubuntu/anaconda3//envs/casa/bin/python3 /home/ubuntu/ovro_data_recorder/dr_tengine.py --address 10.41.0.97 --port 21001 --cores 21,>

>May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/threading.py", line 864, in run
May 23 11:50:42 lwateng dr-tengine[2184896]:     self._target(*self._args, **self._kwargs)
May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/ovro_data_recorder/dr_tengine.py", line 124, in main
May 23 11:50:42 lwateng dr-tengine[2184896]:     status = capture.recv()
May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/packet_capture>
May 23 11:50:42 lwateng dr-tengine[2184896]:     _check(_bf.bfPacketCaptureRecv(self.obj, status))
May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/libbifrost.py">
May 23 11:50:42 lwateng dr-tengine[2184896]:     raise RuntimeError(status_str)
May 23 11:50:42 lwateng dr-tengine[2184896]: RuntimeError: b'BF_STATUS_INTERNAL_ERROR'
May 23 11:50:44 lwateng dr-tengine[2184896]: !! @ 1 33022915795681280 -> 647249149595353088000000  or  0  and  0  at  10

The log indicated an active file and the output of run_sdf.py reported that recording was successful:

`2023-05-23 12:05:40 [INFO   ] Record command succeeded: ({'sequence_id': '228c6154f96211edba9a3cecef613a2e', 'timestamp': 1684843540.0152204, 'status': 'success', 'response': {'filename': '/data/beam01/060087_120600000000228c615'}},)`

but no file was written. I collected simultaneous power beam data, which ran as expected.

jaycedowell, 2023-05-24:
Looking into with `journalctl --user --unit dr-tengine` I'm seeing what underlying problem caused those errors:
```
May 23 11:50:39 lwateng dr-tengine[2184896]: IP is: 1627400458 (10.41.0.97)
May 23 11:50:42 lwateng dr-tengine[2184896]: ring_impl.cpp:310 Condition failed: _sequence_time_tag_map.count(time_tag)==0
May 23 11:50:42 lwateng dr-tengine[2184896]: ring_impl.cpp:310 error 10: BF_STATUS_INVALID_ARGUMENT
May 23 11:50:42 lwateng dr-tengine[2184896]: ring.cpp:136 error 10: BF_STATUS_INVALID_ARGUMENT
May 23 11:50:42 lwateng dr-tengine[2184896]: packet_capture.cpp:406 internal error: BF_STATUS_INVALID_ARGUMENT
May 23 11:50:42 lwateng dr-tengine[2184896]: Exception in thread CaptureOp:
May 23 11:50:42 lwateng dr-tengine[2184896]: Traceback (most recent call last):
May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/threading.py", line 916, in _bootstrap_inner
May 23 11:50:42 lwateng dr-tengine[2184896]:     self.run()
May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/threading.py", line 864, in run
May 23 11:50:42 lwateng dr-tengine[2184896]:     self._target(*self._args, **self._kwargs)
May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/ovro_data_recorder/dr_tengine.py", line 124, in main
May 23 11:50:42 lwateng dr-tengine[2184896]:     status = capture.recv()
May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/packet_capture.py", line 88, in recv
May 23 11:50:42 lwateng dr-tengine[2184896]:     _check(_bf.bfPacketCaptureRecv(self.obj, status))
May 23 11:50:42 lwateng dr-tengine[2184896]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/libbifrost.py", line 111, in _check
May 23 11:50:42 lwateng dr-tengine[2184896]:     raise RuntimeError(status_str)
May 23 11:50:42 lwateng dr-tengine[2184896]: RuntimeError: b'BF_STATUS_INTERNAL_ERROR'
```
That `_sequence_time_tag_map.count(time_tag)==0` line suggests that there was some kind of setup problem in the F or X-engine side of things.  Something that would lead to the T-engine thinking that there were two different sequences of data coming in at the same time.  That could happen with the non-interleaved F-engine setup but I would think that would lead to a different set of symptoms.

jaycedowell, 2023-05-30:
I tried the T-engine today and everything seems ok.  I also tested a fix for https://github.com/lwa-project/ovro_data_recorder/issues/29 and that seems to work as well.

caseyjlaw, 2023-07-28:
This looks like it has been resolved. I'll close.

caseyjlaw, 2023-07-28:
Incidentally, I'd like to give the T-engine recorders a whirl. Is there a doc to describe running an observation? Has anyone used `mnc-python` for this?

closed 2023-07-28 by caseyjlaw

jaycedowell, 2023-07-28:
The best "documentation" is probably what is in [run_sdf.py](https://github.com/ovro-lwa/mnc_python/blob/main/scripts/run_sdf.py).  I think there is also a lingering issue between slow vis. and the voltage beam such that slow vis. needs to be stopped for the T-engine to capture reliably.

marinanderson, 2023-07-31:
I can add a doc (or a section to the system use recipe) describing how I have run the t-engine in the past.

caseyjlaw, 2023-08-15:
I can reproduce this problem when changing the x-engine configuration for mapping channels to GPUs. For example making the bottom two channel ranges lower like this:
```
<     10.41.0.65-10000: [ 272,  368]
<     10.41.0.18-10000: [ 368,  464]
---
>     10.41.0.65-10000: [ 560,  656]
>     10.41.0.18-10000: [ 656,  752]
```

... will produce the reported error in the t-engine.

I'm a bit confused, since I thought this observing setup was already run once before. Is this channel configuration (i.e., standard setup with one set lower) supported?

jaycedowell, 2023-08-15:
What does the rest of the configuration look like?

caseyjlaw, 2023-08-15:
https://github.com/ovro-lwa/mnc_python/blob/cjl/dev/config/lwa_config_calim_haarp.yaml

jaycedowell, 2023-08-15:
If I'm reading this correctly there is a frequency gap between the bottom two channel ranges and the next two.  The T-engine expects a continuous range of frequencies to be sent from the X-engines.

caseyjlaw, 2023-08-15:
Ugh, of course. I'll set up a file with contiguous subbands.