ovro-lwa/lwa-issues #413: Get 24-hour slow vis buffer back online

marinanderson, 2023-05-23:
- [ ] calibration cronjob, once per day
- [ ] automate daytime/nighttime settings changes
- [ ] testing with other obs modes

yupinghuang, 2023-05-24:
I was sorta trying this with the tempcal script.

But I wasn't sure what to do, say, when Cyg A transits, and the correlator isn't sending packets to calim (either because of a F engine reinit or someone shuts down the x engine or if someone restarts the X engine as part of their obsie).

So I think we need to:
- Tighten up the Controller API so that it's harder to accidentally stop slow viz data recording/etc
- Alert on slow viz data not flowing
and then maybe
- the calibration cron jobs wakes up ~30 min in advance, check that the system is in the right state (and if not, tries its best to fix things, ask data recorder to record etc, but also we need to careful about not interrupting schedule beamforming things).
- re the cron job -- it might be better to write a python script that runs forever (as a systemd service) and then calculates the next calibration time and wake up then so that we can work on a sidereal schedule. 
- Also the visibilities data recorder reports the last written timetag https://github.com/lwa-project/ovro_data_recorder/blob/main/dr_visibilities.py#L969 so that hopefully the calibration script doesn't have to `ls` all the time..

caseyjlaw, 2023-05-26:
See also #401

yupinghuang, 2023-06-26:
If you wo functions that uses `subprocess.check_output` to change the setting in https://github.com/ovro-lwa/mantaray/tree/main. It will have a cron-like feature that also works with LST and a real lightweight notify feature when things go wrong via https://ntfy.sh/ovro-lwa-manta-ray-says

...I can sort out the execution bit later.....but rn I'm just having a good time
```
⠀⢀⡀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀
    ⠀⠀⠹⣿⣷⣶⣤⣄⣀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀
    ⠀⠀⠀⠘⣿⣿⣿⣿⣿⣿⣿⣶⣦⣤⣀⠀⠀⠀⠀⢀⡴⠛⠁⠀⠀⠀⠀⠀⠀⠀
    ⠀⠀⠀⠀⠘⣿⣿⣿⣿⣿⣿⣿⡿⢿⣿⣿⣷⣶⣶⣿⣧⠀⠀⠀⠀⢀⠀⠀⠀⠀
    ⠀⠀⠀⠀⠀⠹⣿⣿⣿⣿⡇⢸⣇⠘⣿⣿⣿⣿⣿⣿⣿⣷⣤⣀⣠⠟⠀⠀⠀⠀
    ⠀⠀⠀⠀⠀⠀⢻⣿⣿⣿⣷⠈⢿⣦⣘⣿⣿⣿⣿⣿⣿⣿⣿⡟⠁⠀⠀⠀⠀⠀
    ⠀⠀⠀⠀⠀⠀⠘⣿⣿⣿⣿⣷⣄⣹⣿⣿⣿⣿⣿⣿⣿⣿⣿⣇⠀⠀⠀⠀⠀⠀
    ⠀⠀⠀⠀⠀⠀⠀⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⡟⠿⣿⣿⣿⣿⡄⠀⠀⠀⠀⠀
    ⠀⠀⠀⠀⠀⠀⠀⣿⣿⣿⣿⣿⣿⣿⣿⣿⣏⠙⢿⣦⣄⣉⣹⣿⣿⡀⠀⠀⠀⠀
    ⠀⠀⠀⠀⠀⠀⠀⢿⣿⣿⣿⣿⣿⣿⣿⣿⣿⣷⣤⣌⣉⣉⣿⣿⣿⣧⠀⠀⠀⠀
    ⠀⠀⠀⠀⠀⢀⣤⣴⠾⠟⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⣿⡄⠀⠀⠀
    ⠀⠀⠀⢀⣾⠛⠉⠀⠀⠀⠙⠛⠛⠛⠛⠛⠛⠿⠿⣿⣿⣿⣿⣿⣿⣿⣷⠀⠀⠀
    ⠀⠀⠀⣼⠃⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠈⠙⠛⠿⣿⣿⣿⣇⠀⠀
    ⠀⠀⢰⡇⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠉⠻⢿⡀⠀
    ⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠀⠁⠀
```

caseyjlaw, 2023-06-28:
Next steps:
- Casey and Yuping to turn on recorder and confirm quota is managed
- Someone to create bad ant list in standard location (use caltable location?)
- Deploy Yuping PR to get beamformer using bad ant list.

yupinghuang, 2023-07-18:
OK now that we have slow viz continuous running things are a little easier. Here's the list of todos ordered. The first thing in each category should get us 80% there.

- Flagging
- [ ] Add a bit in the calibration table generation script that looks for excessively large cal amplitudes (corresponding to antennas that don't correlate with the others) and flag them in the solution.
- [ ] A priori bad ants can go in `/home/pipeline/bad_ants_latest.txt` Maybe just merge Greg's bad ants list in there?
- [ ] There are some known bad ants (rotation etc), get them in there too. Maybe turn that into a yaml file or stick them in etcd.

This should produce solutions that are good enough for beamforming and maybe for viz.

- Calibration
- [ ] a scheduling thing that runs calibrator every day/few days with daytime setting
- [ ] a scheduling thing that switches settings in the middle of the night or when calibrator transits and calibrate for both daytime settings
- [ ] keep track of day and night time setting calibration tables and where the system is when a beamformer is run (for now pr

- Imaging
- [ ] the calibration script already makes an image
- [ ] I'll get https://github.com/ovro-lwa/distributed-pipeline to work next week.

binchensun, 2023-07-19:
@surajit4444mondal did some tests for the data taken on 2023/7/7, which were recorded with the same nighttime gain settings for both day and night. He derived calibration tables from the nighttime data and applied them to the daytime data. The image quality is very good. Unless anything changes - for example, if strong daytime RFIs come back - perhaps it is cleaner to hold the gain settings constant for now when we are developing the pipeline. It would be great, though, to have the functionality of a scheduled change to daytime settings planned.

binchensun, 2023-07-19:
Regarding slow visibility imaging, we are interested in developing a solar imaging pipeline that produces near-real-time images at a cadence of ~10 minutes or faster. We would adopt the flagging scheme and apply the calibration tables, so I guess no additional effort there. But we'd like to investigate how to quickly produce a set of say, 16 solar images (one image per subband) every 10 minutes or so, with acceptable quality. Parallel processing should be needed (and better with GPU). Fast disk I/O may also be helpful.

caseyjlaw, 2023-08-16:
calimhead now running backend services to run orca: 
https://github.com/ovro-lwa/distributed-pipeline/commit/422ecec9ee8a7fc4de8d9076f7ee252f979cc2dd

caseyjlaw, 2023-11-30:
Writing to lustre directly:
- Fast vis buffer up to 2 days.
- Slow vis buffer up to 7 days.