ovro-lwa/lwa-issues #414: make dashboard resilient to missing gpu server

caseyjlaw, 2023-05-24:
Correlator pipeline configuration raises a RuntimeError when a gpu server is down. We should get around that if we want dashboard to display monitor points etc, even when one gpu server is down.

yupinghuang, 2023-05-24:
Catching the RuntimeError https://github.com/realtimeradio/caltech-bifrost-dsp/blob/main/pipeline-control/lwa352_pipeline_control/lwa352_pipeline_control.py#L350 and downgrading it to a warning should be good.

And then a null check at the very beginning of any function x engine related.

caseyjlaw, 2023-05-24:
@jack-h Do you think the x-engine pipeline could issue a warning and pass `None` values for pipeline control?

jack-h, 2023-05-24:
Does this achieve what you want -- https://github.com/realtimeradio/caltech-bifrost-dsp/commit/c65125828d701ad3c9e1601fb96fdf60732bad1c

yupinghuang, 2023-05-24:
This is a dream

jack-h, 2023-05-24:
Just testing now that etcd is back and seems to "work", in the sense that this completes ok:
```
c = Lwa352CorrelatorControl(['lxdlwagpu0%d' % i for i in range(1,9)])
```
having ignored (admittedly, not quietly) gpu03 which is currently reprovisioning

yupinghuang, 2023-05-24:
did it log to the terminal?

jack-h, 2023-05-24:
I was running in ipython and get a bunch of :

```
In [1]: from lwa352_pipeline_control import Lwa352CorrelatorControl

In [2]: c = Lwa352CorrelatorControl(['lxdlwagpu0%d' % i for i in range(1,9)])
2023-05-24 19:07:22 [ERROR   ] host lxdlwagpu03 (pipeline None) failed to respond to etcd command!
2023-05-24 19:07:22 [ERROR   ] Control interface on host lxdlwagpu03 failed to respond to ping!
2023-05-24 19:07:22 [ERROR   ] lxdlwagpu03 pipeline 0 was unresponsive and will be ignored
Exception in thread etcd3_watch_7facf7e259e8:
Traceback (most recent call last):
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/etcd3/watch.py", line 126, in _run
    metadata=self._metadata)
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/grpc/_channel.py", line 1209, in __call__
    operationses, event_handler, self._context)
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/grpc/_channel.py", line 1303, in create
    operationses_and_tags, context)
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 474, in grpc._cython.cygrpc.Channel.integrated_call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 294, in grpc._cython.cygrpc._integrated_call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 218, in grpc._cython.cygrpc._call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 258, in grpc._cython.cygrpc._call
ValueError: Cannot invoke RPC: Channel closed!
```
I'm sure I can suppress these

yupinghuang, 2023-05-24:
This looks good actually. Information is good

closed 2023-05-25 by jack-h

caseyjlaw, 2023-05-25:
This is deployed and tested in development environment. Also added a feature to actually *show* the missing pipelines in red in go/no-go status.