ovro-lwa/lwa-issues #415: etcd failover

yupinghuang, 2023-05-24:


caseyjlaw, 2023-05-24:
Notes:
- initially the mcs server had a bad disk.
- later, the server got shut down by the high temperature monitoring service
- both failures shut us down for a while and caused considerable confusion. it is worth spending money to prevent this.
- Q: second server vs redundant disks?
- Latter is a good starting point. Just need to set up RAID1 (mirror) of OS and data store. Rick to check on this.
- Former is more complex, as we'd want a failover mechanism for services looking for data on the new mcs server. TBD.

rh-codebase, 2023-05-25:
All the backup servers in the world won't stop user errors. Following  workflows may have prevented the second shutdown. A sign is posted in the shelter to shutdown the overtemperature service when doing work inside the shelter although there may be a clause regarding judgment whether the work may cause higher than normal temps. Maybe the shutdown service needs to be disabled on every entry into the shelter.

better unit tests against the current system config hopefully would have caught the wrong threshold temperature setting tor the MCS host. This lower setting was installed when the host was re-provisionied after the bad disk. Discovering the cause of the MCS shutdown was fairly visible in the Grafana Dashboards. Bringing more people up to speed on using these may help diagnose issues sooner. 

last but not least was my error for not updating the threshold temp in the provisioning scripts for the mcs host. It's actually stored in the shutdown tarball so the deploy framework, when that gets up and running would have caught this since the latest version of the repo does have the correct threshold.

None of this is meant to replace OS mirrors and other nice HW redundancy measures but only to point out their limits.

rh-codebase, 2023-06-02:
changed title since this issue is more focused on the host which runs a bunch of services and etcd is just one of them.