ovro-lwa/lwa-issues #416: lwa x-eng fast config gets reset when followed by other configure_xengine commands

caseyjlaw, 2023-05-26:
@iveydavis Can you describe the way this goes wrong?
An assumption of the `configure_xengine` method is that the list of `recorders` is what should be configured. That may produce unintuitive behavior if issuing that command multiple times.
Ultimately, we'd like to get to the point where this method is not even used. Only start/stop of data recorders.

iveydavis, 2023-05-27:
The way I remember this error occurring is that if you configure for one of the datarecorders (whether that be fast or a beam I believe), if you try to configure for the slow visibilities at another time, the xengines conk out and I believe some failures crop up with the datarecorders. I'll need to double check the exact errors again.

jack-h, 2023-05-30:
A single correlator feeds both the fast and slow outputs, so configuring the slow after fast will likely upset things.
The beamformer is independent, so messing with the beamformer should affect the correlator outputs, and similarly messing with the correlator(s) shouldn't affect the beamformers

yupinghuang, 2023-12-01:
fixed by https://github.com/ovro-lwa/mnc_python/commit/dd6dcb17fd524da2f6a3fbc55641d74a364f31f3

closed 2023-12-01 by yupinghuang