ovro-lwa/lwa-issues #420: save beam calibration state

caseyjlaw, 2023-06-01:
The x-engine beamformer control code currently creates and calibrates at the same time. It should be possible to create a handle for an already calibrated beam, but the control code currently assumes it is uncalibrated.
For more efficient observing, we need to find a way store calibration state for a beam and check it when creating a control instance.

jaycedowell, 2023-06-01:
This is more of an x-engine state problem than something with the x-engine beamformer control module.  I had thought about this before and worked on https://github.com/realtimeradio/caltech-bifrost-dsp/compare/main...beamformer_cal_flag  The idea being that the x-engine beamformer control module could use the `cal_gains` monitoring point to determine if there were a sufficient number of calibrated inputs to declare a beam "working".

caseyjlaw, 2023-06-29:
This has been deployed to cal-im and the gpu servers. Could you check that it deployed properly? I needed to work around some deployment issues.

jaycedowell, 2023-06-29:
I'm on travel for the next couple of weeks but I will take a look at it when I get back.

caseyjlaw, 2023-07-17:
I see something called `cal_gains` at: `p.beamform.get_bifrost_status()['stats']['coeffs']`.
Is that where you expected it?

jaycedowell, 2023-07-17:
I would have thought that it would have been directly under 'stats' and called `cal_gains0`, `cal_gains1`, etc.  Each `cal_gainsN` should be a list with 704 True/False values.  I was going to take a closer look but it looks like I still can't connect to OVRO.

caseyjlaw, 2023-07-17:
Unfortunately, there are routing issues to OVRO currently. If you have an account to a server on Caltech campus, then you could hop in that way.

Here's more detail on the calgains field that happened to be in my terminal (under "new_coeffs" this time):
```
> st['stats']['new_coeffs']
"{'type': 'calgains', 'input_id': 15, 'beam_id': 1, 'data': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0]}"
```

I don't see anything with a different `beam_id`. Perhaps only beam 1 had calibration applied at this time?

jaycedowell, 2023-07-17:
Hrm.  That doesn't look like what I would expect.  Are there any `cal_gainN` keys in the full dictionary that `get_bifrost_status()` returns?

caseyjlaw, 2023-07-17:
I see these keys: `['pid', 'name', 'cmd', 'core', 'acquire', 'process', 'reserve', 'total', 'gbps', 'time', 'time_tag', 'sync_time', 'seq0', 'chan0', 'nchan', 'system_nchan', 'fs_hz', 'sfreq', 'bw_hz', 'nstand', 'pipeline_id', 'npol', 'complex', 'nbit', 'stats']`.

The full value dict returned by the status function:

```
{'pid': 8172,
 'name': 'Beamform',
 'cmd': '/home/ubuntu/anaconda3/envs/casa/bin/python /home/ubuntu/anaconda3/envs/casa/bin/lwa352-pipeline.py --nchan 96 --ibverbs --gpu 0 --pipelineid 0 --useetcd --etcdhost etcdv3service.sas.pvt --ip 10.41.0.66 --port 10000 --bufgbytes 0 --cores 1,2,3,4,4,4,4,4,4,4,4,4,4,4,4 --logfile /home/ubuntu/xpipeline.lxdlwagpu02.0.log ',
 'core': 4,
 'acquire': 0.0172731876373291,
 'process': 0.010779142379760742,
 'reserve': 2.3365020751953125e-05,
 'total': 0.028075695037841797,
 'gbps': 48.15272882697751,
 'time': 1689624954.3460753,
 'time_tag': 1,
 'sync_time': 0,
 'seq0': 40419545948160,
 'chan0': 752,
 'nchan': 96,
 'system_nchan': 3072,
 'fs_hz': 196000000,
 'sfreq': 17992187.5,
 'bw_hz': 2296875.0,
 'nstand': 352,
 'pipeline_id': 0,
 'npol': 2,
 'complex': 'True',
 'nbit': 4,
 'stats': {'curr_sample': 40425597049920,
  'update_pending': True,
  'last_cmd_proc_time': 1689372042.9714475,
  'new_coeffs': "{'type': 'calgains', 'input_id': 15, 'beam_id': 1, 'data': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0]}",
  'last_cmd_time': 1689372214.9723601,
  'last_cmd_response': 0,
  'coeffs': "{'type': 'calgains', 'input_id': 15, 'beam_id': 1, 'data': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0, 0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, -0.0, 0.0, 0.0]}"}}

```

jaycedowell, 2023-07-17:
Definitely not there.  Are we sure that the right software version is running on the x-engines?

caseyjlaw, 2023-07-17:
It looks like the new commit was not tagged for deployment. And... as it turns out, the networking issue makes software deployment fail, too. I fixed the former, but will need to wait on the latter.

caseyjlaw, 2023-07-18:
The networking issue is fixed, so you can log in. I've deployed the latest commit that includes this feature.
Unfortunately, I still don't see the new field in `stats` when running on lwacalim10 in `deployment` env. I guess there is some problem with how I'm deploying code, but I'm not seeing it.
Could you test this, @jaycedowell?

jaycedowell, 2023-07-18:
I don't see anything under the 'stats' key right now.

jaycedowell, 2023-07-18:
Doesn't look like the new version deployed.  On gpu01 If I look in `~/proj/lwa-shell/caltech-bifrost-dsp/pipeline/lwa352_pipeline/blocks` I see the updated `beamform_block.py`.  If I look in `~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline/blocks/` where I import `lwa352_pipeline` from the timestamp on `beamform_block.py` is from June 29.

caseyjlaw, 2023-07-18:
Ok, thanks. I think the deployment system isn't working correctly, so I'll do it manually.

caseyjlaw, 2023-07-18:
The `deploy` script does quite a lot, so I'd like to make it work. However, it seems to fail on the first part, which is `make install` on bifrost. Have you tested that recently?

jaycedowell, 2023-07-18:
No but I don't think anything in that fork of Bifrost has changed recently.

caseyjlaw, 2023-07-18:
I don't see a Makefile in the submodule.

jaycedowell, 2023-07-18:
Woah, it looks like the submodule has had a lot of stuff deleted:
```
$ git status
HEAD detached at 474d870
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	deleted:    .travis_deploy_docs.sh
	deleted:    Doxyfile
	deleted:    LICENSE
	deleted:    Makefile
	deleted:    NOTICE
	deleted:    ROADMAP.md
	deleted:    deploy_key.enc
	deleted:    docs/.nojekyll
	deleted:    docs/Dockerfile
	deleted:    docs/Makefile
	deleted:    docs/README.rst
	deleted:    docs/docker_build_docs.sh
	deleted:    docs/source/Common-installation-and-execution-problems.rst
	deleted:    docs/source/Cpp-Development.rst
	deleted:    docs/source/Create-a-pipeline.rst
	deleted:    docs/source/How-Python-and-C---fits-together.rst
	deleted:    docs/source/Python-API.rst
	deleted:    docs/source/bfmap.rst
	deleted:    docs/source/conf.py
	deleted:    docs/source/cpp_reference.rst
	deleted:    docs/source/index.rst
	deleted:    docs/source/intro.rst
	deleted:    docs/source/tools-intro.rst
	deleted:    docs/source/views.rst
	deleted:    docs/source/your-first-blocks.rst
	deleted:    python/bifrost/addon/leda/test/test_blocks.py
	deleted:    python/bifrost/blocks/accumulate.py
	deleted:    python/bifrost/blocks/audio.py
	deleted:    python/bifrost/blocks/convert_visibilities.py
	deleted:    python/bifrost/blocks/copy.py
	deleted:    python/bifrost/blocks/fdmt.py
	deleted:    python/bifrost/blocks/guppi_raw.py
	deleted:    python/bifrost/blocks/print_header.py
	deleted:    python/bifrost/blocks/psrdada.py
	deleted:    python/bifrost/blocks/quantize.py
	deleted:    python/bifrost/blocks/scrunch.py
	deleted:    python/bifrost/blocks/unpack.py
	deleted:    python/bifrost/header_standard.py
	deleted:    python/bifrost/temp_storage.py
	deleted:    python/bifrost/units.py
	deleted:    python/bifrost/views/__init__.py
	deleted:    python/testsrc/naming_convention.txt
	deleted:    src/.gitignore
	deleted:    src/ArrayIndexer.cuh
	deleted:    src/Complex.hpp
	deleted:    src/EnvVars.hpp
	deleted:    src/IndexArray.cuh
	deleted:    src/Jones.hpp
	deleted:    src/ObjectCache.hpp
	deleted:    src/ShapeIndexer.cuh
	deleted:    src/Vector.hpp
	deleted:    src/affinity.cpp
	deleted:    src/array.cpp
	deleted:    src/array_utils.hpp
	deleted:    src/assert.hpp
	deleted:    src/autodep.mk
	deleted:    src/bifrost/Common.hpp
	deleted:    src/bifrost/Ring.hpp
	deleted:    src/bifrost/affinity.h
	deleted:    src/bifrost/array.h
	deleted:    src/bifrost/common.h
	deleted:    src/bifrost/cuda.h
[... 69 lines cut ...]
	deleted:    test/benchmarks/performance_vs_serial/benchmarks_ben2.log.txt
	deleted:    test/benchmarks/performance_vs_serial/benchmarks_ben5.log.txt
	deleted:    test/benchmarks/performance_vs_serial/linear_fft_pipeline.py
	deleted:    test/benchmarks/performance_vs_serial/run_benchmark.sh
	deleted:    test/benchmarks/performance_vs_serial/skcuda_fft_pipeline.py
	deleted:    test/benchmarks/setup.py
	deleted:    test/test_accumulate.py
	deleted:    test/test_address.py
	deleted:    test/test_fdmt.py
	deleted:    test/test_guantize.py
	deleted:    test/test_gunpack.py
	deleted:    test/test_header_standard.py
	deleted:    test/test_ndarray.py
	deleted:    test/test_quantize.py
	deleted:    test/test_resizing.py
	deleted:    test/test_romein.py
	deleted:    test/test_scrunch.py
	deleted:    test/test_sigproc.py
	deleted:    test/test_unpack.py
	deleted:    testbench/README.md

no changes added to commit (use "git add" and/or "git commit -a")
```

Not good.

rh-codebase, 2023-07-18:
@caseyjlaw The deployment framework needs to be properly configured. By default it does not go into every repo and run 'deploy'. It uses a config file to instruct it which repos to run deploy so we can tune which software gets installed on which nodes. 

@jaycedowell Looks like the working tree is detached and just needs a checkout. Not sure which host and directory your output came from though. Try checking out the tip.

jaycedowell, 2023-07-18:
That `git status` was from ` ~/proj/lwa-shell/caltech-bifrost-dsp/bifrost` on gpu01.  I also see similar problems with the xgpu submodule in `~/proj/lwa-shell/caltech-bifrost-dsp/xgpu`.  I would have thought that all of the submodule init/updating should have been handled as part of deployment process because it's listed in `.mrconfig_production`.

rh-codebase, 2023-07-18:
Thanks for the addition info. I'm with you, myrepos should have handled it. I'm pretty sure it did with Jack's repo which also uses submodules.  Maybe this repo was whacked somehow earlier. Can you put it back to how it should be on gpu01? I can do a more localalized test regarding myreps and the deployment framework. This looks more like myrepos since that's all the deployment framework runs initially. Anyway. I can take a closer look.

jaycedowell, 2023-07-19:
I've restored the deleted files in the bifrost and xgpu submodules.

rh-codebase, 2023-07-19:
can't reproduce on lxdlwagpu01 in proj/lwa-shell using

 mr -c .mrconfig_production update

 maybe the repo was in this state before deployment ran. Recommend launching deployment from the webUI and then recheck this repo on gpu01. The deployment config is such that the deploy script is not run in caltech-bifrost-dsp so run/debug that script manually. After the deploy script runs successfully, the deployment config can be changed to deploy this repo.

caseyjlaw, 2023-07-20:
I see the field after pulling `main` and installing the x-engine pipelines and control with the latest commit. 
```
> st = p.beamform.get_bifrost_status()
> st['stats']['cal_gains0']
'[False, False, False, ...
```

caseyjlaw, 2023-07-22:
Sorry, but I think this isn't quite resolved yet.
I tried to set the `cal_gains3` stats to be `True`, but can't see it in another pipeline instance/session.
The standard pipeline control uses `p.beamform.update_calibration_gains` to set gains, which also sets the `cal_gains#` stats to True. If I load another session, I still see all `cal_gains#` beams antpols are `False`.

jaycedowell, 2023-07-22:
Set it to `True`?  It should be set on the x-engine side when the calibration is loaded.

caseyjlaw, 2023-07-22:
I meant to report that the x-engine is not setting it to `True`. I'm trying to set it by applying calibration. I'm reading it later and still seeing it as `False`.

jaycedowell, 2023-07-22:
Could it be an indexing error somewhere, i.e., is `cal_gains2` being updated instead?

caseyjlaw, 2023-07-22:
No, I confirmed that all beams and all antennas are still `False`.

jaycedowell, 2023-07-22:
Oh, yeah, I see it now.  There needs to be a:
```
self.update_stats({'cal_gains%d' % b: self.stats['cal_gains%d' % b]})
```
after [line 335 on beamform_block.py](https://github.com/realtimeradio/caltech-bifrost-dsp/blob/main/pipeline/lwa352_pipeline/blocks/beamform_block.py#L335C20-L335C49).  It might be better to put it at the end of the command update method,  after line 350, with a loop through all beams:
```
beam_state = {}
for b in range(nbeam):
    beam_state['cal_gains%d' % b] = self.stats['cal_gains%d' % b]
self.update_stats(beam_state)
```

jaycedowell, 2023-07-24:
I'd like to think that this is working under https://github.com/realtimeradio/caltech-bifrost-dsp/commit/740d7335450492737035f075dad88a7017e72a81.

caseyjlaw, 2023-07-27:
I'm seeing the cal_gains# field in the stats. After setting calibration as usual via `mnc-python` interface to the beamformer control code, I see this for a "calibrated" beam:

```
> st['stats']['cal_gains3']
'[False, True, False, True, False, True, False, True, False, True, False, True, False, True, False, True, False, True, False, True, False, True, False, True, False, True, False,...
```

There are 704 boolean values that alternate between True and False. Does that imply that only one pol is calibrated?

Also, I see that the `BeamPointingControl` has a `cal_set` property. Is that be connected to the state shown in cal_gains#?

jaycedowell, 2023-07-27:
> There are 704 boolean values that alternate between True and False. Does that imply that only one pol is calibrated?

Yes, that looks like only one pol. is calibrated.

> Also, I see that the BeamPointingControl has a cal_set property. Is that be connected to the state shown in cal_gains#?

Yes, there should be some mechanism in `BeamPointingControl.__init__` that uses this field to initialize the state of `_cal_set`.

caseyjlaw, 2023-07-31:
I see something a bit odd in the calibration state in the new cal_gains# field. The setup is that I restarted the x-engines and applied calibration to dr3 beamformer. I haven't looked at the tables, but the logging seems reasonable during application.
All cal_gains# fields are False, except for pipelines 4/5. They show complementary values:

```
> st['stats']['cal_gains4']
'[True, False, True, False,  [and so on, alternating....]
```

And on the next pipeline
```
> st['stats']['cal_gains5']
'[False, True, False, True,  [and so on, complementary alternating...]
```

I expected all to be True. Could there be an iteration but during application?
I see relevant code at https://github.com/ovro-lwa/mnc_python/blob/main/mnc/xengine_beamformer_control.py#L306.

jaycedowell, 2023-08-01:
The indexing values being fed into `push_gains` look ok.  I'm not familiar with asyncio but maybe some of the gain updates are getting dropped there?  Are there any warnings in the logs about failing to send commands?

jaycedowell, 2023-08-02:
I just realized what this is, it's a difference in what a "beam" is to observers and what a "beam" is for the X-engines.  For the X-engines a beam corresponds to a single polarization of what an observer would call a beam.  So observer beam 1 is X-engine beams 0 and 1, observer beam 2 is X-engine beams 2 and 3, etc.

Maybe the correct way to address this is to not change what happens in the X-engines but update how `xengine_beamformer_control.py` uses the stored state to determine the overall calibration state.  `xbf.py` might also need to explicitly set the unused inputs for that X-engine beam to zero so that it is clear that something has happened.

caseyjlaw, 2023-08-02:
For the first part of this solution, is it as simple as setting the "calibrated" threshold to compare against the status of the even/odd values for pipelines that are even/odd?
For example, around here: https://github.com/ovro-lwa/mnc_python/blob/main/mnc/xengine_beamformer_control.py#L146C36-L146C36.

jaycedowell, 2023-08-02:
That's similar to what I was working on in [jdd/cal_state_fixes](https://github.com/ovro-lwa/mnc_python/tree/jdd/cal_state_fixes).

caseyjlaw, 2023-08-10:
This is merged into main. I tested it in the development environment and confirmed that it works as expected.
Thanks for the help with this.

closed 2023-08-10 by caseyjlaw