ovro-lwa/lwa-issues #422: missing version number from latest fpga firmware?

yupinghuang, 2023-06-03:
```
Jun 03 06:50:13 lwacalim10 lwa_start_feng_[2321025]: 2023-06-03 06:50:13,881 - lwa_f.blocks.block:snap01 - WARNING -  - Software supports FW rev 7.3., but not 0.0.0.0
Jun 03 06:50:14 lwacalim10 lwa_start_feng_[2321025]: 2023-06-03 06:50:14,515 - lwa_f.blocks.block:snap01 - WARNING - pfb - Shift register is being overridden by firmware.
Jun 03 06:53:04 lwacalim10 lwa_start_feng_[2321025]: 2023-06-03 06:53:04,138 - lwa_f.blocks.block:snap01 - WARNING - pfb - Firmware implements some hardcoded shift stages. Setting shift to 0x1fff
Jun 03 18:12:53 lwacalim10 lwa_start_feng_[2321025]: 2023-06-03 18:12:53,153 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jun 03 18:12:53 lwacalim10 lwa_start_feng_[2321025]: 2023-06-03 18:12:53,185 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jun 03 18:12:53 lwacalim10 lwa_start_feng_[2321025]: 2023-06-03 18:12:53,217 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jun 03 18:12:53 lwacalim10 lwa_start_feng_[2321025]: 2023-06-03 18:12:53,249 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
```

dashboard says they're programmed with `snap2_f_200msps_64i_4096c_2023-04-26_1706.fpg`

jack-h, 2023-06-03:
> Assigned #422 <https://github.com/ovro-lwa/lwa-issues/issues/422> to
> @jack-h <https://github.com/jack-h>.

Not by a computer, but this is what would happen if the snaps haven't yet
been initialized (prior to which, all registers, including version, return
zero)

caseyjlaw, 2023-06-05:
I'll reprogram now.

caseyjlaw, 2023-06-05:
F-engines are back up and I'm planning some control code/doc updates.
However, I'm seeing an error on x-engine initialization. I can get an etcd control instance, but running `stop_pipelines` produces an error:
`2023-06-05 11:10:07,297 - mnc.control.Lwa352CorrelatorControl - ERROR - host lxdlwagpu01 (pipeline None) failed to respond to etcd command!`

Here's the log on the etcd service on gpu01:
```
Jun 05 11:08:25 lxdlwagpu01 python[20212]: 2023-06-05 11:08:25,225 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa3
Jun 05 11:08:25 lxdlwagpu01 python[20212]: 2023-06-05 11:08:25,228 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa3
Jun 05 11:08:25 lxdlwagpu01 python[20212]: 2023-06-05 11:08:25,231 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa3
Jun 05 11:08:25 lxdlwagpu01 python[20212]: 2023-06-05 11:08:25,232 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa3
Jun 05 11:08:29 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[20212]: terminate called after throwing an instance of 'Verbs::Error'
Jun 05 11:08:29 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[20212]:   what():  Failed to create queue pair: (1) Operation not permitted
Jun 05 11:08:29 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[20212]: terminate called after throwing an instance of 'Verbs::Error'
Jun 05 11:08:29 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[20212]:   what():  Failed to create queue pair: (1) Operation not permitted

```

jack-h, 2023-06-05:
This is an error I've seen in the past, which I thought the server
provisioning incorporated a fix for.

I ran `/etc/init.d/openibd restart` on all the GPU nodes. This should make
this error go away

caseyjlaw, 2023-06-05:
The logging on the GPU server looks ok, but the controller still fails with this (slightly different error):
`2023-06-05 12:39:58,907 - mnc.control.Lwa352CorrelatorControl - ERROR - host lxdlwagpu01 (pipeline 2) failed to respond to etcd command!`
Restarted the etcd control services with no effect. Using an `Lwa352CorrelatorControl` instance also does not work. It times out with this error:
`2023-06-05 12:46:17,770 - lwa352_pipeline_control.lwa352_pipeline_control - WARNING - Timeout waiting for pipelines to come up after 60.0 seconds`

jack-h, 2023-06-05:
Of the 4 pipelines on each server, numbers 2&3 are the ones with big ram
buffers. You'll need a timeout longer than 60 seconds (suggest 3 mins) for
these. I changed the default relatively recently (
https://github.com/realtimeradio/caltech-bifrost-dsp/blob/0fa9ef67bded838eb8b71912724a251353a1514c/pipeline-control/lwa352_pipeline_control/lwa352_pipeline_control.py#L90)
to try and account for this. Either you can specify your own timeout of
update the control library which should get you new defaults which work

caseyjlaw, 2023-06-06:
I updated the `development` environment to allow the timeout argument to pass to x-eng startup.
Restarting is working with longer timeout, although it seems that if a gpu server is out, the `start_pipeline` method times out without starting the correlator. Is that expected behavior?

jack-h, 2023-06-06:
`start_pipeline` doesn't know if a pipeline exists, so would indeed timeout
if a server is absent. But a pipeline should never be present in the
`CorrelatorControl.pipelines` unless it has responded to a ping