ovro-lwa/lwa-issues #424: calim node temperatures -- OK?

rh-codebase, 2023-06-05:
Grafana is now plotting data from the calim nodes. Are the temperatures expected.

Dashboard: Linux Hosts, Job: CalimHosts

yupinghuang, 2023-06-28:
I think for the most part they make sense? ones doing data recording are warmer than the ones not doing it. Although they're quite a bit warmer than I expected
[image]
[image]

And...what's going on with the head node?? Does it have two different sets of temperature sensors..?
[image]

rh-codebase, 2023-06-28:
Somewhere on the Grafana plot shows the critical temperature which is up around 90C IIRC. This value hopefully is coming back from the host so it seems that 60C, which does seem high, is well below critical.

head node:  does seem like a different sensor. Maybe the bios or some driver is mapping sensors differently into prometheus. That's a different matter so closing this out.

closed 2023-06-28 by rh-codebase