ovro-lwa/lwa-issues #425: Issue to request access to server

Deekshit-Vedula, 2023-06-07:
Just joined the team from Rice University. Need access to the server.

caseyjlaw, 2023-06-15:
@kembyct any progress on this?

kembyct, 2023-06-15:
Can I have the public key.
I got access to lxd110hxx machine and noticed that prometheus-node-exporter have been running on all the node for nine months.

Thanks

caseyjlaw, 2023-06-15:
@Deekshit-Vedula can you share your public key here?

Deekshit-Vedula, 2023-06-19:
Hi

My public key is: ssh-rsa
AAAAB3NzaC1yc2EAAAADAQABAAABgQDNg+AJYLurrwNNDMdEUopx2f+sFRpPKsP3SH3d0n5iyxXMys2EfkDfnkNL1F/hEY6dF3SR66RrkH9wC5or1vBqyPT6u5/mLWVHDPuwfT2HMVIqomB/KCxhnI1MPR1clLRmsCFP4Ur/4wGj5sO/VgwV4566cK7CN7lvDYZ5HgD/DmRK0X2+ho9tKlXSSKWuJaaMYrxOQI1VZ1kjOpnrszeb6mMmZzdwUXnTXatk+XAwWS3NmT+DtgzDNoviqE5PZrBMvUyNuahKpOxxoYTAGyg+IlJQ98nI/+UOJ35Lo0zwIbIVTy+Mrjrdzx8FONynT/HQxBWclKXc9rescaMNmA0NpMD1QUetA28RnzD0lBHTIsiEFuB6A7s3CWqOSobONlGelw/JVwjU4ePLnOw2+URFFMKhqnBYPw5w5cqLVZYzyhlIjhtz0WvvVA6d7EzxTNYsQV8RF61XlWANgM8yXpyPK0RQkdMCS7H+0h2DAz7XbqvjozEdgTNaG9E7GpVh17k=
***@***.***

Deekshit-Vedula, 2023-07-06:
Hey @kembyct. Any update on this?

kembyct, 2023-07-10:
Hello @Deekshit-Vedula 
I'm sorry I just noticed your notes. I wonder what your username on the system is. I wonder if you have an account on dana server, so I can use it.
Thanks.

Deekshit-Vedula, 2023-07-11:
Hi @kembyct 

I don’t have an account on the dana server.

Deekshit-Vedula, 2023-07-11:
sv50

Deekshit-Vedula, 2023-07-17:
Hi @kembyct. I need access to cal-im which requires an account on ssh.ovro.caltech.edu and a public key on lwamass

kembyct, 2023-07-17:
I think you are all set. Your username is "sv50" all across the system.

Deekshit-Vedula, 2023-07-17:
I can connect to ovro server but when i try to connect to lwamaas, I get an error saying "ubuntu@lwamaas.lwa.pvt: Permission denied (publickey)."

caseyjlaw, 2023-07-17:
I suspect you need to get ssh key forwarding working.
If you have sharepoint access, you can find some instructions on getting ssh to work at https://caltech.sharepoint.com/sites/ovro/projects/LWA%20Documents/Tutorials/Tutorial%20Links.docx?d=wa76338bc7cc243569d170ab933a92641&csf=1&web=1&e=Do39hy.

kembyct, 2023-07-18:
Can you try to connect one more time, and let know if it works.

Deekshit-Vedula, 2023-08-01:
Hi @kembyct, I am able to connect to the ovro gateway server but I don't my public key is on the lwamaas and calim00, because when I try to connect to them, it says "Permission Denied (public key)"

kembyct, 2023-08-01:
I think you should be able to connect.
You  gotta jump to lwamaas from ovro, then lwacalim. In that order. You can use ssh-agent, "ssh -YA hostname" to connect to different machine. Usually, We create users on lwacalimhead, which requires a different setup. Give it a shot and let me know.
Thanks

Deekshit-Vedula, 2023-08-03:
Got into lwamaas from ovro but then when I ssh into lwacalim00 from lwamaas it asks for a password

kembyct, 2023-08-03:
Done!
You should be able connect

Deekshit-Vedula, 2023-08-08:
Its asking me for a password

kembyct, 2023-08-08:
Did you try to ssh as root?
Can you copy and send exactly what displays on your screenwhenever you try to ssh.
Have try ssh -YA -vvv hostnname
Let me know

kembyct, 2023-08-08:
This ' ***@***.*** ' at end of your public key could be the problem. Maybe RedHat system is kicking you out because of that. If it persists, you' will need to generate another ssh key and send it to me, so we can move forward.

Deekshit-Vedula, 2023-08-09:
When I try to use the command ssh -YA -vvv sv50@lwacalim00.mcs.pvt I get the following

`OpenSSH_7.6p1 Ubuntu-4ubuntu0.5, OpenSSL 1.0.2n  7 Dec 2017
debug1: Reading configuration data /home/ubuntu/.ssh/config
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug2: resolving "lwacalim00.mcs.pvt" port 22
debug2: ssh_connect_direct: needpriv 0
debug1: Connecting to lwacalim00.mcs.pvt [10.40.0.20] port 22.
debug1: Connection established.
debug1: identity file /home/ubuntu/.ssh/id_rsa type 0
debug1: key_load_public: No such file or directory
debug1: identity file /home/ubuntu/.ssh/id_rsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/ubuntu/.ssh/id_dsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/ubuntu/.ssh/id_dsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/ubuntu/.ssh/id_ecdsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/ubuntu/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/ubuntu/.ssh/id_ed25519 type 3
debug1: key_load_public: No such file or directory
debug1: identity file /home/ubuntu/.ssh/id_ed25519-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_7.6p1 Ubuntu-4ubuntu0.5
debug1: Remote protocol version 2.0, remote software version OpenSSH_8.0
debug1: match: OpenSSH_8.0 pat OpenSSH* compat 0x04000000
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to lwacalim00.mcs.pvt:22 as 'sv50'
debug3: hostkeys_foreach: reading file "/home/ubuntu/.ssh/known_hosts"
debug3: record_hostkey: found key type ECDSA in file /home/ubuntu/.ssh/known_hosts:206
debug3: load_hostkeys: loaded 1 keys from lwacalim00.mcs.pvt
debug3: order_hostkeyalgs: prefer hostkeyalgs: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521
debug3: send packet: type 20
debug1: SSH2_MSG_KEXINIT sent
debug3: receive packet: type 20
debug1: SSH2_MSG_KEXINIT received
debug2: local client KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group16-sha512,diffie-hellman-group18-sha512,diffie-hellman-group-exchange-sha1,diffie-hellman-group14-sha256,diffie-hellman-group14-sha1,ext-info-c
debug2: host key algorithms: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,ssh-ed25519-cert-v01@openssh.com,ssh-rsa-cert-v01@openssh.com,ssh-ed25519,rsa-sha2-512,rsa-sha2-256,ssh-rsa
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: MACs ctos: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: MACs stoc: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: compression ctos: none,zlib@openssh.com,zlib
debug2: compression stoc: none,zlib@openssh.com,zlib
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug2: peer server KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group14-sha256,diffie-hellman-group16-sha512,diffie-hellman-group18-sha512,diffie-hellman-group-exchange-sha1,diffie-hellman-group14-sha1
debug2: host key algorithms: rsa-sha2-512,rsa-sha2-256,ssh-rsa,ecdsa-sha2-nistp256,ssh-ed25519
debug2: ciphers ctos: aes256-gcm@openssh.com,chacha20-poly1305@openssh.com,aes256-ctr,aes256-cbc,aes128-gcm@openssh.com,aes128-ctr,aes128-cbc
debug2: ciphers stoc: aes256-gcm@openssh.com,chacha20-poly1305@openssh.com,aes256-ctr,aes256-cbc,aes128-gcm@openssh.com,aes128-ctr,aes128-cbc
debug2: MACs ctos: hmac-sha2-256-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha2-256,hmac-sha1,umac-128@openssh.com,hmac-sha2-512
debug2: MACs stoc: hmac-sha2-256-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha2-256,hmac-sha1,umac-128@openssh.com,hmac-sha2-512
debug2: compression ctos: none,zlib@openssh.com
debug2: compression stoc: none,zlib@openssh.com
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ecdsa-sha2-nistp256
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug3: send packet: type 30
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug3: receive packet: type 31
[... 60 lines cut ...]
[... cut ...]
debug1: Offering public key: ED25519 SHA256:8k8fyFaTRLc8xrIhQh+kXTdCukWpDtIlMakDLuWYuwE /home/ubuntu/.ssh/id_ed25519
debug3: send_pubkey_test
debug3: send packet: type 50
debug2: we sent a publickey packet, wait for reply
debug3: receive packet: type 51
debug1: Authentications that can continue: publickey,gssapi-keyex,gssapi-with-mic,password
debug2: we did not send a packet, disable method
debug3: authmethod_lookup password
debug3: remaining preferred: ,password
debug3: authmethod_is_enabled password
debug1: Next authentication method: password
sv50@lwacalim00.mcs.pvt's password: 
`

yupinghuang, 2023-08-09:
I don't know what happened but on calim00 I see

```
   0 drwx------.  4 deekshitv  deekshitv   120 Jul 11 17:02 sv50
```

and in `/var/log/secure` I see "invalid user" for `sv50`, which would make sense.

and then on calimhead I see.
```
   0 drwx------.  4 sv50       deekshitv   120 Jul 11 17:02 sv50
```

@kembyct how did you create the user on calim?

kembyct, 2023-08-09:
I usually just run the script on lwacalimhead, then I append the user's public keys to the ~/.ssh/authorized_keys. But I think I had to modified the username to sv50 upon his request running "usermod" command.

yupinghuang, 2023-08-09:
Ok I’ll take a look tomorrow. Should be just running a groupmod and then
propagate the changes across NIS

kembyct, 2023-08-09:
Or you can just run "chown -R sv50:sv50 /home/sv50/.ssh/" to fix everything.

yupinghuang, 2023-08-09:
@kembyct thanks for mod'ing the group name -- that was mostly an aesthetic thing.

I ran `sudo make -C /var/yp` so that the changes propagate across the nodes.

Deekshit-Vedula, 2023-08-09:
Worked! Thank you so much @kembyct and @yupinghuang

closed 2023-08-21 by kembyct