ovro-lwa/lwa-issues #426: lwa352_start_xeng_etcd_service.py matplotlib error...?

yupinghuang, 2023-06-12:
```
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]: Traceback (most recent call last):
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/bin/lwa352-pipeline.py", line 329, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     sys.exit(main(sys.argv))
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/bin/lwa352-pipeline.py", line 324, in main
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     build_pipeline(args)
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/bin/lwa352-pipeline.py", line 38, in build_pipeline
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from bifrost.address import Address
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/__init__.py", line 41, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from bifrost import views
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/views/__init__.py", line 28, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from .basic_views import astype, split_axis, merge_axes, rename_axis, \
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/views/basic_views.py", line 32, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from bifrost.units import convert_units
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/units.py", line 28, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     import pint
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pint/__init__.py", line 14, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from .context import Context
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pint/context.py", line 15, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from .definitions import Definition, UnitDefinition
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pint/definitions.py", line 13, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from .converters import LogarithmicConverter, OffsetConverter, ScaleConverter
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pint/converters.py", line 12, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from .compat import HAS_NUMPY, exp, log  # noqa: F401
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pint/compat.py", line 155, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from pandas import Series
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/__init__.py", line 55, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from pandas.core.api import (
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/core/api.py", line 24, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from pandas.core.groupby import Grouper, NamedAgg
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/core/groupby/__init__.py", line 1, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from pandas.core.groupby.generic import (  # noqa: F401
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/core/groupby/generic.py", line 44, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from pandas.core.frame import DataFrame
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/core/frame.py", line 115, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from pandas.core.series import Series
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/core/series.py", line 84, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     import pandas.plotting
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/plotting/__init__.py", line 59, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from pandas.plotting._core import (
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/plotting/_core.py", line 17, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     import pandas.plotting._matplotlib  # noqa
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/plotting/_matplotlib/__init__.py", line 3, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from pandas.plotting._matplotlib.boxplot import (
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/pandas/plotting/_matplotlib/boxplot.py", line 4, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     from matplotlib.artist import setp
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/matplotlib/__init__.py", line 1162, in <module>
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     rcParams = rc_params()
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/matplotlib/__init__.py", line 996, in rc_params
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     fname = matplotlib_fname()
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/matplotlib/__init__.py", line 840, in matplotlib_fname
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     for fname in gen_candidates():
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/matplotlib/__init__.py", line 829, in gen_candidates
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]:     yield os.path.join(six.moves.getcwd(), 'matplotlibrc')
Jun 12 23:46:43 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[24926]: FileNotFoundError: [Errno 2] No such file or directory
```

jack-h, 2023-06-13:
my word

caseyjlaw, 2023-06-13:
Is this on an gpu server in `casa` env?
If I go there, I am able to import some of these functions without an issue.

yupinghuang, 2023-06-13:
> Looking upstream in the logs, there is an early sign that os.getcwd() is
> returning No such file or directory, which is odd. I'm led to believe
> this only happens if the directory a script is launched from is
> subsequently deleted or renamed.
> I added some debug stuff to the lwa352_start_xeng_etcd_service.py script,
> but as soon as I restarted the process I couldn't ever recreate this issue.
>
> Unsatisfactory close?

Does swdeploy deploy to gpu nodes?

caseyjlaw, 2023-06-15:
I'll close.
Incidentally, swdeploy is working to gpu servers now (#296).

caseyjlaw, 2023-06-29:
@jack-h It seems that this is error is back after I used the webUI to deploy software to the gpu servers.
I installed manually with `pip install .` in the pipeline directory. After that, and restarting services on each gpu server, the error goes away.
Could you take a look at what's happening with the deploy script?

rh-codebase, 2023-06-29:
How is Jack's deploy script being fired off? The deploy framework is not running it.

caseyjlaw, 2023-06-29:
The repo has a `deploy` script, but running it manually failed. Likely that even a working deploy framework would fail too. How do I know if the deploy framework is trying to execute the script in the repo?
I thought the installation changed after I deployed via the webUI, but perhaps not. Sorry for the confusion.

rh-codebase, 2023-06-29:
the deploy framework is working as configured as far as the logs show. All my services log to /var/log/syslog. Each deploy service reads a config file which determines which repos to run the deploy script in. I have not updated that config since I'm still in debug mode. For GPU hosts, only lwa-pyutils and swdeploy are the only repos the deploy framework is running deploy in. The logs indicate this yet Jack states his deploy script is executed so I'm still trying to reproduce that statement without success.

closed 2023-08-23 by caseyjlaw