ovro-lwa/lwa-issues #429: Log each time restoreSettings.sh is run

marinanderson, 2023-06-13:
In addition to tracking settings changes in etcd, `./scripts/restoreSettings.sh` should output to a log file every time it is run, as a another method for tracking changes to system settings.

caseyjlaw, 2023-06-15:
Larry implemented this by writing to a file at `/home/pipeline/proj/lwa-shell/mnc_python/data/arxAndF-settings.log`.
Example line:
```
Wed Jun 14 18:53:09 2023	1686768789.4905016	ldaddario	20230608-settingsAll-night.mat	1686346303.0589998
```

An improvement would be to set these same values to an etcd key. A proposal:
- key name "/cmd/settings",
- values of "mjd", "user", "filename".

This could also be displayed on the dashboard and ingested to influx/grafana.
Thoughts @rh-codebase?

caseyjlaw, 2023-10-12:
`mnc-python` has a `settings.py` module that can set/get the settings. I'll close this, but we could still implement an etcd cfg key for this.

closed 2023-10-12 by caseyjlaw