ovro-lwa/lwa-issues #430: pythone-etcd3 callback is leaking threads

yupinghuang, 2023-06-15:
# Here's a thread count experiment.

**All the numbers should be 1 (the main thread) since all commands succeeded.** And I did try to `join()` the threads and they would not join.

```
(deployment) [yuping@lwacalim10 ~]$ cat experiment.py 
from mnc.control import Controller
import threading
import gc
from mnc import mcs
import time
import yaml

from lwa352_pipeline_control import Lwa352CorrelatorControl

def con():
    con = Controller()

def con_plus_mcs():
    con = Controller()
    con.start_dr('drvs')

def beep_xeng():
    with open('/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim.yaml', 'r') as fh:
        conf = yaml.load(fh, Loader=yaml.CSafeLoader)
    p = Lwa352CorrelatorControl(conf["xengines"]["xhosts"], npipeline_per_host=conf["xengines"]["nxpipeline"], etcdhost=conf["etcd"]["host"])
   # send an extra two commands just for fun
    p.pipelines[0].check_connection()
    p.pipelines[-1].check_connection()

if __name__=='__main__':
    gc.disable()
    beep_xeng() # switch out for each of the above function
    print(threading.active_count())
    gc.collect()
    time.sleep(5)
    print(threading.active_count())
```

And then for each function the following happens (numbers are pre and post gc thread counts):
```
(deployment) [yuping@lwacalim10 ~]$ python experiment.py # running con()
Read antpos from xlsx file in repo
2023-06-15 02:09:15,479 - mnc.control - INFO - Loaded configuration for 8 x-engine host(s) running 4 pipeline(s) each
2023-06-15 02:09:15,480 - mnc.control - INFO - Supported recorder modes are: slow,fast,beam1
2023-06-15 02:09:15,480 - mnc.control - INFO - etcd server being used is: etcdv3service
97
97
```
Controller has 97 threads left after gc. Not good.

controller plus using mcs.client to send a command bumps up the thread count to 99, gc'ing forces mcs. Client to close the gRPC channel which gives the "you pulled the rug from under my feet" error (which, is fine and probably indicates that there's no deadlocking issues) and then the thread counts went back down to 97.
```
(deployment) [yuping@lwacalim10 ~]$ python experiment.py  # con_plus_mcs
Read antpos from xlsx file in repo
2023-06-15 02:11:16,647 - mnc.control - INFO - Loaded configuration for 8 x-engine host(s) running 4 pipeline(s) each
2023-06-15 02:11:16,647 - mnc.control - INFO - Supported recorder modes are: slow,fast,beam1
2023-06-15 02:11:16,647 - mnc.control - INFO - etcd server being used is: etcdv3service
2023-06-15 02:11:16,648 - mnc.control - INFO - Starting recorders ['drvs'] at 60110.09116490779 (currently 60110.09116491611)
2023-06-15 02:11:16,834 - mnc.control - INFO - recording on drvs
99
Exception in thread etcd3_watch_7fdd58435320:
Traceback (most recent call last):
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/etcd3/watch.py", line 126, in _run
    metadata=self._metadata)
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/grpc/_channel.py", line 1209, in __call__
    operationses, event_handler, self._context)
  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/grpc/_channel.py", line 1303, in create
    operationses_and_tags, context)
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 474, in grpc._cython.cygrpc.Channel.integrated_call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 294, in grpc._cython.cygrpc._integrated_call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 218, in grpc._cython.cygrpc._call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 258, in grpc._cython.cygrpc._call
ValueError: Cannot invoke RPC: Channel closed!

97
```
and here's the xengine calls (with sending an additional two commands) -- note the the additional commands didn't increase the eventual thread count. 
```
(deployment) [yuping@lwacalim10 ~]$ python experiment.py #beep_xeng
97
# a bunch of debug outputs
97
```
# This hurts my head
Callbacks are implemented as threads in `python-etcd3`. So it looks like 2-3 threads never terminate for each etcd instance that does a callback. My best theory rn is that python decided that some callback functions are alien (i.e. numpy and gRPC extensions that release the GIL) because of of the scopes of the things in the callback function (being closures, being member functions, etc., referencing to `self` is....a fun thought experiment) so it would not know if they ended. But that doesn't quite explain why threads for additional commands do terminate just fine.

# Here's a workaround
close the etcd connection and be ok with lots of RPC channel closed errors...they come from threads that are not supposed to be alive anyway.

# oh and here's something that would be horrible
`multiprocessing` which forks the process with an etcd client in it and then do callbacks.

caseyjlaw, 2023-06-15:
Thanks for digging in on this. The traceback is a bit unclear, so can you confirm that this is all mediated by the `dsa_store` module (https://github.com/dsa110/dsa110-pyutils/blob/master/dsautils/dsa_store.py)?
If so, is that the right layer of code to implement your suggested workaround?

jack-h, 2023-06-15:
I _think_ the summary is that etcd3 doesn't close threads out when cancelling a callback watch, but, it also only creates threads when the first watch is instantiated (hence the lack of thread growth when repeatedly issuing commands)

```
In [1]: import threading

In [2]: import etcd3

In [3]: ecs = []

In [4]: watches = []

In [5]: def fake_watch(blah):
   ...:     print("Shouldn't actually trigger")
   ...:     

In [6]: threading.active_count()
Out[6]: 2

In [7]: for i in range(10):
   ...:     ecs += [etcd3.Etcd3Client('etcdv3service')]
   ...:     

In [8]: threading.active_count()
Out[8]: 2

In [9]: ec0 = ecs[0]

In [10]: for i in range(10):
    ...:     watches += [ec0.add_watch_callback('fake_key', fake_watch)]
    ...:     

In [11]: threading.active_count()
Out[11]: 5

In [12]: for i in range(10):
    ...:     watches += [ec0.add_watch_callback('fake_key', fake_watch)]
    ...:     

In [13]: threading.active_count()
Out[13]: 5

In [14]: for i in range(10):
    ...:     ec0.cancel_watch(watches[i])
    ...:     
    ...:     

In [15]: threading.active_count()
Out[15]: 5

In [16]: threading.active_count()
Out[16]: 5

In [17]: import gc

In [18]: gc.collect()
Out[18]: 28

In [19]: threading.active_count()
Out[19]: 5

In [20]: threading.active_count()
Out[20]: 5

In [21]: threading.active_count()
Out[21]: 5

In [22]: for i in range(10):
    ...:     watches += [ec0.add_watch_callback('fake_key', fake_watch)]
    ...:     

In [23]: threading.active_count()
Out[23]: 5

In [25]: for i in range(10):
    ...:     watches += [ecs[i].add_watch_callback('fake_key', fake_watch)] # one watch per ETCD connection

In [26]: threading.active_count()
Out[26]: 32
```

I can either:
1. Make  `send_command` in lwa352_pipeline_control.etcd_control.EtcdControl` close and restart the entire etcd connection (I think this is @yupinghuang 's suggestion)
2. Make `CorrControl` share one etcd interface -- this doesn't really address the issue, but should reduce the number of threads by the total number of pipelines (32), since currently every pipeline controller spawns its own interface client

jack-h, 2023-06-15:
Skimming the code, it looks like the behaviour is intended -- the first instantiation of a callback starts a callback thread -- https://github.com/kragniz/python-etcd3/blob/e58a899579ba416449c4e225b61f039457c8072a/etcd3/watch.py#L83 -- which is never stopped.

So, my question is, it the system leaking threads, or just spawning more threads than is desirable? I'm going to implement my modification (2) above, which should at least make behaviour more reasonable

yupinghuang, 2023-06-15:
Sorry for the confusions. In my example, since all the objects were defined in functions. They should be garbage collected after the functions run and `gc.collect()` is called (which is what python automatically does anyway). And if the threads these objects own do not terminate and there's no way to access them (because we don't have references to the objects), then that's leakage.

I think that's what's happening with https://github.com/ovro-lwa/lwa-issues/issues/361 and https://github.com/lwa-project/ovro_data_recorder/issues/31 where long-running processes that send commands ended up using up a lot of CPUs. Closing the etcd clients in each case did seem to resolve the issues. The funny thing is that they are not supposed to use up more than 100% of CPU anyway because of the GIL.

So the takeaway here is to

- be careful with python-etcd callbacks
- watch out for locally scoped objects that do callbacks (they're kinda everywhere in `control.Controller` but I think they're probably best handled at the subsystem controls level, maybe with a context manager)
- I don't like this python wrapper...
- It'd still be good to understand this issue. Calling `etcd.close()` on mnc client/pipeline control destruction is a good starting point but that would spam the stderr with Exceptions.

@caseyjlaw it does not seem like  dsastore would handle this issue. If a dsastore object gets gc'd after using it for callbacks, the threads will remain (the codes are basically the same as in mcs.Client). But this needs to be tested and confirmed and semantics for cleaning up made clear to users.

@jack-h the code snippet you shared...I didn't see an inifinite loop in `_run` so it seemed like the thread should have terminated. But in any case I don't think having threads that run forever and the only way to terminate them is to have them exception out is an intended behavior.

I was not suggesting that `send_command` calls `etcdclient.close()`. What I would suggest is that `CorrInterface` implements a destructor that calls `etcdclient.close()` and give `Lwa352CorrelatorControl` a destructor that destructs its child control objects (because things like someone referencing the `pipelines` attribute may prevent GC from kicking in).

"it the system leaking threads, or just spawning more threads than is desirable" I don't mind having a few hundred threads (especially if it's pure python threads) for a few hours. But if they are non-GIL'd threads that keep running for months and there's no way to reference them in the code, I'd count that as leaking.

sharing an ectd client would be an excellent move if the shared etcd client object is garbaged collected correctly (which it probably is) and if the etcd client takes care of closing its connection on destruction (which it does not). It does mitigate the issues. And if some user (e.g. dashboard) keeps creating new locally scoped pipelinecontrol objects (or even the controller object)...that can still be problematic.

yupinghuang, 2023-06-15:
ok this is just also kinda fun. I'm not saying that anything needs to be done atm...

jack-h, 2023-06-15:
I clearly didn't read you message properly, because I implemented the destructors and am now drowning in fake exceptions as you said would happen.

jack-h, 2023-06-15:
Looks like maybe this was fixed ant then never released to PyPi

https://github.com/kragniz/python-etcd3/commit/eb2ecd8a47c7375b6c40ad5dd60bbb11995de001

caseyjlaw, 2023-06-16:
I see a few issues in the python client asking for a new release.
We might want to build from source, rather than wait.
https://github.com/kragniz/python-etcd3/issues/2197#issue-1755842894

caseyjlaw, 2023-08-07:
@jack-h Did you have any ideas for reducing or catching the exceptions from the threads?

jack-h, 2023-08-07:
As far as I could tell in my brief playing around they couldn't be caught except by messing with the client code (at which point we may as well just use the latest source in github rather than waiting on pypi

caseyjlaw, 2023-08-10:
Incidentally, I ran into this error when playing with the scheduler, because I'm forking processes with python etcd clients:
`Other threads are currently calling into gRPC, skipping fork() handlers`

Setting `GRPC_ENABLE_FORK_SUPPORT=true` resolves it for our version of the python etcd client.

caseyjlaw, 2023-12-14:
Are we ok closing this issue with a workaround described above?

jaycedowell, 2023-12-14:
I'm ok with it but I will note that setting `GRPC_ENABLE_FORK_SUPPORT=true` was a disaster for the data recorders.

caseyjlaw, 2023-12-14:
An alternative would be to rebuild the python client (see #458).

caseyjlaw, 2024-01-22:
I'd like to revive this topic for discussion.
It seems that `GRPC_ENABLE_FORK_SUPPORT=true` environment variable is bad for the data recorders. However, I'm finding that I *require* it for the way I've configured the scheduler. I am using the python `ProcessPooolExecutor` to create a processes to issue commands for data recorders.
Somehow I thought that was working fine, but now I'm seeing that the scheduler is not issuing any beam tracking commands. The tracking in particular does require some async execution, because it is run as a loop that blocks the process during tracking.
Do we actually believe that rebuilding etcd3 from source will resolve this and remove the need for the GRPC env variable?

yupinghuang, 2024-01-24:
- You can set different env variables for different processes?
- Have you tried https://docs.python.org/3.6/library/multiprocessing.html#multiprocessing.set_start_method to spawn?
- lmk what the symptoms look like? I am curious what `GRPC_ENABLE_FORK_SUPPORT` has to do with what you're seeing.

Gonna note that `GRPC_ENABLE_FORK_SUPPORT=true` does not help with the thread leaking issue. It just stops the cascade of error messages when a process stop.

caseyjlaw, 2024-01-24:
Thanks for the ideas.
- I think I need one env for both the schedule management and communication to etcd. That seems to introduce this incompatibility on the GRPC?
- What alternatives to spawning would you recommend?
- Symptoms:

I can reproduce a problem when issuing tracking commands in the scheduler. First, I execute the commands from ipython with the `GRPC_ENABLE_FORK_SUPPORT=true` env variable set. Then I use a `ProcessPoolExecutor` to create a pool to submit commands.
I misstated where the failure occurs. It actually happens when controlling the data recorders via etcd. So if I submit a command, I get an etcd related error:
```
> con.start_dr(...)
Exception in thread etcd3_watch_7fb571a39ba8:
Traceback (most recent call last):
  File "/opt/devel/pipeline/envs/development/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/opt/devel/pipeline/envs/development/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/etcd3/watch.py", line 126, in _run
    metadata=self._metadata)
  File "/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/grpc/_channel.py", line 1209, in __call__
    operationses, event_handler, self._context)
  File "/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/grpc/_channel.py", line 1303, in create
    operationses_and_tags, context)
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 474, in grpc._cython.cygrpc.Channel.integrated_call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 294, in grpc._cython.cygrpc._integrated_call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 218, in grpc._cython.cygrpc._call
  File "src/python/grpcio/grpc/_cython/_cygrpc/channel.pyx.pxi", line 258, in grpc._cython.cygrpc._call
ValueError: Cannot invoke RPC: Channel closed due to fork
```

And if I check on the future returned by the pool, I get a warning from the mnc-python code:
```
/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/mnc/mcs.py:694: RuntimeWarning: Error sending command to '/cmd/dr3/record': timeout
  warnings.warn("Error sending command to '/cmd/%s/%s': %s" % (subsystem, command, str(e)), RuntimeWarning)
2024-01-22 21:14:58,318 - mnc.control - WARNING - no response from dr3
/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/mnc/mcs.py:476: RuntimeWarning: Error reading monitor point '/mon/dr3/summary': Cannot invoke RPC on closed channel!
  warnings.warn("Error reading monitor point '/mon/%s/%s': %s" % (id, name, str(e)), RuntimeWarning)
```

yupinghuang, 2024-01-25:
- re the envs, can you use `Environment=` in your systemd service file? Or just stick it in front of the executable if you're doing it manually.
 - Python 3.6's multiprocessing pool forks by default. See if changing it to spawn helps. Also, if you instantiate your ectd instances post pool creation (i.e. post fork/spawn), that might work better.

I'll talk to you tmr. Meanwhile, how are you "submitting" the `con.start_dr` command through the pool in your example?

caseyjlaw, 2024-01-25:
Ok, I have a solution to my problem. It is also not related to the issue in this thread.

To return to the OP, @rh-codebase, we need to address these issues in how we use etcd (in python):
- We'd like to support long-running processes, but thread leaking is a problem. How can we fix that?
- What is the expected behavior when forking a process in python? If not supported, we should let people using etcd clients know that.
- python-etcd3 client is not being updated. What are we going to do about that?
- Code creating python clients creates redundant grpc connection clients. We should control that in the client creation code.

These are long term concerns, but we need to get a handle on them if we're going all in with etcd.

rh-codebase, 2024-01-29:
some info on fork and threading:
https://discuss.python.org/t/concerns-regarding-deprecation-of-fork-with-alive-threads/33555

@casey " I am using the python ProcessPooolExecutor to create a processes to issue commands for data recorders."  Can you give a bit more insight as to why(for architecture, not why for sending commands)? When I think of parallelism or concurrency I think of architectures trying to solve CPU bound or I/O bound problems. Does sending commands to DR's fit into one of these categories?