ovro-lwa/lwa-issues #431: lxdlwagpu01 and 03 in a bad state

caseyjlaw, 2023-06-19:
I can't ssh into lxdlwagpu01 and 03. It has been like this for a few days.

rh-codebase, 2023-06-20:
console screen on lxdlwagpu01 shows lots of processes including sshd being killed due to out of memory. On lxdlwagpu03, hung task messages for lwa352-pipeline. It was then killed by the OOM as well as other processes. rebooted machines via ipmi.

lxdlwagpu01 is now reachable via ssh
lxdlwagpu03 is now reachable via ssh

caseyjlaw, 2023-06-20:
Thanks, Rick.
Is there any way to protect or prioritize the lwa352-pipeline process? It is critical to operations.

jack-h, 2023-06-20:
I think the most likely way to OOM these machines is to launch too many pipelines. Which shouldn't be possible, but maybe some old pipelines are not shutting down properly(?)
Or, something else has a memory leak

yupinghuang, 2023-06-20:
On gpu01 I see
```
un 16 18:23:51 lxdlwagpu01 python[11783]: 2023-06-16 18:23:51,308 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa352_start_xeng_etcd_service.py - INFO - Responded to command 'stop_pipeline' (ID 1686939831294455): OK? True
Jun 16 18:23:51 lxdlwagpu01 lwa352_start_xe[10884]: 2023-06-16 18:23:51,309 - /home/ubuntu/anaconda3/envs/casa/bin/lwa352_start_xeng_etcd_service.py - INFO - Responded to command 'stop_pipeline' (ID 1686939831294455): OK? True
Jun 16 18:23:51 lxdlwagpu01 python[11783]: 2023-06-16 18:23:51,316 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa352_start_xeng_etcd_service.py - INFO - Removing pidfile /home/ubuntu/xpipeline.1.pid
Jun 16 18:23:51 lxdlwagpu01 lwa352_start_xe[10884]: 2023-06-16 18:23:51,316 - /home/ubuntu/anaconda3/envs/casa/bin/lwa352_start_xeng_etcd_service.py - INFO - Removing pidfile /home/ubuntu/xpipeline.1.pid
Jun 16 18:23:51 lxdlwagpu01 python[11783]: 2023-06-16 18:23:51,316 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa352_start_xeng_etcd_service.py - INFO - Removing pidfile /home/ubuntu/xpipeline.1.pid
Jun 16 18:23:51 lxdlwagpu01 lwa352_start_xe[10884]: 2023-06-16 18:23:51,316 - /home/ubuntu/anaconda3/envs/casa/bin/lwa352_start_xeng_etcd_service.py - INFO - Removing pidfile /home/ubuntu/xpipeline.1.pid
Jun 16 18:23:51 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[10884]: rm: cannot remove '/home/ubuntu/xpipeline.1.pid': No such file or directory
```

and then
```
Jun 16 18:24:01 lxdlwagpu01 lwa352_start_xe[10884]: 2023-06-16 18:24:01,746 - /home/ubuntu/anaconda3/envs/casa/bin/lwa352_start_xeng_etcd_service.py - INFO - Running:/usr/bin/taskset 0x1e lwa352-pipeline.py --nchan 96 --ibverbs --gpu 0 --pipelineid 0 --useetcd --etcdhost etcdv3service.sas.pvt --ip 10.41.0.65 --po
Jun 16 18:24:01 lxdlwagpu01 python[11783]: 2023-06-16 18:24:01,746 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa352_start_xeng_etcd_service.py - INFO - Running:/usr/bin/taskset 0x1e lwa352-pipeline.py --nchan 96 --ibverbs --gpu 0 --pipelineid 0 --useetcd --etcdhost etcdv3service.s
Jun 16 18:24:01 lxdlwagpu01 python[11783]: 2023-06-16 18:24:01,748 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa352_start_xeng_etcd_service.py - INFO - Setting pidfile /home/ubuntu/xpipeline.0.pid to 11859
Jun 16 18:24:01 lxdlwagpu01 lwa352_start_xe[10884]: 2023-06-16 18:24:01,750 - /home/ubuntu/anaconda3/envs/casa/bin/lwa352_start_xeng_etcd_service.py - INFO - Setting pidfile /home/ubuntu/xpipeline.0.pid to 11860
Jun 16 18:24:01 lxdlwagpu01 python[11783]: 2023-06-16 18:24:01,750 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa352_start_xeng_etcd_service.py - INFO - Responded to command 'start_pipeline' (ID 1686939841744290): OK? True
```
and then
```
Jun 16 18:24:19 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[10884]: memory.cpp:123 cudaGetErrorString(cuda_ret) = out of memory
Jun 16 18:24:19 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[10884]: memory.cpp:123 Condition failed: cuda_ret == cudaSuccess
Jun 16 18:24:19 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[10884]: memory.cpp:123 error 32: BF_STATUS_MEM_ALLOC_FAILED
Jun 16 18:24:19 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[10884]: ring_impl.cpp:163 Condition failed: bfMalloc((void**)&new_buf, new_nbyte, _space) == BF_STATUS_SUCCESS
Jun 16 18:24:19 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[10884]: ring_impl.cpp:163 error 32: BF_STATUS_MEM_ALLOC_FAILED
Jun 16 18:24:19 lxdlwagpu01 lwa352_start_xeng_etcd_service.py[10884]: ring.cpp:55 error 32: BF_STATUS_MEM_ALLOC_FAILED
```

and then

```
Jun 16 18:26:53 lxdlwagpu01 kernel: INFO: task bifrost_etcd_br:11103 blocked for more than 120 seconds.
Jun 16 18:26:53 lxdlwagpu01 kernel:       Tainted: P           OE    4.15.0-163-generic #171-Ubuntu
Jun 16 18:26:53 lxdlwagpu01 kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jun 16 18:26:53 lxdlwagpu01 kernel: bifrost_etcd_br D    0 11103  10423 0x00000000
Jun 16 18:26:53 lxdlwagpu01 kernel: Call Trace:
Jun 16 18:26:53 lxdlwagpu01 kernel:  __schedule+0x24e/0x890
Jun 16 18:26:53 lxdlwagpu01 kernel:  ? terminate_walk+0xe5/0x100
Jun 16 18:26:53 lxdlwagpu01 kernel:  schedule+0x2c/0x80
Jun 16 18:26:53 lxdlwagpu01 kernel:  rwsem_down_read_failed+0xf0/0x160
Jun 16 18:26:53 lxdlwagpu01 kernel:  ? __alloc_pages_nodemask+0x11c/0x2c0
Jun 16 18:26:53 lxdlwagpu01 kernel:  call_rwsem_down_read_failed+0x18/0x30
Jun 16 18:26:53 lxdlwagpu01 kernel:  ? call_rwsem_down_read_failed+0x18/0x30
Jun 16 18:26:53 lxdlwagpu01 kernel:  down_read+0x20/0x40
Jun 16 18:26:53 lxdlwagpu01 kernel:  proc_pid_cmdline_read+0xcf/0x4e0
Jun 16 18:26:53 lxdlwagpu01 kernel:  __vfs_read+0x1b/0x40
Jun 16 18:26:53 lxdlwagpu01 kernel:  ? __vfs_read+0x1b/0x40
Jun 16 18:26:53 lxdlwagpu01 kernel:  vfs_read+0x8e/0x130
Jun 16 18:26:53 lxdlwagpu01 kernel:  SyS_read+0x5c/0xe0
Jun 16 18:26:53 lxdlwagpu01 kernel:  do_syscall_64+0x73/0x130
Jun 16 18:26:53 lxdlwagpu01 kernel:  entry_SYSCALL_64_after_hwframe+0x41/0xa6
Jun 16 18:26:53 lxdlwagpu01 kernel: RIP: 0033:0x7f2b8f097474
Jun 16 18:26:53 lxdlwagpu01 kernel: RSP: 002b:00007ffedab1f550 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
Jun 16 18:26:53 lxdlwagpu01 kernel: RAX: ffffffffffffffda RBX: 000000000000002b RCX: 00007f2b8f097474
Jun 16 18:26:53 lxdlwagpu01 kernel: RDX: 0000000000002000 RSI: 000055cfe3165b90 RDI: 000000000000002b
[... cut ...]
Jun 16 18:26:53 lxdlwagpu01 kernel: RBP: 000000002080200a R08: 00007f6e2b082d50 R09: 000000000000000f
Jun 16 18:26:53 lxdlwagpu01 kernel: R10: 00007f6e2b082e30 R11: 0000000000000293 R12: 00007f6e2b082d6c
Jun 16 18:26:53 lxdlwagpu01 kernel: R13: 00000000c1d00044 R14: 000000005c000002 R15: 00007f6e2b082d50
```

Dunno what these mean (there's page fault and but there's also vfs_read??) .

But if you're allocating a big chunk of the memory, weird things might happen if you stop the pipelines and then immediately start them.

caseyjlaw, 2023-06-22:
That can happen if people use the standard stop then start procedure. I'll add a `sleep` in the mnc-python control code.