ovro-lwa/lwa-issues #432: x-engine shutting down

caseyjlaw, 2023-06-20:
The x-engine was down, so I logged into each lxdlwagpu0x server and restarted the "lwa-xeng-etcd" and "bifrost_etcd_bridge" services. After a few seconds, they all shut down again. Any idea why?

```
> systemctl --user status lwa-xeng-etcd.service
Jun 20 03:49:16 lxdlwagpu02 systemd[27627]: Started LWA etcd X-Engine command etcd service.
Jun 20 03:49:16 lxdlwagpu02 lwa352_start_xe[27753]: 2023-06-20 03:49:16,851 - /home/ubuntu/anaconda3/envs/casa/bin/lwa352_start_xeng_etcd_service.
Jun 20 03:49:16 lxdlwagpu02 lwa352_start_xe[27753]: 2023-06-20 03:49:16,858 - /home/ubuntu/anaconda3/envs/casa/bin/lwa352_start_xeng_etcd_service.
Jun 20 03:49:21 lxdlwagpu02 systemd[27627]: Stopping LWA etcd X-Engine command etcd service...
Jun 20 03:49:21 lxdlwagpu02 systemd[27627]: Stopped LWA etcd X-Engine command etcd service.
```

```
> systemctl --user status bifrost_etcd_bridge.service
Jun 20 03:49:10 lxdlwagpu02 systemd[27627]: Started Bifrost <-> etcd bridge service.
Jun 20 03:49:21 lxdlwagpu02 systemd[27627]: Stopping Bifrost <-> etcd bridge service...
Jun 20 03:49:21 lxdlwagpu02 systemd[27627]: Stopped Bifrost <-> etcd bridge service.
```

jack-h, 2023-06-20:
:rage: 

Just restarted these services -- they've been running for ~8 mins so far. Will keep an eye on it.

FWIW, as part of supporting the deployment infrastructure, I moved the services from root-managed to `systemctl --user` managed. Up until then these services stopping hadn't been an issue

rh-codebase, 2023-06-20:
never had luck with systemctl --user. The deployment framework is run as a user with sudo priv. so keeping those services under root should be favorable. In fact, I'd vote to standardize all services under root. Unless someone can show that --user works reliably.

caseyjlaw, 2023-06-21:
Let's collec t a list of science services and setup. Here's what I see with a quick look:
- lwa-feng-etcd (per snap) -- pipeline@lwacalim10 with --user
- low-freq-sky -- pipeline@lwacalim10 with --user
- dashboard -- pipeline@lwacalim10 with --user
- lwa-xeng-etcd (per gpu) -- ubuntu@lxdlwagpuxx with --user
- bifrost_etcd_bridge (per gpu) -- ubuntu@lxdlwagpuxx with --user

jack-h, 2023-06-21:
I believe the complete list is/was here -- https://caltech.sharepoint.com/:w:/r/sites/ovro/projects/_layouts/15/Doc.aspx?sourcedoc=%7B923FF148-BB7E-42C8-B7AB-7CA975A02B5D%7D&file=Servers%2C%20Containers%2C%20and%20Services.docx&action=default&mobileredirect=true

yupinghuang, 2023-06-21:
Oh I forgot about this! https://wiki.archlinux.org/title/Systemd/User#Automatic_start-up_of_systemd_user_instances

The services probably got stopped as the user got logged out. Running the `enable-linger` thing should probably help.

jack-h, 2023-06-21:
Aha!
Before I do this, let's talk in the meeting later about whether this is
better or worse than just using root's systemctl

jack-h, 2023-06-21:
Snap F services and Correlator services now all run under the system systemd

yupinghuang, 2023-06-22:
Jack: "I'd like the 1s polling script to stop. The periodic cron stuff I have no problem with"

"This one can cause issues when reprogramming, which takes a few minutes and results in a massive callback queue forming. Mostly it works, but for polling at 1s cadence there are better ways to go than spamming commands at etcd"

me: Re the 1s polling -- how about setting a timeout for each command issuance and then if it fails then back off exponentially (1s, 5s, 25s, etc)?

Casey: I’m hearing two actions here. First, shut off the 1s f-eng polling. Second, update the “Servers, Containers, and Services” doc on SP.

rh-codebase, 2023-06-22:
another approach would be a priority queue where certain commands usurp. polling IMO is rarely a desired framework. If refactoring to allow a cadence to be specified it may make the priority queue easier to implement. 1 second monitor data has been a requirement from the beginning but admit I don't fully understand why it can't be met.

jack-h, 2023-06-23:
I think the main issue here is that there is a mechanism to push out 1s
monitor data from the F-engines which explicitly doesn't involve polling
commands. What seems to have happened is that the list of monitor points
desired at 1s cadence has grown beyond those covered by this mechanism, so
polling of these extras was set up during the 24hr data run.
If there is a clear list of what metadata is actually wanted (and at what
cadence) these could all be covered by the default stats pushing mechanism.

caseyjlaw, 2023-06-28:
I've updated the "Servers, Containers, and Services" with updated info (including Jack's which are no longer using --user).
@greghell Can you update the table in that doc with the cron jobs you set up for f-engine stats?

caseyjlaw, 2023-06-28:
Cross-link to #437 on coordinating a few M&C developments for data quality/flagging purposes.

jack-h, 2023-07-28:
Think the underlying issue here was fixed by moving away from using users' systemd

closed 2023-07-28 by jack-h