ovro-lwa/lwa-issues #433: Merging solar and standard configurations

dgary50, 2023-06-23:
@caseyjlaw @yupinghuang 
There are a few small differences between the configuration for solar observations and that for standard observations, listed below.

1. The beamforming calibrations use only core antennas to keep the beam large enough not to resolve the Sun.
2. The selection of antennas for fast visibilities may differ from others.
3. The current standard configuration file removed dr2 in order to "protect" the solar beam.  

These differences are small, but to maintain them as we do now the solar observers have to take the current standard configuration and edit it to correct for these issues.  To be safe, in the case of possible changes to the standard configuration, we are doing this every observation.

Gregg Hallinan suggests that we should have only one configuration for all, so to achieve this we need to identify ways to solve these small differences.  Since issue #1 pertains only to the solar beam 2, perhaps we can have beam-specific locations for cal files.  Issue #2 may not be a real issue if we are the only users of fast visibilities (or if other users of fast visibilities are happy enough with our selection).  Issue #3 is easy to fix by just adding dr2 back to the list of recorders in the standard configuration file.  There is already a warning issued if someone tries to start dr2, so that might be enough.

caseyjlaw, 2023-06-25:
Your suggestion to resolve Issue 1 is a good one. We could probably get started on this by opening a request for the calibration pipeline script to generate two sets of solutions in two locations.
I agree that issue 2 isn't an obvious problem yet. It is also easy to fix by overloading the configuration.
Issue 3 here can be controlled via the `recorder` argument that is supported by (I think) all `mnc-python` commands. The logic in the configuration and recorder control should handle it, so I think this is just a matter of training and documentation.

marinanderson, 2023-06-27:
Yuping has active pull request out on this. Waiting for review by Casey?

caseyjlaw, 2023-06-27:
great addition. yuping can merge when ready.

dgary50, 2023-06-27:
Can someone provide a description of the change and what is happening under the hood?  Is it that there are two calibration file locations, one for beam 2 and one for "not beam 2"?  Is this handled automatically when the calibratebeams=True switch is set in configure_xengine, depending on which recorders are specified?  If someone were to configure, say, dr1 and dr2 in the same command, would the two get different calibration files?  Thanks.

yupinghuang, 2023-06-27:
(it's not deployed yet but here's a description)
Antenna selection for the beams is done at `control_bf`, not at `configure_xengine`.

So for core-only beamforming, you don't have to do anything different from before since the default for the newly added `weight` argument is `core` (i.e. non-core antennas get a weight of zero). 

The goal is for all the beams to use the same calibration files and handle additional antenna flags at `control_bf`. For now we'll try to keep the daytime settings calibration files in there, but I'll follow up with a change that handles day/night settings.

"If someone were to configure, say, dr1 and dr2 in the same command, would the two get different calibration files? "
No they will not.

dgary50, 2023-06-27:
Okay, that makes sense, thanks.

caseyjlaw, 2023-12-14:
I think this is resolved and tested well via solar observing.

closed 2023-12-14 by caseyjlaw