ovro-lwa/lwa-issues #434: New account on calim for Vinand

marinanderson, 2023-06-23:


kembyct, 2023-06-23:
I wonder what is the ssh public keys.

VinAstro, 2023-06-24:
Hi Kemby,

I have attached two ssh keys. One for my linux box and the other for my
windows machine. Please add them to the calim machines.

Cheers
Vinand

caseyjlaw, 2023-08-02:
@VinAstro do you have access? I see your public key on some servers.

VinAstro, 2023-08-02:
I have access from my linux machine but not from my windows box. I gave
both keys to Kemby.

Cheers,
Vinand

VinAstro, 2023-08-02:
Also, there is a directory with my name on it but I cannot cd into it. It
says permission denied.

Vi

kembyct, 2023-08-02:
There must be something wrong with the Windows keys. You can generate a new key and send it. I'll check the permission the directory.

caseyjlaw, 2023-10-12:
I think Vinand has an account on cal-im, so I'll close this issue. Reopen if I misunderstood.